xbianonpi/xbian #656: Screen adjustments are not persistant

NilsNojje, 2015-01-06:
Every time that the Pi reboots, the settings for screen size adjustment are reset.

anaconda, 2015-01-07:
Please ensure you're using the latest version of xbian-package-xbmc.
If not, upgrade, try to reproduce and provide a debug log (see http://kodi.wiki/view/Debug_log/Easy) - I believe this has been fixed a long time ago.

NilsNojje, 2015-01-07:
I updated all but same problem persists.
One thing I noticed last reboot, was that the initial resolution was all OK, but after a few seconds into the Xbian UI it got messed up again.

UI resolution is set to 1080p60 and the Pi is connected to a Onkyo receiver and is all set up with CEC and Samsung Anynet for the TV. Everything has worked just fine untill about a week ago.

Pastebin link of last log -> http://pastebin.com/qPqx2rds

mk01, 2015-01-13:
@NilsNojje 

I assume it is the same as this one https://github.com/xbianonpi/xbian/issues/648

it is fixed in latest packages, ... but all are already Kodi, v14. 
uncomment STAGING repository from /etc/apt/sources.list.d/xbian.list, run
```
apt-get update
apt-get install xbian-package-xbmc
```

NilsNojje, 2015-01-15:
@mk01 Uncommented staging in xbian.list and yes, that did the trick.
So resolution problem solved, but after performing this I could no longer connect to my SMB server.

password.xml contain the correct info and even if typing manually while browsing the network, it always fail on connecting to my media server.

mkreisl, 2017-09-15:
Closing due to lack of activity. Reopen if you feel this issue is still relevant.

closed 2017-09-15 by mkreisl