xbianonpi/xbian #657: xbian-update obliterates dhclient.conf

palswim, 2015-01-13:
Whenever `xbian-update` has an update, I lose all of the data in `/etc/dhcp/dhclient.conf` configuration file, save for three lines:
> initial-interval 1;
> reboot 4;
> script "/etc/xbian-udhcpc/dhclient-script-xbian";
Thankfully, I had the initial `dhclient.conf` from the base XBian (1.0r2) image from which to restore the configuration, but I have to restore that backup each time `xbian-update` updates.

mk01, 2015-01-14:
can you try 
```
touch /run/noreboot
wget http://download.serveftp.net/other/xbian-update20150114-0.deb
dpkg -i xbian-update20150114-0.deb
 ```

palswim, 2015-01-14:
A diff confirms that I have an identical `/etc/dhcp/dhclient.conf` file to the file I had before the update. Thanks!

closed 2015-01-14 by palswim