xbianonpi/xbian #659: Xbian-config not autostarting

Smultie, 2015-01-28:
Since a couple of days, when I SSH into my pi, Xbian-config no longer autostarts.

"sudo xbian-config" does the trick, of course.
I've checked the settings, SSH-2-6 and it says "Successfully enabled xbian-config autostart".

CurlyMoo, 2015-01-28:
Could it be that you use the manual override:
```
cat /home/xbian/.xbian-config-start
0
```
If you make it a `1` is should work again.

Smultie, 2015-01-28:
Nope, it's at 1

CurlyMoo, 2015-01-28:
Does this file still exists?
`/home/xbian/.xbian-config-start.sh`

And do you have this at the end of `/home/xbian/.bashrc`?
```
if [ -z "${STY}" ] && [ -z "${TMUX}" ] \
  && [ -e /home/xbian/.xbian-config-start.sh ]; then
    . /home/xbian/.xbian-config-start.sh
fi
```

Smultie, 2015-01-28:
xbian@xbian ~ $ cat /home/xbian/.xbian-config-start.sh
if grep -qx '1' /home/xbian/.xbian-config-start; then
    sudo xbian-config
    echo
    echo '# to disable xbian-config:'
    echo 'echo 0 > ~/.xbian-config-start'
fi
________________________________________________________
cat /home/xbian/.bashrc

if [ -z "${STY}" ] && [ -z "${TMUX}" ] \
  && [ -e /home/xbian/.xbian-config-start.sh ]; then
    . /home/xbian/.xbian-config-start.sh
fi

Bastlwastl2010, 2015-01-29:
Same Problem here all is set right
I make a fresh install and after update the error occurs, also the dist-upgrade do not work see:
http://forum.xbian.org/thread-2655-post-26477.html#pid26477

RolfTimmerman, 2015-01-30:
I can confirm that I have the same issue (on the forum I'm known as Carpenter)

CurlyMoo, 2015-01-31:
What xbian-config package is this: `apt-cache policy xbian-package-config-shell`

Smultie, 2015-01-31:
xbian@xbian ~ $ apt-cache policy xbian-package-config-shell
xbian-package-config-shell:
  Installed: 2.1.15-2
  Candidate: 2.1.15-2
  Version table:
 *** 2.1.15-2 0
        500 mirror://apt.xbian.org/mirror.txt/ stable/main armhf Packages
        100 /var/lib/dpkg/status
     2.1.15-1 0
        500 mirror://apt.xbian.org/mirror.txt/ stable/main armhf Packages
     2.1.15 0
        500 mirror://apt.xbian.org/mirror.txt/ stable/main armhf Packages
     2.1.14-2 0
        500 mirror://apt.xbian.org/mirror.txt/ stable/main armhf Packages
     2.1.14 0
        500 mirror://apt.xbian.org/mirror.txt/ stable/main armhf Packages
     2.1.13 0
        500 mirror://apt.xbian.org/mirror.txt/ stable/main armhf Packages
     2.1.12 0
        500 mirror://apt.xbian.org/mirror.txt/ stable/main armhf Packages
     2.1.11 0
        500 mirror://apt.xbian.org/mirror.txt/ stable/main armhf Packages
     2.1.10 0
        500 mirror://apt.xbian.org/mirror.txt/ stable/main armhf Packages
     2.1.9 0
        500 mirror://apt.xbian.org/mirror.txt/ staging/main armhf Packages
     2.1.8-6 0
        500 mirror://apt.xbian.org/mirror.txt/ stable/main armhf Packages
     2.1.8-5 0
        500 mirror://apt.xbian.org/mirror.txt/ stable/main armhf Packages
     2.1.8-4 0
        500 mirror://apt.xbian.org/mirror.txt/ stable/main armhf Packages
     2.1.8-3 0
        500 mirror://apt.xbian.org/mirror.txt/ stable/main armhf Packages

CurlyMoo, 2015-01-31:
First check if you still have a `/home/xbian/.profile`. If not try this:
```
cat << \EOF > /home/xbian/.profile
if [ -n "$BASH_VERSION" ]; then
    # include .bashrc if it exists
    if [ -f "$HOME/.bashrc" ]; then
        . "$HOME/.bashrc"
    fi
fi
EOF
```

Bastlwastl2010, 2015-01-31:
.profile ist not present in my home
After create this xbian config start at login
Still have the error in dist-upgrade

CurlyMoo, 2015-01-31:
That's why you shouldn't use `dist-upgrade`.

Smultie, 2015-01-31:
That fixed it for me CurlyMoo

closed 2015-01-31 by CurlyMoo

anaconda, 2015-02-07:
This turned out to be a bug in xbian-package-xbianhome and has been fixed.

To be honest, I don't think there's any bug as the idea was that that package should only be installed when building the image - I think it should be removed from the repo.

CurlyMoo, 2015-02-07:
I don't agree. An example is that we recently added the xbian user to the cdrom group. That change has been pushed by xbianhome.

Smultie, 2015-07-14:
The issue somehow returned after me putting back a backup on my SD/USB.
However, the above didn't work this time. 
Any way to get the autostarting back?

CurlyMoo, 2015-07-14:
You should be able to enable xbian-config autostart inside xbian-config.

Smultie, 2015-07-14:
That's the first thing I tried, but that didn't work.
I created /home/xbian/.xbian-config-start.sh and after that it worked again.