xbianonpi/xbian #665: RPi 2 not booting

Smultie, 2015-02-05:
I know this has already been mentioned, but might be better to get it on github as well.

rikardo1979, 2015-02-07:
same here, it just hangs on card resizing. do not have any other card to ry now, but with same HW OE build works just fine

CurlyMoo, 2015-02-07:
The version from yesterday is confirmed to work by various users and myself.

rikardo1979, 2015-02-07:
thanks for reply @CurlyMoo 
are u refering to this one? http://sourceforge.net/projects/xbian/files/testing/XBian_2015.02.06_rpi2.img.gz
cos thats the one I have tried and it doesnt work here as I mentioned above

CurlyMoo, 2015-02-07:
That's the one. Make sure your power supply is sufficient. There are a lot of issues about power supplies and the Raspberry Pi 2. Also on the OpenElec forum. @smultie posted this about the image `2015.02.06` which we released w/o testing.

There is also an `weekly` from today.

rikardo1979, 2015-02-07:
sure,
My PSU is solid stable 2.5A and as mentioned same hardware (RPi2, PSU, MicroSD card, cables, and whole setup) 
tried OpenElec and it boots straight, no issues, all super fast etc
It just hangs on resizing and it just froze an stays forever. when I tried unplug and plug again it wont even go back to that stage

CurlyMoo, 2015-02-07:
Did you try the latest `weekly`? Problem is that i can't confirm and a lot of users are using that version w/o issues. So i have no clue where to look. Maybe @mk01?

rikardo1979, 2015-02-07:
just started to play with it now, so have tried only that build so far 
downloading this now http://sourceforge.net/projects/xbian/files/release/XBian_2015.02.07_rpi2.img.gz
so will see in few minutes

rikardo1979, 2015-02-07:
so with that build it ends here http://oi60.tinypic.com/2r6368j.jpg

CurlyMoo, 2015-02-07:
I have no clue. Need to test myself later.

rikardo1979, 2015-02-07:
tried to power cycle but its no go http://oi59.tinypic.com/28lgx2d.jpg
probably no help at all

CurlyMoo, 2015-02-07:
Booting just fine with latest image:
```
Uncompressing Linux... done, booting the kernel.
▒To dropdown to shell, press and hold shift on usb attached keyboard...
udevd[93]: error opening ATTR{/sys/devices/platform/mmc-bcm2835.0/mmc_host/mmc0/mmc0:b368/block/mmcblk0/queue/iosched/target_latency} for writing: Permission denied

Loading initram modules ...
Looking for keyboards
Scanning for Btrfs filesystems

Disk /dev/mmcblk0: 245216 cylinders, 4 heads, 16 sectors/track
Old situation:
Units = sectors of 512 bytes, counting from 0

   Device Boot    Start       End   #sectors  Id  System
/dev/mmcblk0p1   *      2048     71679      69632   b  W95 FAT32
/dev/mmcblk0p2        141312   1135487     994176  83  Linux
/dev/mmcblk0p3             0         -          0   0  Empty
/dev/mmcblk0p4             0         -          0   0  Empty
New situation:
Units = sectors of 512 bytes, counting from 0

   Device Boot    Start       End   #sectors  Id  System
/dev/mmcblk0p1   *      2048     71679      69632   b  W95 FAT32
/dev/mmcblk0p2        141312  15693823   15552512  83  Linux
/dev/mmcblk0p3             0         -          0   0  Empty
/dev/mmcblk0p4             0         -          0   0  Empty
udevd[92]: error opening ATTR{/sys/devices/platform/mmc-bcm2835.0/mmc_host/mmc0/mmc0:b368/block/mmcblk0/queue/iosched/target_latency} for writing: Permission denied

Successfully wrote the new partition table

Re-reading the partition table ...

Partition resized...
udevd[92]: error opening ATTR{/sys/devices/platform/mmc-bcm2835.0/mmc_host/mmc0/mmc0:b368/block/mmcblk0/queue/iosched/target_latency} for writing: Permission denied

Resize '/rootfs' of 'max'
FSSync '/rootfs'
Resize '/rootfs' of '-250M'
FSSync '/rootfs'
Checking that no-one is using this disk right now ...
OK

Disk /dev/mmcblk0: 245216 cylinders, 4 heads, 16 sectors/track
Old situation:
Units = mebibytes of 1048576 bytes, blocks of 1024 bytes, counting from 0

   Device Boot Start   End    MiB    #blocks   Id  System
/dev/mmcblk0p1   *     1     34     34      34816    b  W95 FAT32
/dev/mmcblk0p2        69   7662   7594    7776256   83  Linux
/dev/mmcblk0p3         0      -      0          0    0  Empty
/dev/mmcblk0p4         0      -      0          0    0  Empty
New situation:
udevd[92]: error opening ATTR{/sys/devices/platform/mmc-bcm2835.0/mmc_host/mmc0/mmc0:b368/block/mmcblk0/queue/iosched/target_latency} for writing: Permission denied
Units = mebibytes of 1048576 bytes, blocks of 1024 bytes, counting from 0

   Device Boot Start   End    MiB    #blocks   Id  System
/dev/mmcblk0p1   *     1     34     34      34816    b  W95 FAT32
/dev/mmcblk0p2        69   7417   7349    7525376   83  Linux
/dev/mmcblk0p3         0      -      0          0    0  Empty
/dev/mmcblk0p4         0      -      0          0    0  Empty
Successfully wrote the new partition table

Re-reading the partition table ...

If you created or changed a DOS partition, /dev/foo7, say, then use dd(1)
to zero the first 512 bytes:  dd if=/dev/zero of=/dev/foo7 bs=512 count=1
(See fdisk(8).)
Checking that no-one is using this disk right now ...
OK

Disk /dev/mmcblk0: 245216 cylinders, 4 heads, 16 sectors/track
Old situation:
Units = sectors of 512 bytes, counting from 0

   Device Boot    Start       End   #sectors  Id  System
/dev/mmcblk0p1   *      2048     71679      69632   b  W95 FAT32
/dev/mmcblk0p2        141312  15192063   15050752  83  Linux
/dev/mmcblk0p3             0         -          0   0  Empty
/dev/mmcblk0p4             0         -          0   0  Empty
New situation:
Units = sectors of 512 bytes, counting from 0

   Device Boot    Start       End   #sectors  Id  System
/dev/mmcblk0p1   *      2048     71679      69632   b  W95 FAT32
/dev/mmcblk0p2        141312  15192063   15050752  83  Linux
/dev/mmcblk0p3      15192064  15693823     501760  82  Linux swap / Solaris
/dev/mmcblk0p4             0         -          0   0  Empty
Successfully wrote the new partition table

Re-reading the partition table ...

If you created or changed a DOS partition, /dev/foo7, say, then use dd(1)
to zero the first 512 bytes:  dd if=/dev/zero of=/dev/foo7 bs=512 count=1
(See fdisk(8).)
udevd[92]: error opening ATTR{/sys/devices/platform/mmc-bcm2835.0/mmc_host/mmc0/mmc0:b368/block/mmcblk0/queue/iosched/target_latency} for writing: Permission denied

Setting up swapspace version 1, size = 256897024 bytes
UUID=c456b7a1-4bc5-45ac-964f-0396738b7deb
Mounting root as: mount -t btrfs -o rw,subvol=root/@,thread_pool=2,autodefrag,compress=lz4,commit=120 /dev/mmcblk0p2 /rootfs
udevd[92]: error opening ATTR{/sys/devices/platform/mmc-bcm2835.0/mmc_host/mmc0/mmc0:b368/block/mmcblk0/queue/iosched/target_latency} for writing: Permission denied

udevd[92]: error opening ATTR{/sys/devices/platform/mmc-bcm2835.0/mmc_host/mmc0/mmc0:b368/block/mmcblk0/queue/iosched/target_latency} for writing: Permission denied

Resize '/rootfs' of 'max'
FSSync '/rootfs'
Moving root
udevd[92]: error opening ATTR{/sys/devices/platform/mmc-bcm2835.0/mmc_host/mmc0/mmc0:b368/block/mmcblk0/queue/iosched/target_latency} for writing: Permission denied

udevd[92]: error opening ATTR{/sys/devices/platform/mmc-bcm2835.0/mmc_host/mmc0/mmc0:b368/block/mmcblk0/queue/iosched/target_latency} for writing: Permission denied

udevd[92]: error opening ATTR{/sys/devices/platform/mmc-bcm2835.0/mmc_host/mmc0/mmc0:b368/block/mmcblk0/queue/iosched/target_latency} for writing: Permission denied

Switching root
▒rpcbind: cannot create socket for udp6
rpcbind: cannot create socket for tcp6
udevd[1186]: error opening ATTR{/sys/devices/platform/mmc-bcm2835.0/mmc_host/mmc0/mmc0:b368/block/mmcblk0/queue/iosched/target_latency} for writing: Permission denied

```

After the switch_root i can login after 5 minutes.

belese, 2015-02-07:
#rikardo1979
i've got a sdcard whith similar behaviour on pi1
try to plug the sd card on pc  (after resising freeze), and resize second partition with gparted, replug on pi and maybe it will work ;--)
The problem on my sd card was the partiton was resising, but not the filesystem.
other os was working too.
but it was one year late at least, so try and see ;-)

rikardo1979, 2015-02-07:
just wondering as OE has no problem with same card, and I have no keyboard attached to RPi (dont have one at all) so I cant do much.
may try that @belese

CurlyMoo, 2015-02-07:
@rikardo1979, does the red led blink when booting?

rikardo1979, 2015-02-07:
RED and GREEN blinks once, than RED stay solid and thats all I get 
it stays like that. Even in multiple attempts to powercycle is always the same. it ends in same point

rikardo1979, 2015-02-07:
thanks @belese 
finally I found the time to try that and it does work ;) but for some reason I have 4 partition
But at least XBian booted now, even in weird @640 resolution.
but now after I have done the basic video settings and reboot it hangs on something else
will investigate in about an hour again

rikardo1979, 2015-02-07:
not sure if this is OK? http://snag.gy/qxfm9.jpg
what I have done
flashed the latest image, than as there were two unallocated partitions I have created those and formated to ext4
after that I have plugged into RPi and than it finally booted straight into XBian in correct resolution etc
than I have set all the basics, like resolution, audio out etc
but when I tried to reboot it hangs on network detection now
Any clue?

rikardo1979, 2015-02-08:
only way how I can make this working is by commenting 1st line
#initramfs initramfs.gz  0x1000000

rikardo1979, 2015-02-08:
looks the issue is resolved. seems that the problem was with my computer running Ubuntu
need to test bit more but looks all is good

closed 2015-02-10 by CurlyMoo