xbianonpi/xbian #666: mv command to cifs mount hangs rpi.  cp does not.

ggggh, 2015-02-06:
Using the "mv" command to move a large (>500Mb) file to a cifs mounted share hangs the rpi at some variable point while moving the files (file only partly reaches target, is not deleted from source, and rpi has to be hard-rebooted to regain ssh terminal).  cp works fine, on the other hand.

-Using RPI B
-Tried with a brand new installation on a new SD card - same issue.
-Rpi doesn't seem to show any power-related issues, unless there's some reason mv has an extraordinary power requirement
-All packages are up to date

ggggh, 2015-02-09:
Actually, just had a "cp" hang the rpi too.  It is less common than wit "mv", but also ocurring.

f1vefour, 2015-02-27:
Try lowering your overclock.

ggggh, 2015-02-27:
Ok, had it on standard xbian setup though. Just lowered to "Modest" and will monitor.

By the way, realized that the behaviour was quite strange.  One of the times the RPi seemed frozen to ssh and UI.., but every few minutes it was, for a few seconds, transferring a burst of data (could monitor the network traffic and file size increase in the NAS).  Left alone for a few hours it actually finished the transfer.

f1vefour, 2015-02-27:
When it does this again grab dmesg and post it to pastebin and link here please.

closed 2017-09-07 by mkreisl