xbianonpi/xbian #667: Update to Kodi - Settings gone and no more LAN

RolfTimmerman, 2015-02-07:
Please refer to http://forum.xbian.org/thread-2704.html.

I have the same issue as mentioned in the post above but I don't have LAN access as well. Doesn't matter if I select Static or DHCP.

Fabio72, 2015-02-07:
I symlinked .xmbc to .kody and kody.bin start fine, according to the logs

video seems 640*480
also I get ERROR: Unable to load libmpeg2.so.0, reason: libmpeg2.so.0: cannot open shared object file: No such file or directory

http://pastebin.com/HiCVUR7t

I reinstalled libmpeg2-4 with apt-get and I get rid of the previous error, but the video is still stuttering like it's without hardware acceleration

Fabio72, 2015-02-07:
seems only iso dvd play is broken

CurlyMoo, 2015-02-07:
@Fabio72, your problem has been fixed with this commit: https://github.com/xbianonpi/xbian-package-xbmc/commit/0f73fd6686b82f56e0baa4e5cc9994a78891a6b5

@RolfTimmerman, does execution this command work?
`chmod 777 /home/xbian/.kodi/userdata/guisettings.xml`

RolfTimmerman, 2015-02-07:
I rolled back one of my snapshots in the hope that  it would work again but it got only worse so I will have to reinstall everything again. Also after executing the command nothing changed .

CurlyMoo, 2015-02-07:
I also notice that somehow Kodi reverts the `guisettings.xml` permissions:

```
# ls -Al /home/xbian/.kodi/userdata/ | grep guisettings
lrwxrwxrwx 1 xbian xbian  34 Feb  7 22:12 guisettings.xml -> /media/config/guisettings.xml.kodi
# ls -Al /media/config/ | grep guisettings
-rwxrwxrwx 1 root root 26153 Feb  8 00:10 guisettings.xml.kodi
```

```
service xbmc restart
```

```
# ls -Al /home/xbian/.kodi/userdata/ | grep guisettings
lrwxrwxrwx 1 xbian xbian  34 Feb  7 22:12 guisettings.xml -> /media/config/guisettings.xml.kodi
# ls -Al /media/config/ | grep guisettings
-r-xr-xr-x 1 root root 42652 Feb  8 00:31 guisettings.xml.kodi
```

For some reason `guisettings.xml` get reset on `Kodi` start triggering the issue users have.

CurlyMoo, 2015-02-07:
Ok, found the issue. Kodi opens the `guisettings.xml` with permission 555:
```
stat64("/home/xbian/.kodi/userdata/guisettings.xml", {st_mode=S_IFREG|0555, st_size=42652, ...}) = 0
open("/home/xbian/.kodi/userdata/guisettings.xml", O_RDONLY|O_LARGEFILE) = 10
```
Thereby resetting the permissions so it can't be written by itself anymore. @anaconda, can you checked if this is fixed upstream or we should patch it?

RolfTimmerman, 2015-02-09:
I've installed a fresh copy and the issues are gone so it must be an upgrade bug or something. If you need some files for investigation, I've made a backup before installing a fresh copy.

CurlyMoo, 2015-02-10:
Can you guys check if setting the `guisettings.xml` permission to 644 works?

First shutdown Kodi:
`service xbmc stop`

`chmod 644 /home/xbian/.kodi/userdata/guisettings.xml`

And start kodi:
`service xbmc start`

RolfTimmerman, 2015-02-11:
As mentioned I've installed a fresh copy and everything is working flawlessly, so I won't be of any help I'm afraid. This LAN issue I had was probably due to issue #672 because I had installed OpenVPN with an iptables entry.

CurlyMoo, 2015-02-11:
Ok

f1vefour, 2015-02-27:
This is still an issue.  I installed the latest RPI 2 build and updated it, kodi still loses set resolution (actually loses video calibration).

f1vefour, 2015-02-27:
No matter what I do my calibration wont stick, I even calibrated and chmod 444 guisettings.xml as a test. Nothing works.

CurlyMoo, 2015-02-27:
`644`?

f1vefour, 2015-02-27:
No 444 read only so whatever is changing it couldn't change it.

CurlyMoo, 2015-02-27:
It was a suggestion :smile:

f1vefour, 2015-02-27:
I calibrate, restart kodi and something is erasing my calibration. I calibrated, made file read only, kodi still not using my calibration. Something is really wrong.

f1vefour, 2015-02-27:
I had it 644, I tried 777...nothing. What is changing it?

f1vefour, 2015-02-27:
Also everytime I stop kodi it turns my tv off.

mk01, 2015-02-27:
@f1vefour 

the last happens based on CEC setting "Turn devices XYZ off when KODI stops". But that same setting is used as device iteration when Screensaver activation should turn off devices.

Patch that is not a problem, it bothers me too. Only never that much, that I change the particular line. It was worse before when XBMC based on that setting was turning TV off even when exiting due to REBOOT or SHUTDOWN call, what assumes (at least my case) you WANT to see the screen, but that we fixed (so reboot and shutdown IS recognised as not XBMC QUIT).

Hellcube, 2015-03-19:
I have same problem with calibration. It have some thing to do with CEC initialization. When Kodi starts calibration is ok then i see sign initializing CEC then calibration goes wild.