xbianonpi/xbian #668: DVD ISO Playback broken after updating to kodi

Fabio72, 2015-02-07:
After the last update to kodi, iso dvd playback suffers lots of problem.
I had to fix the lack of libmpeg2 but I still have issues:
- if I enable the bypass of introduction menu, the playback doesn't start: I can ear audio for a while then everything stop.
- very often I get an xbmc crash and restart
- Video and audio stutter like software decoding
If I play directly the vob files the playback is fine
http://pastebin.com/HiCVUR7t
I will paste later a log whith debug enabled

CurlyMoo, 2015-02-10:
> if I enable the bypass of introduction menu, the playback doesn't start: I can ear audio for a while then everything stop.

I do we do this?

> very often I get an xbmc crash and restart

Random or at a certain amount of time. Good to know to test ourselves.

> Video and audio stutter like software decoding If I play directly the vob files the playback is fine 

I have no issues in regard to this point.

Fabio72, 2015-02-11:
Logfiles:
first try of playback -> kodi restart immediatly
http://pastebin.com/E1JFLfhN
the log is kodi.old.log

second try, with or without bypass of introduction enabled
CPU show 100% anda video and audio stutter
in the log I try to play two or three times
http://pastebin.com/0X3Tsud1

In the log you can search "lost and found" as dvd playback start
The same iso played fine with xbmc

mkreisl, 2015-02-13:
Same here (Kodi 14.1)
Kodi always crashes if Settings->Videos->Acceleration->Allow hardware acceleration (OMXPlayer) is checked.
If only Allow hardware acceleration (MMALPlayer) is checked, Kodi does not crash but video is in slow motion and stuttering, and video resolution changes from 1920x1080p to 800x600

anaconda, 2015-02-13:
@popcornmix any idea here? I see lots of
`ERROR: CMMALRenderer::GetImage - mmal_queue_get failed`
in http://pastebin.com/0X3Tsud1 ("second try")
This is a vanilla Helix build (no newclock4).

popcornmix, 2015-02-13:
@anaconda can you confirm if helix_rbp_backports has the same issue?

Fabio72, 2015-02-13:
Now kodi restart even playing vob files.

20:36:57 T:3034943488  NOTICE: DVDPlayer: Opening: /srv/DualServer/Videos/lostandfound/VTS_02_1.VOB
20:36:57 T:3034943488 WARNING: CDVDMessageQueue(player)::Put MSGQ_NOT_INITIALIZED
20:36:57 T:2901992464  NOTICE: Thread CMMALRenderer start, auto delete: false
20:36:57 T:2755650576  NOTICE: Thread DVDPlayer start, auto delete: false
20:36:57 T:2755650576  NOTICE: Creating InputStream
20:36:57 T:2755650576  NOTICE: Creating Demuxer
20:36:58 T:2755650576  NOTICE: OMXPlayerUnsuitable OMXPlayer unsuitable due to video codec (2:0/1)
20:37:03 T:3034210304  NOTICE: special://profile/ is mapped to: special://masterprofile/
20:37:03 T:3034210304  NOTICE: -----------------------------------------------------------------------
20:37:03 T:3034210304  NOTICE: Starting Kodi (14.1 Git:2015-02-05-38e4046-dirty). Platform: Linux ARM 32-bit

popcornmix, 2015-02-13:
> 20:36:58 T:2755650576 NOTICE: OMXPlayerUnsuitable OMXPlayer unsuitable due to video codec (2:0/1)

suggests you don't have an MPG2 codec licence.

Fabio72, 2015-02-13:
I lost the entry in config.txt...
Ok, now vob play is fixed so it's not an mpeg2 issue

anaconda, 2015-02-13:
I will try to test tomorrow, but I don't think there would be any difference. That error seems to come from:
https://github.com/popcornmix/xbmc/blob/helix_rbp_backports/xbmc/cores/VideoRenderers/MMALRenderer.cpp#L260-L263
which is identical in both master and Helix.
Also note that this seems to work fine for others: https://github.com/xbianonpi/xbian/issues/668#issuecomment-73808661

http://pastebin.com/0X3Tsud1 - here the user reports "_CPU show 100% and video and audio stutter_" (https://github.com/xbianonpi/xbian/issues/668#issuecomment-73928085) but according to the log MPG2 licence is enabled.

Fabio72, 2015-02-17:
no news:
kodi reboot playing iso files
http://pastebin.com/yQxJ6DK1
http://pastebin.com/DdnEL2Eu

Fabio72, 2015-02-23:
I found this thread:
http://forum.kodi.tv/showthread.php?tid=192380&pid=1823127#pid1823127
And some GITHUB issues about ISO or IFO playback on smb or NFS shares
https://github.com/xbmc/xbmc/pull/5534
https://github.com/OpenELEC/OpenELEC.tv/issues/3692

My ISO are all on NFS system mountpoints.
I cannot play also the VIDEO_TS folder as long as there's an IFO file inside.
VOB files play perfectly while an ISO play makes kodi.bin restart

mkreisl, 2015-02-23:
Strange things happens ...

After updating to Kodi 14.2 Beta 2 and Fabio's last post I played a little bit more on DVD-ISO's on my test-system.

Now I have following situation:
1. Kodi crashes always if OMXPlayer and MMAL hw accel is checked and it does not crash if only one of them is checked.
2. Kodi plays in slow motion with 100%CPU usage if MMAL accel is checked, video and audio is completely async.
3. Kodi plays perfectly with around 30-50% CPU usage if OMXPlayer is checked
4. It makes NO difference if the iso file is on a locally attached usb disk (ext4 fs), on smb share which is mounted by Kodi or on NFS share which is mounted by autofs (which is my default)
5. The screen resolution change to 800x600 has gone (other Kodi issue which has been fixed since 14.2 Beta 2)

I tested it with 3 different ISO files and one VIDEO_TS folder

Fabio72, 2015-02-23:
After disabling MMAL I can play an ISO again. I must do more testing because I'm now playing an home made dvd with slide show and audio sometimes stutters, but maybe is the DVD...

closed 2015-03-04 by Fabio72