xbianonpi/xbian #669: System wants sysvinit

florian-asche, 2015-02-08:
wohnzimmer1:/etc/apt# apt-get -f install
Reading package lists... Done
Building dependency tree
Reading state information... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
wohnzimmer1:/etc/apt# apt-get dist-upgrade
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following packages will be REMOVED:
  upstart xbian-package-config-xbmc xbian-package-lirc xbian-package-xbmc xbian-update
The following NEW packages will be installed:
  sysvinit
WARNING: The following essential packages will be removed.
This should NOT be done unless you know exactly what you are doing!
  xbian-update xbian-package-lirc (due to xbian-update) upstart (due to xbian-update) xbian-package-xbmc (due to xbian-update)
0 upgraded, 1 newly installed, 5 to remove and 0 not upgraded.
Need to get 129 kB of archives.
After this operation, 82.8 MB disk space will be freed.
You are about to do something potentially harmful.
To continue type in the phrase 'Yes, do as I say!'
 ?]

CurlyMoo, 2015-02-08:
What did you try to install before you got to this point?

florian-asche, 2015-02-08:
Hi,
nothing. Is there a way to find out, why he wants to install it ?

wohnzimmer1:~# apt-get clean all

wohnzimmer1:~# apt-get update
Hit http://mirrordirector.raspbian.org wheezy Release.gpg
Hit http://mirrordirector.raspbian.org wheezy Release
Hit http://mirrordirector.raspbian.org wheezy/main armhf Packages
Hit http://mirrordirector.raspbian.org wheezy/contrib armhf Packages
Hit http://www.mirrorservice.org/sites/apt.xbian.org/ stable Release.gpg
Hit http://www.mirrorservice.org/sites/apt.xbian.org/ stable Release
Hit http://mirrordirector.raspbian.org wheezy/non-free armhf Packages
Hit http://mirrordirector.raspbian.org wheezy/rpi armhf Packages
Hit http://www.mirrorservice.org/sites/apt.xbian.org/ stable/main armhf Packages
Hit http://www.mirrorservice.org/sites/apt.xbian.org/ stable/rpi-wheezy armhf Packages
Ign http://www.mirrorservice.org/sites/apt.xbian.org/ stable/main Translation-en_US
Ign http://www.mirrorservice.org/sites/apt.xbian.org/ stable/main Translation-en
Ign http://www.mirrorservice.org/sites/apt.xbian.org/ stable/rpi-wheezy Translation-en_US
Ign http://www.mirrorservice.org/sites/apt.xbian.org/ stable/rpi-wheezy Translation-en
Ign http://mirrordirector.raspbian.org wheezy/contrib Translation-en_US
Ign http://mirrordirector.raspbian.org wheezy/contrib Translation-en
Ign http://mirrordirector.raspbian.org wheezy/main Translation-en_US
Ign http://mirrordirector.raspbian.org wheezy/main Translation-en
Ign http://mirrordirector.raspbian.org wheezy/non-free Translation-en_US
Ign http://mirrordirector.raspbian.org wheezy/non-free Translation-en
Ign http://mirrordirector.raspbian.org wheezy/rpi Translation-en_US
Ign http://mirrordirector.raspbian.org wheezy/rpi Translation-en
Reading package lists... Done

wohnzimmer1:~# apt-get upgrade
Reading package lists... Done
Building dependency tree
Reading state information... Done
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

wohnzimmer1:~# apt-get dist-upgrade
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following packages will be REMOVED:
  upstart xbian-package-config-xbmc xbian-package-lirc xbian-package-xbmc xbian-update
The following NEW packages will be installed:
  sysvinit
WARNING: The following essential packages will be removed.
This should NOT be done unless you know exactly what you are doing!
  xbian-update xbian-package-lirc (due to xbian-update) upstart (due to xbian-update) xbian-package-xbmc (due to xbian-update)
0 upgraded, 1 newly installed, 5 to remove and 0 not upgraded.
Need to get 129 kB of archives.
After this operation, 82.8 MB disk space will be freed.
You are about to do something potentially harmful.
To continue type in the phrase 'Yes, do as I say!'
 ?]

wohnzimmer1:~# cat /etc/issue
XBian GNU/Linux 7 \n \l

wohnzimmer1:~# uname -a
Linux wohnzimmer1 3.17.7-ck2+ #3 PREEMPT Sat Jan 17 15:08:48 CET 2015 armv6l GNU/Linux

wohnzimmer1:/etc/apt# cat sources.list
###http://forum.xbian.org/thread-2451-page-11.html
###deb http://mirrordirector.raspbian.org/raspbian/ wheezy main non-free
###deb http://mirrordirector.raspbian.org/raspbian/ wheezy main contrib non-free rpi
###deb http://xbian.brantje.com stable main
#deb http://apt.xbian.org stable main
#deb http://apt.xbian.org staging main
#deb http://mirrordirector.raspbian.org/raspbian/ stable main contrib non-free rpi
#deb http://xbian.brantje.com stable main
#deb-src http://mirrordirector.raspbian.org/raspbian/ wheezy main contrib non-free rpi
deb http://mirrordirector.raspbian.org/raspbian/ wheezy main contrib non-free rpi

wohnzimmer1:/etc/apt/sources.list.d# cat xbian.list
deb mirror://apt.xbian.org/mirror.txt     stable   main    rpi-wheezy
### deb mirror://apt.xbian.org/mirror.txt     staging   main    rpi-wheezy
### deb mirror://apt.xbian.org/mirror.txt     devel   main    rpi-wheezy
wohnzimmer1:/etc/apt/sources.list.d#

CurlyMoo, 2015-02-08:
Why did you use the `-f` in the first place?

florian-asche, 2015-02-08:
try to look for missing packages. but as you can see, problem is also there at dist-upgrade

anaconda, 2015-02-08:
Some of XBian packages are conflicting with Debian (and hence Raspbian) packages.
That's normal and `dist-upgrade` should not be used.

florian-asche, 2015-02-08:
Is there a better solution in future?

CurlyMoo, 2015-02-08:
> try to look for missing packages

What did you do before it complained about it. It doesn't happen normally.

florian-asche, 2015-02-08:
Upgrade from RC3 to latest version.

CurlyMoo, 2015-02-08:
If you install just `xbian-update` the rest should be resolved normally.

mk01, 2015-02-27:
this topic repeats itself over and over, like swiss trains - always on time ;)

- for user:
```
dist-upgrade is suposed to handle distribution ugrades (from one generation to new: for instance wheezy -> jessie). to be able to do this 'automatically', it does not follow standard apt rules and settings. in case of any conflicts it absolutely puts priority to default distro template. in this scenario, it may (also automatically) give preference to system defaults over user's manually installed packages.

in case of wheezy it is trying to install sysvinit system (which of course conflicts with upstart -> that means with xbian).

that particular preference is hardcoded deep into wheezy's repo/distro templates and despite 'public' opinion Debian never changed that.

but this is no more case for jessie. so once jessie hits stable, xbian on RPI goes jessie too, this pain will 
disappear. 

just btw: speaking generally dist-upgrade is great tool for vanilla disto installations. for other cases it should not be tool to rely on because - already said - it can ignore actual system's package state and its dependencies (even for essential packages) - in favor of vanilla upstream. that means it is practically use-full ONLY if run by someone who understands what it does and understands the upgraded system well.
```

CurlyMoo, 2015-02-27:
Users now get this warning:
```
 # apt-get dist-upgrade
 ----------- WARNING -----------
 Running 'apt-get dist-upgrade'
 can break your system badly.
 Are you sure you want to take
 that risk?
 ----------- WARNING -----------
 [y/n]
```

f1vefour, 2015-02-27:
No should be default.

closed 2015-11-30 by mkreisl