xbianonpi/xbian #672: iptables / NAT kernel module appears missing after update

mikehawke1337, 2015-02-10:
Hi,

as reported in http://forum.xbian.org/thread-2729.html I received an error trying to set my masquerading rule with iptables after an apt-get update&upgrade. I cannot tell you which version I had in the very beginning, as I have set up my raspberry from scratch to maybe have a quick fix, but this issue is still there with kernel version 3.17.7-ck2+ #3.

If you require additional information from me let me know, I'd be happy to help.

Thanks

CurlyMoo, 2015-02-10:
Can you create a pull request in our kernel repository with the update needed to our kernel config?

mikehawke1337, 2015-02-10:
I just checked out the pull request section and I have no idea what to do. Sorry, I'm not a coder and a novice to github..
I'm not even entirely sure which exact kernel module would fix the problem, I don't have a kernel config on my machine right now to check on that.

anaconda, 2015-02-10:
It seems it's `CONFIG_IP_NF_NAT`.
```
found in Linux kernels: 3.17–3.19
modules built: ip_nat, iptable_nat
Help text

This enables the `nat' table in iptables. This allows masquerading, port forwarding and
other forms of full Network Address Port Translation.
```

It's not set in our 3.17 config:
https://github.com/xbianonpi/xbian-package-kernel/blob/master/extra-files/rpi-3.17.y/.config#L816

mikehawke1337, 2015-02-10:
Yep, looks about right.

Has that been unset on purpose? I would assume quite a few people are using NAT for VPN forwarding and port forwarding..
Any chance this setting will be enabled again? If yes, in which timeframe?

I don't know of any workarounds other than recompile my own kernel, which I wouldn't want to do on a system like my xbian raspi..

CurlyMoo, 2015-02-10:
So @mikehawke1337, can you do that pull request?

mikehawke1337, 2015-02-10:
I have no idea what I did, but I think I created a pull request...

CurlyMoo, 2015-02-10:
Did you also check the other configs?

mikehawke1337, 2015-02-10:
just checked rpi2-3.18.y aswell and changed it

CurlyMoo, 2015-02-10:
Was that all?

mikehawke1337, 2015-02-10:
I hope so?

CurlyMoo, 2015-02-10:
The new kernels should arrive at 0:05 UTC+1

CurlyMoo, 2015-02-10:
Confirmed working.

closed 2015-02-10 by CurlyMoo

RolfTimmerman, 2015-02-11:
Can you please indicate how to update to the new kernel?

CurlyMoo, 2015-02-11:
Post the output of `find /lib -name *nat*`

RolfTimmerman, 2015-02-11:
/lib/modules/3.17.7-ck2+/kernel/net/bridge/netfilter/ebt_snat.ko
/lib/modules/3.17.7-ck2+/kernel/net/bridge/netfilter/ebt_dnat.ko
/lib/modules/3.17.7-ck2+/kernel/net/bridge/netfilter/ebtable_nat.ko
/lib/modules/3.17.7-ck2+/kernel/net/ipv6/netfilter/nf_nat_ipv6.ko
/lib/modules/3.17.7-ck2+/kernel/net/sched/act_nat.ko
/lib/modules/3.17.7-ck2+/kernel/net/netfilter/nf_nat.ko
/lib/modules/3.17.7-ck2+/kernel/net/netfilter/nf_nat_amanda.ko
/lib/modules/3.17.7-ck2+/kernel/net/netfilter/nf_nat_tftp.ko
/lib/modules/3.17.7-ck2+/kernel/net/netfilter/nf_nat_ftp.ko
/lib/modules/3.17.7-ck2+/kernel/net/netfilter/nf_nat_proto_udplite.ko
/lib/modules/3.17.7-ck2+/kernel/net/netfilter/nf_nat_proto_sctp.ko
/lib/modules/3.17.7-ck2+/kernel/net/netfilter/nf_nat_sip.ko
/lib/modules/3.17.7-ck2+/kernel/net/netfilter/nf_nat_proto_dccp.ko
/lib/modules/3.17.7-ck2+/kernel/net/netfilter/nf_nat_irc.ko
/lib/modules/3.17.7-ck2+/kernel/net/ipv4/netfilter/nf_nat_pptp.ko
/lib/modules/3.17.7-ck2+/kernel/net/ipv4/netfilter/nf_nat_ipv4.ko
/lib/modules/3.17.7-ck2+/kernel/net/ipv4/netfilter/nf_nat_proto_gre.ko
/lib/modules/3.17.7-ck2+/kernel/net/ipv4/netfilter/nf_nat_snmp_basic.ko
/lib/modules/3.17.7-ck2+/kernel/net/ipv4/netfilter/nf_nat_h323.ko
/lib/modules/3.17.7-ck2+/kernel/drivers/net/phy/national.ko

CurlyMoo, 2015-02-11:
Try `modprobe nf_nat` first.

RolfTimmerman, 2015-02-11:
xbian@xbian /boot $ sudo modprobe nf_nat
xbian@xbian /boot $ sudo find /lib -name *nat*
/lib/modules/3.17.7-ck2+/kernel/net/bridge/netfilter/ebt_snat.ko
/lib/modules/3.17.7-ck2+/kernel/net/bridge/netfilter/ebt_dnat.ko
/lib/modules/3.17.7-ck2+/kernel/net/bridge/netfilter/ebtable_nat.ko
/lib/modules/3.17.7-ck2+/kernel/net/ipv6/netfilter/nf_nat_ipv6.ko
/lib/modules/3.17.7-ck2+/kernel/net/sched/act_nat.ko
/lib/modules/3.17.7-ck2+/kernel/net/netfilter/nf_nat.ko
/lib/modules/3.17.7-ck2+/kernel/net/netfilter/nf_nat_amanda.ko
/lib/modules/3.17.7-ck2+/kernel/net/netfilter/nf_nat_tftp.ko
/lib/modules/3.17.7-ck2+/kernel/net/netfilter/nf_nat_ftp.ko
/lib/modules/3.17.7-ck2+/kernel/net/netfilter/nf_nat_proto_udplite.ko
/lib/modules/3.17.7-ck2+/kernel/net/netfilter/nf_nat_proto_sctp.ko
/lib/modules/3.17.7-ck2+/kernel/net/netfilter/nf_nat_sip.ko
/lib/modules/3.17.7-ck2+/kernel/net/netfilter/nf_nat_proto_dccp.ko
/lib/modules/3.17.7-ck2+/kernel/net/netfilter/nf_nat_irc.ko
/lib/modules/3.17.7-ck2+/kernel/net/ipv4/netfilter/nf_nat_pptp.ko
/lib/modules/3.17.7-ck2+/kernel/net/ipv4/netfilter/nf_nat_ipv4.ko
/lib/modules/3.17.7-ck2+/kernel/net/ipv4/netfilter/nf_nat_proto_gre.ko
/lib/modules/3.17.7-ck2+/kernel/net/ipv4/netfilter/nf_nat_snmp_basic.ko
/lib/modules/3.17.7-ck2+/kernel/net/ipv4/netfilter/nf_nat_h323.ko
/lib/modules/3.17.7-ck2+/kernel/drivers/net/phy/national.ko

CurlyMoo, 2015-02-11:
I meant before you tried the iptables rule.

RolfTimmerman, 2015-02-11:
still the same issue

CurlyMoo, 2015-02-11:
If you know what kernel module you're missing you can do a pull request on our kernel configs and it will appear automatically in our repositories.

RolfTimmerman, 2015-02-11:
Pull request 22 should have fixed the issues from what I can understand. When performing `lsmod` I don't see `iptable_nat` as mentioned on some websites. Could that be the issue?

RolfTimmerman, 2015-02-11:
Is there a way to really establish that I'm running the latest kernel?