xbianonpi/xbian #673: Can't initialize iptables table `nat'

RolfTimmerman, 2015-02-11:
Currently I'm trying to set up OpenVPN however when executing:
iptables -t nat -A POSTROUTING -s 10.8.0.0/24 -o eth0 -j SNAT --to-source 192.168.XX.X

I'm getting the following error:
iptables v1.4.14: can't initialize iptables table `nat': Table does not exist (do you need to insmod?)
Perhaps iptables or your kernel needs to be upgraded.

Is this perhaps also related to issue #672?

I've updated my Pi with the latest packages (20150210-4)

CurlyMoo, 2015-02-11:
It is related and as you can see fixed when you install the latest kernels of last night.

closed 2015-02-11 by CurlyMoo

RolfTimmerman, 2015-02-11:
Are the new kernels in the staging repository?

CurlyMoo, 2015-02-11:
First try upgrading everything in the stable repository. If it contains a kernel update and the new kernel contains your module. You're fine.

RolfTimmerman, 2015-02-11:
I've updated/upgraded everything in the stable repository. My kernel is:
Linux xbian 3.17.7-ck2+ #3 PREEMPT Sat Jan 17 15:08:48 CET 2015 armv6l GNU/Linux

However I've still got the same issue. I can join IRC or something if you want.

CurlyMoo, 2015-02-11:
Please continue in the other issue.