xbianonpi/xbian #674: [RPI] Video Resolution changing on video playback

bairdy, 2015-02-12:
on video playback since Kodi 14.1 Git:2015-02-05-38e4046-dirty it changes the resolution before playing a video file and makes the kodi overlay look blurry and knocking the aspect ratio out on the video

on checking the log on loading videos its adjusting the video output
```23:24:45 T:2667574288  NOTICE: Display resolution ADJUST : 640x480 (640x480) @ 72.00 - Full Screen (32) (weight: 0.001)```
but the video has a resolution of 720x404 and on a differant video it adjusts as follows

```23:28:44 T:2909250576  NOTICE: Display resolution ADJUST : 1024x768 (1024x768) @ 59.94 - Full Screen (39) (weight: 0.000)```
and the videos resolution is 1280x536

a full log is available at http://xbmclogs.com/show.php?id=423750

anaconda, 2015-02-12:
This should have been fixed - fix will be in next update. Can you add this to your advancedsettings.xml and report whether I have identified the cause correctly (Kodi needs a restart):
```
<video>
  <adjustdisplayrestomatchvideo>false</adjustdisplayrestomatchvideo>
</video>
```
If a `<video>` sections exists already, add it there.

bairdy, 2015-02-12:
@anaconda  
that does the job and fixes it

closed 2015-02-12 by anaconda