xbianonpi/xbian #675: Mac OSX Installer

ManusSs, 2015-02-12:
I have an issue with the installer for MacOS. I have two disc in my MacBook Pro, 1 SSD with a partition named "Macintosh OS" and 1 HDD (partition "Macintosh HD") for big data storage.

Message error :
[image]

anaconda, 2015-02-12:
@calhaus

CurlyMoo, 2015-02-12:
I think it's already fixed but we need someone to recompile the installer for mac.

calhaus, 2015-02-13:
Hi @ManusSs 
First, i don't have a mac, so it can be hard to help you.
Second, I've got OS X Yosemite running on Vmware (it's so slooow and laggy it's ridiculous, but it works i guess).

With that said, i've tested both installers, Git v1.21 and the sourceforge v1.00 (both with root privileges), and both times the usb drive was formatted to fat32 and both times worked fine.
Can you tell me which version of installer are you using, are you running the app with root privileges and is your flash drive formatted or already with Xbian? Also when are you getting this errors? (I didn't really get it, you info was a bit confussing and little for me :/ )

-------------------------------------------------------------
@CurlyMoo I've managed to install qt5 n OS X, only dynamically and can't get qmake to run, but compiled the installer and works, but it's not static and i don't know yet why it does not askes me for root permissions like the one on sourceforge, will try to do compile qt statically and try to resolve this next...

bairdy, 2015-02-13:
Hi guys 
Ive compiled the installer from git but cant test it properly as its been done on a machine i have stuck colocated on the other side of the world. it seems to run but thats as far as i can test due not being able plug an sd card in but if anyone is able willing to test it and let me know if it works. 

It can be downloaded at https://dl.dropboxusercontent.com/u/76655167/xbian-installer.zip

bairdy, 2015-02-13:
Hi Guys

Ive created the installer with static libraries unfortunately im unable to test any further than loading it as it was built on a machine collocated the other side of the world for me.
you can get it at https://www.dropbox.com/s/es9krw72vlfpmxs/XBian-installer.dmg?dl=0 let me know if it doesn't work and i can investigate and rebuild

CurlyMoo, 2015-02-13:
Can you describe the steps needed?

bairdy, 2015-02-13:
download the file file from https://www.dropbox.com/s/es9krw72vlfpmxs/XBian-installer.dmg?dl=0and mount the Xbian-installer.dmg in OSX and then run the Xbian-installer inside. select the device,  insert a sd card, select verion to download and click install and with a bit of luck all should work

CurlyMoo, 2015-02-13:
Ok, i was unclear :smile: Steps needed to compile like you did?

calhaus, 2015-02-13:
@bairdy First, dude thanks a lot for helping.

Second, i've first tested the first one (not static app you've added here) and it was not working. But this one is a really god start since is static. Three really minor problems only:
-*.dmg does not work, the app download the Xbian inside the folder, since dmg is mounted archive, the image will not be downloaded. It super easy, just extract the *.app and run it and works fine. The best way is probably use *.zip so users extract the app in the first place;
-The icon is is not showing for the app;
-The app is not calling root privileges needed to write on the usb devices;
I've found this: http://stackoverflow.com/questions/18404884/c-show-osx-permission-dialog to add a root prompt but my qt creator is not opening on the vmware with osx now :s

bairdy, 2015-02-13:
1. install xcode from apple app store
2. download qt from http://qt-project.org/
3. install qt and choose the the version of qt you want to build against
4. install git
5. clone xbian-installer-universal
6. load up qt creator and load the xbian-installer.pro from the xbian-installer-universal directory
7. select the version of qt to compile against and click release

[image]
8. click the hammer to compile for local use
9. open terminal and go to the where xbian-installer.app is now located
10. to make deployable and a dmg image file execute macdepolyqt in terminal
 ```/DIR TO QT COMPILED AGAINST/macdedeployqt Xbian-installer.app -dmg ```
11. mount dmg created and in terminal type to check its running all from the package type
```otool -L /Volumes/Xbian-installer/Xbian-installer.app/Contents/MacOS/Xbian-installer```
to get an output like
```
/Volumes/XBian-installer/XBian-installer.app/Contents/MacOS/XBian-installer:
        /usr/lib/libz.1.dylib (compatibility version 1.0.0, current version 1.2.5)
        @executable_path/../Frameworks/QtWidgets.framework/Versions/5/QtWidgets (compatibility version 5.4.0, current version 5.4.0)
        @executable_path/../Frameworks/QtGui.framework/Versions/5/QtGui (compatibility version 5.4.0, current version 5.4.0)
        @executable_path/../Frameworks/QtCore.framework/Versions/5/QtCore (compatibility version 5.4.0, current version 5.4.0)
        /System/Library/Frameworks/DiskArbitration.framework/Versions/A/DiskArbitration (compatibility version 1.0.0, current version 1.0.0)
        /System/Library/Frameworks/IOKit.framework/Versions/A/IOKit (compatibility version 1.0.0, current version 275.0.0)
        @executable_path/../Frameworks/QtXml.framework/Versions/5/QtXml (compatibility version 5.4.0, current version 5.4.0)
        @executable_path/../Frameworks/QtNetwork.framework/Versions/5/QtNetwork (compatibility version 5.4.0, current version 5.4.0)
        /System/Library/Frameworks/OpenGL.framework/Versions/A/OpenGL (compatibility version 1.0.0, current version 1.0.0)
        /System/Library/Frameworks/AGL.framework/Versions/A/AGL (compatibility version 1.0.0, current version 1.0.0)
        /usr/lib/libstdc++.6.dylib (compatibility version 7.0.0, current version 104.1.0)
        /usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1213.0.0)
```
if its running from local library files it returns something like
```/Users/administrator/apps/build-xbian-installer-Desktop_Qt_5_4_0_clang_64bit-Release/XBian-installer.app/Contents/MacOS/XBian-installer:
        /usr/lib/libz.1.dylib (compatibility version 1.0.0, current version 1.2.5)
        /Users/administrator/Qt/5.4/clang_64/lib/QtWidgets.framework/Versions/5/QtWidgets (compatibility version 5.4.0, current version 5.4.0)
        /Users/administrator/Qt/5.4/clang_64/lib/QtGui.framework/Versions/5/QtGui (compatibility version 5.4.0, current version 5.4.0)
        /Users/administrator/Qt/5.4/clang_64/lib/QtCore.framework/Versions/5/QtCore (compatibility version 5.4.0, current version 5.4.0)
        /System/Library/Frameworks/DiskArbitration.framework/Versions/A/DiskArbitration (compatibility version 1.0.0, current version 1.0.0)
        /System/Library/Frameworks/IOKit.framework/Versions/A/IOKit (compatibility version 1.0.0, current version 275.0.0)
        /Users/administrator/Qt/5.4/clang_64/lib/QtXml.framework/Versions/5/QtXml (compatibility version 5.4.0, current version 5.4.0)
        /Users/administrator/Qt/5.4/clang_64/lib/QtNetwork.framework/Versions/5/QtNetwork (compatibility version 5.4.0, current version 5.4.0)
        /System/Library/Frameworks/OpenGL.framework/Versions/A/OpenGL (compatibility version 1.0.0, current version 1.0.0)
        /System/Library/Frameworks/AGL.framework/Versions/A/AGL (compatibility version 1.0.0, current version 1.0.0)
        /usr/lib/libstdc++.6.dylib (compatibility version 7.0.0, current version 104.1.0)
        /usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1213.0.0)```

bairdy, 2015-02-13:
@calhaus 
Ive done another build with the added security to test and download from https://adchimp.co.uk/xbian-installer.zip

i havent put it into a dmg file as for the icon i will take a look into later on tonight

ManusSs, 2015-02-13:
@calhaus 
This error message show at the launch of the app.
I think the problem is in the code of the installer.

My system volume is named "Macintosh OS" and is mounted at "/Volumes/Macintosh OS". But the app installer want a volume named "Macintosh" and mounted in "/Volumes/Macintosh".

So with the actual app, we can't rename the principal partition of MacOS System.

calhaus, 2015-02-13:
@bairdy the icon part is fixed. and thanks for the tip with "macdedeployqt" ;)
Checking the 1.00 version, the app has an exetra executable asking for root privileges. I'll see what i can do with it.

@ManusSs 
If bairdy version works it's cool. It'll not work inside the dmg because it will not download the Xbian image. But if you extract it and run it as root it will probably ;) Tell me if it will, just to be sure

calhaus, 2015-02-13:
@bairdy the icon part is fixed on git. and thanks for the tip with "macdedeployqt" ;)

For the root part, i've seen the app v1.00 from sourceforge, and this one has an extra executable/script.
I've added this to the app and it works.
Added the executable applet and folder: XBian-installer.app/Contents/Resources/Scripts/* and changed the info.plist to execute first the applet.

@ManusSs 
I've found/reproduced the bug! :)
It's related to what i'm talking above, about the script that asks for root privileges.
I added that to the new version and if i execute it from a removable drive i get your error also.
Everything works fine if i run it on a local folder.

You need root to write on a flash drive, so i'll look if i can fix the executable, for now:
-Extract the app from @bairdy *.dmg; and
-Run the app with root privileges from a terminal: "sudo open /path/to/the/installer/*.app"

calhaus, 2015-02-14:
I've updated the OSX installer: http://sourceforge.net/projects/xbian/files/installers/
The bug is still there for now, but i've added a readme so people know they can't execute the file from a removable drive.

bairdy, 2015-02-14:
@calhaus 

sorry never got back in touch been cloning my server so to run on vmware to be able to test locally and spotted the fix for the icon :)

the new installer on sourceforge wont load on the real machine or in vm for me just get this message on both

[image]

 which i think is down to the script which is the best way to raise the privilages with osx in the installer as it  will be a bitch after doing some research as its not native osx code and a major lack of info integrating osx api's into qt5 if any

CurlyMoo, 2015-02-14:
@bairdy, you can ofc implement ideas directly in the installer so we have a OSX developer on their as well :+1:

calhaus, 2015-02-18:
@bairdy The app lost executable permissions. :s 
I've compress it on linux dunno if that was the problem.

I've runned chmod again and compressed it inside OS X. Can you tell me if it works right away now? Or you still have that problem?
Link: http://www12.zippyshare.com/v/Q1NhdMM5/file.html

bairdy, 2015-02-18:
Yeah will be able to have a look once once I get home from work later on. Linux probably did do something funky with the permissions if it doesn't work I'll repackage it with the correct permissions 😄

bairdy, 2015-02-18:
@calhaus 

Checked the install and it runs fine :)

calhaus, 2015-02-20:
@bairdy , only saw this now. Ty for the feedback, if it's executable for you also, I'll upload this version :)

bairdy, 2015-02-20:
Yeah that version is working and fine to upload 
I'm getting a Mac dev account next week so will be able to sign the installer to avoid any issues with signing and not having to disable the app security in OS X or have to right click an open to bypass

mk01, 2015-04-28:
@bairdy 

can we implement option to choose a locally stored img ?
it would serve as on-click restore of XBian's backup files (images, or clones to files) (((all the terms is the same. bootable image containing the actual system at the time of backup)))

bairdy, 2015-04-29:
@mk01, yeah should be able to add a restore option, but it wont be for a week as im working away with work and dont have access to my desktop.

mk01, 2015-04-29:
time was never a concern. 
just keep it mind and once you will be updating it, try to implement this.
(the img is not gziped, pure data).

Smultie, 2015-05-19:
Any update @bairdy?

Smultie, 2015-06-10:
@bairdy how are you doing? :)

bairdy, 2015-06-10:
sorry i ain't got back but been a little busy working round europe but back at home now, I have it all most done just the GUI to finish off and to test on all platforms so should be around 2-3 weeks once ive tested everything as i don't want to release it broken

Smultie, 2015-06-10:
Sure, no problem @bairdy 
Good to know it will come eventually !

3vi13d, 2016-01-02:
Hey Folks,

Any progress on this as I'm getting this error from OSX version of the installer from Sourceforge 
(about to install El Capitan though - so that might change it!)

/bin/sh: -c: line 0: syntax error near unexpected token `3'
/bin/sh: -c: line 0: `/Users/Ed/Downloads/XBian-installer-OSX (3)/XBian-installer.app/Contents/MacOS/XBian-installer' (2)

mkreisl, 2016-03-25:
Is anybody here who is able to build a new version of OSX installer? I've modified installer sources (Rasberry Pi 3 support) and now there is need to build new installer for OSX

bairdy, 2016-03-25:
Is it on the source forge rss feeds under rpi3 or similar? As it'll be a easy add for the source which I can do then make the installer for all platforms

mkreisl, 2016-03-25:
https://github.com/xbianonpi/xbian-installer-universal
Installer for Win and Linux are already built an uploaded to SF

bairdy, 2016-03-25:
Yeah I will build it tonight

mkreisl, 2016-03-25:
Great! Thanks

bairdy, 2016-03-26:
sorry about the delay my i hadn't tried building the app with the latest xcode and os x which seems to break macdeployqt which bundles the frameworks into the app bundle and qt are a bit rubbish about osx support, it's going to be a day or two as im going to have to sort a mavericks machine out from a time machine backup and build from there

bairdy, 2016-03-28:
i have it now built for OSX its available at https://www.dropbox.com/s/fvop09ynxbshzg8/XBian-Installer-OSX.zip?dl=0

mkreisl, 2016-03-29:
Thank you. Uploaded to Sourceforge :smile:

3vi13d, 2016-03-30:
I get this error...

/bin/sh: -c: line 0: syntax error near unexpected token `2'
/bin/sh: -c: line 0: `/Users/edco/Downloads/XBian-Installer-OSX
(2)/XBian-installer.app/Contents/MacOS/XBian-installer'

/bin/sh: -c: line 0: syntax error near unexpected token `2'
/bin/sh: -c: line 0: `/Users/edco/Downloads/XBian-Installer-OSX
(2)/XBian-installer.app/Contents/MacOS/XBian-installer' (2)

bairdy, 2016-03-30:
what version of osx is it on?

3vi13d, 2016-03-30:
El Capitan 10.11.4 (15E65)

bairdy, 2016-03-30:
Sorry I cant' seem to replicate in a clean install on a vm or on the mac it's self have you disabled gatekeeper as the package is not signed.