xbianonpi/xbian #676: [Cubox-i] Cannot update: Unmet dependancies error

zilexa, 2015-02-16:
After installing the latest Cubox-i image (filename says 7-Feb, after installing XBian-Config says 20150206) it is not possible to update. This can be reproduced easily by writing this image to SD and after first boot is finished, try to update. Also tried after a couple of reboots and without XBMC running. Same behaviour. Screenshots below:

[image]

[image]

CurlyMoo, 2015-02-16:
Can you post the output of `apt-get upgrade` in the terminal?

zilexa, 2015-02-16:
Via command line, i get a very different warning:
[code]Reading package lists...
Building dependency tree...
Reading state information...
The following packages were automatically installed and are no longer required:
  libepoxy0 libmtdev1 libxshmfence1
Use 'apt-get autoremove' to remove them.
The following NEW packages will be installed:
  libdrm-freedreno1 libdrm-nouveau2 libdrm-radeon1 libelf1 libgbm1 libllvm3.5
  libwayland-client0 libwayland-server0
The following packages have been kept back:
  xbian-package-config-xbmc
The following packages will be upgraded:
  bsdutils firmware-imx grep libblkid1 libcryptsetup4 libgudev-1.0-0
  libldap-2.4-2 libmount1 libpam-systemd libpython-stdlib libsmartcols1
  libsystemd0 libudev1 libuuid1 linux-image-3.14.28-armmp mount ntp python
  python-minimal sudo systemd udev util-linux xbian-package-config-shell
  xbian-package-samba xbian-package-xbmc-6q xbian-package-xbmc-scripts
  xbian-update
The following packages will be DOWNGRADED:
  libdrm2 xserver-common xserver-xorg-core xserver-xorg-input-evdev
  xserver-xorg-input-kbd xserver-xorg-input-mouse xserver-xorg-input-synaptics
  xserver-xorg-video-modesetting
28 upgraded, 8 newly installed, 8 downgraded, 0 to remove and 1 not upgraded.
Need to get 108 MB of archives.
After this operation, 19.6 MB of additional disk space will be used.
Do you want to continue? [Y/n] Get:1 http://apt.xbian.org stable/imx6-jessie linux-image-3.14.28-armmp armhf 3.0.0-1423983312 [18.4 MB]
Get:2 http://http.debian.net/debian/ jessie/main bsdutils armhf 1:2.25.2-5 [92.7 kB]
Get:3 http://http.debian.net/debian/ jessie/main grep armhf 2.20-4.1 [314 kB]
Get:4 http://http.debian.net/debian/ jessie/main mount armhf 2.25.2-5 [165 kB]
Get:5 http://http.debian.net/debian/ jessie/main util-linux armhf 2.25.2-5 [800 kB]
Get:6 http://http.debian.net/debian/ jessie/main libelf1 armhf 0.159-4.2 [159 kB]
Get:7 http://http.debian.net/debian/ jessie/main libdrm-freedreno1 armhf 2.4.58-2 [20.4 kB]
Get:8 http://http.debian.net/debian/ jessie/main libdrm-nouveau2 armhf 2.4.58-2 [20.8 kB]
Get:9 http://http.debian.net/debian/ jessie/main libdrm-radeon1 armhf 2.4.58-2 [28.4 kB]
Get:10 http://http.debian.net/debian/ jessie/main libllvm3.5 armhf 1:3.5-9 [7,011 kB]
Get:11 http://http.debian.net/debian/ jessie/main libwayland-client0 armhf 1.6.0-2 [18.7 kB]
Get:12 http://http.debian.net/debian/ jessie/main libuuid1 armhf 2.25.2-5 [64.3 kB]
Get:13 http://http.debian.net/debian/ jessie/main libblkid1 armhf 2.25.2-5 [146 kB]
Get:14 http://http.debian.net/debian/ jessie/main libudev1 armhf 215-11 [47.6 kB]
Get:15 http://http.debian.net/debian/ jessie/main udev armhf 215-11 [856 kB]
Get:16 http://http.debian.net/debian/ jessie/main libgbm1 armhf 10.3.2-1 [1,273 kB]
Get:17 http://http.debian.net/debian/ jessie/main sudo armhf 1.8.10p3-1+deb8u1 [834 kB]
Get:18 http://http.debian.net/debian/ jessie/main ntp armhf 1:4.2.6.p5+dfsg-5 [343 kB]
Get:19 http://http.debian.net/debian/ jessie/main systemd armhf 215-11 [2,310 kB]
Get:20 http://http.debian.net/debian/ jessie/main libwayland-server0 armhf 1.6.0-2 [22.9 kB]
Get:21 http://http.debian.net/debian/ jessie/main python-minimal armhf 2.7.8-3 [40.0 kB]
Get:22 http://http.debian.net/debian/ jessie/main libpython-stdlib armhf 2.7.8-3 [19.5 kB]
Get:23 http://http.debian.net/debian/ jessie/main python armhf 2.7.8-3 [151 kB]
Get:24 http://http.debian.net/debian/ jessie/main libsystemd0 armhf 215-11 [78.4 kB]
Get:25 http://http.debian.net/debian/ jessie/main libcryptsetup4 armhf 2:1.6.6-5 [96.2 kB]
Get:26 http://http.debian.net/debian/ jessie/main libpam-systemd armhf 215-11 [110 kB]
Get:27 http://http.debian.net/debian/ jessie/main libmount1 armhf 2.25.2-5 [153 kB]
Get:28 http://http.debian.net/debian/ jessie/main libldap-2.4-2 armhf 2.4.40-4 [194 kB]
Get:29 http://http.debian.net/debian/ jessie/main libsmartcols1 armhf 2.25.2-5 [103 kB]
Get:30 http://http.debian.net/debian/ jessie/main libgudev-1.0-0 armhf 215-11 [36.3 kB]
Get:31 http://apt.xbian.org stable/imx6-jessie libdrm2 armhf 2.4.58-2 [30.8 kB]
Get:32 http://apt.xbian.org staging/imx6-jessie xserver-xorg-video-modesetting armhf 0.9.0-1 [17.5 kB]
Get:33 http://apt.xbian.org staging/imx6-jessie xserver-xorg-input-synaptics armhf 1.8.1-1 [205 kB]
Get:34 http://apt.xbian.org staging/imx6-jessie xserver-xorg-input-mouse armhf 1:1.9.1-1 [64.3 kB]
Get:35 http://apt.xbian.org staging/imx6-jessie xserver-xorg-input-kbd armhf 1:1.8.0-1 [38.1 kB]
Get:36 http://apt.xbian.org staging/imx6-jessie xserver-xorg-input-evdev armhf 1:2.6.0-1 [82.9 kB]
Get:37 http://apt.xbian.org staging/imx6-jessie xserver-xorg-core armhf 2:1.14.5-1 [1,070 kB]
Get:38 http://apt.xbian.org stable/imx6-jessie firmware-imx armhf 2.0.0 [36.9 MB]
Get:39 http://apt.xbian.org staging/imx6-jessie xserver-common all 2:1.14.5-1 [1,531 kB]
Get:40 http://apt.xbian.org stable/main xbian-package-xbmc-scripts armhf 1.1.8 [26.3 kB]
Get:41 http://apt.xbian.org stable/main xbian-package-samba armhf 20150210-0 [1,586 B]
Get:42 http://apt.xbian.org stable/imx6-jessie xbian-package-xbmc-6q armhf 13.2-RLS-1424001723 [33.7 MB]
Get:43 http://apt.xbian.org stable/main xbian-package-config-shell armhf 20150209-1 [555 kB]
Get:44 http://apt.xbian.org stable/main xbian-update armhf 20150210-4 [78.5 kB]
Fetched 108 MB in 45s (2,403 kB/s)
E: This installation run will require temporarily removing the essential package firmware-imx:armhf due to a Conflicts/Pre-Depends loop. This is often bad, but if you really want to do it, activate the APT::Force-LoopBreak option.
E: Reverse conflicts early remove for package 'libgbm1:armhf' failed
[/code]

zilexa, 2015-02-20:
Suggested workarounds from the forum did not work for me and lead to the same error I as shown in the first 2 screenshots.

CurlyMoo, 2015-02-20:
Can you post the content of `/etc/apt/sources.list` and `/etc/apt/sources.list.d/xbian.list`?

zilexa, 2015-02-20:
`xbian@CuboxTV ~ $ sudo cat /etc/apt/sources.list
deb http://http.debian.net/debian jessie main non-free`

and 
`xbian@CuboxTV ~ $ cat /etc/apt/sources.list.d/xbian.list
deb mirror://apt.xbian.org/mirror.txt      stable  main imx6-jessie
deb mirror://apt.xbian.org/mirror.txt  staging main imx6-jessie
### deb mirror://apt.xbian.org/mirror.txt  devel   main imx6-jessie`

zilexa, 2015-02-20:
Did some more testing: 
forum.xbian.org/thread-2607-post-27365.html#pid27365

mk01, 2015-02-26:
I have the feeling I fixed this in firmware-imx's last update - if I remember correctly - for NEW img build only (without hassling with apt-get a bit).

That means unfortunately that apt-get's output 
```
activate the APT::Force-LoopBreak option.
```

is right, you will need to use that option:
```
apt-get -o APT::Force-LoopBreak=1 upgrade
```

the older firmware-imx packages had libgbm1 as conflict, the last one (more polite) is depending on it - that means old version firmware-imx needs to be removed first - then libgbm1 installed - then firmware-imx installed (the new version).

and because it is essential package, apt-get refuses to do it automatically. if the force-loop will not help, always will work:
```
dpkg -r --force-all firmware-imx
apt-get download libffi6 libgbm1 firmware-imx
dpkg -i --force-all libffi6*.deb libgbm1*.deb
dpkg -i --force-all firmware-imx*.deb
dpkg --reconfigure -a
```

zilexa, 2015-03-02:
Would you want me to test the new image XBian_2015.02.28_imx6.img.gz to see if it has the same issue? To be sure it is fixed in that one.

CurlyMoo, 2015-03-02:
Yes

zilexa, 2015-03-03:
I tried 3 microSD cards and I redownloaded once (first "latest" then XBian_2015.02.28_imx6.img.gz) but my Cubox-i does not even turn on when I use that image. 
When I restore my backup to one of the SD cards, Cubox boots. 

Please bake a new image.

zilexa, 2015-03-07:
I'll try the image from today in a few hrs.

zilexa, 2015-03-07:
I would say this ticket can be closed as the image from March 7 does work, only 2 packages needed updating (something with shell script and another one) which went ok.

closed 2015-03-07 by zilexa

mk01, 2015-03-07:
@zilexa 

we did a "hack" via xbian-update to allow firmware-imx update automatically, but if you reinstalled we can't test on you anymore :)))

but should help others.