xbianonpi/xbian #680: [Kodi] SuperRepo not updated for Helix

calhaus, 2015-02-18:
Hi,
Just got my rpi2 today and installed the latest image from sourcefourge (XBian_Latest_rpi2.img.gz = XBian_2015.02.14_rpi2.img.gz ).
It's Kodi 14.2-beta1.

This version of Kodi still got the repositories for Gotham and not (Kodi) Helix. 
Same thing goes for the source of SuperRepo. In the file manager, the link for SuperRepo is http://mirrors.superrepo.org/addons/gotham .
Please change this.

Found lots of files with superrepo info and links and don't know what they do. But here are at least some packages and the files with superrepo in it:
https://github.com/xbianonpi/xbian-update/search?utf8=%E2%9C%93&q=superrepo
https://github.com/xbianonpi/xbian-package-xbianhome/search?utf8=%E2%9C%93&q=superrepo
https://github.com/xbianonpi/xbian-package-xbmc/search?utf8=%E2%9C%93&q=superrepo

You can probably change the word gotham to helix and it will work fine. I think.

@anaconda

BartOtten, 2015-05-19:
The right source to set starts with http://srp.nu/

mkreisl, 2016-06-20:
see here https://github.com/xbianonpi/xbian/issues/824

closed 2016-06-20 by mkreisl