xbianonpi/xbian #682: [Cubox-i] Online content streaming is broken

attila123456, 2015-02-20:
As mentioned in the [forum](http://forum.xbian.org/thread-2741-post-27118.html): some XBMC add-ons that play online content are broken in the xbian jessie release for the Cubox. Affected add-ons (so far tested) are Radio, Youtube - however, this may be a libcurl or xbmc issue, not a bug in the add-ons.
From the [logs](http://pastebin.com/bYnXHAgi) it appears as if the HTTP responses of streams were empty (`Curl::Debug - TEXT: Empty reply from server`). The Cubox is connected to the network, and the same streams are accessible by wget from the command line. There was a [known issue](http://sourceforge.net/p/curl/bugs/1426/) with libcurl 7.38 but the cubox build uses 7.37.

attila123456, 2015-02-20:
**Update:** kodi 14.2 in xbian-package-xbmc-nightly-6q works beautifully.

mk01, 2015-02-27:
@attila123456 

yes, we provide forced 7.37 since Debian started pushing 7.38 (few months back) as update and removed 7.37.
(and it still does, only the broken 7.38).
```
root@cubox:/tmp# apt-cache policy libcurl3
libcurl3:
  Installed: 7.37.1-1
  Candidate: 7.37.1-1
  Package pin: 7.37.1-1
  Version table:
     7.38.0-4 1001
        500 http://ftp.de.debian.org/debian/ jessie/main armhf Packages
 *** 7.37.1-1 1001
        500 mirror://apt.xbian.org/mirror.txt/ stable/imx6-jessie armhf Packages
```