xbianonpi/xbian #684: Missing file /etc/init/rpcbind-boot.conf

mkreisl, 2015-02-20:
Automatically starting portmapper and nfs-common during boot requires a rpcbind-boot.conf file.
I need this because i mount my nfs shares via autofs with the nfsvers=3 option to have acl's

CurlyMoo, 2015-02-20:
Can you make this a pull request?

mkreisl, 2015-02-20:
Yes, will do it tomorrow

mk01, 2015-02-26:
that (package default concept) is very slow (and LATE) for our needs. 
rpcbind is in XBIAN started unconditionally - immediately after local drives are mounted.

then only statd/idmapd are started(or not) depending on actual configuration. also autofs config (and maps) are build automatically during boot based on info in /etc/fstab. simplified process looks like this:

- rpcbind is started (portmap)
- autofs is started (fstab nfs3/4 and smb entries are recognised and /etc/auto.master.d/auto.xbian map for autofs is build - also fstab change triggers immediate map rebuild and autofs reload)
- statd/idmapd are checking if nfs3/4 mounts are defined and started accordingly. 

as nfs4 is considered:
```
fstype=nfs with option vers=4 or without option
```
as nfs3 is considered:
```
fstype=nfs with option vers=3 (mandatory)
```

(this is actually the same logic as libnfs is using (and mount.nfs documents)
beside that "auto" logic, all standard config files (autofs and nfs-utils /etc/default/nfs-common) are available to user and processed normally, so for instance statd/idmapd can be forced to load anyhow.

it is also documented on wiki http://wiki.xbian.org/doku.php/mount_nfs 
basically without changes working since Beta1 - more than 1.5years. I remember a bug fix from few months back which fixed not properly processed spaces in mount folder names. 

you haven't told what actually is broken for your install - so hard to help in particular. anyhow it looks you should be able to pinpoint the problem (with the description of nfs/smb mounting workflow above). 
so just do pull request if needed, or ask/tell more.

mkreisl, 2015-03-02:
@mk01 
Thank you for your good answer, now things are more clear for me, however I am not an upstart expert and I can't see how starting of portmap works.

My problem was on my Rpi which runs 24/7, that from time to time mount of nfsv3 shares did not work and I looked into /etc/init and was a little bit confused about the comments.which refers to rpcbind-boot[.conf].
Perhaps is is a good idea to remove such comments.

For me it is solved and I closed that issue

closed 2015-03-02 by mkreisl

mk01, 2015-03-04:
@mkreisl 

some files (.conf in /etc/init) are original upstart jobs delivered with packages and we do not touch them. 
```
upstart mini howto:
```
upstart allows local "overriding" of those files (or configuration values) via XXX.override files. they have the same structure as .conf files and any section / configuration variable inside .override file takes precedence over original .conf.

so in your case - portmap.override - redefined is "start on", "STATEDIR" and "pre-start script". portmap is no longer starting on "start-rpcbind" event, but as soon as mountall job is started (what happens on each boot).

for more informations run
```
man 5 init
man 8 init
man 8 initctl
```

mkreisl, 2015-03-06:
@mk01
Thank you again for your excellent explanation how upstart works in that case. 
I've noticed the portmap.override, but I looked not deep inside this file. Reading man pages is a good idea, however I'm not a fan of this, I like it more learning things with working examples ;-)