xbianonpi/xbian #685: Streams broken in 14.2 build

rikardo1979, 2015-02-21:
most if not all online streams seems to be broken on latest 14.2 build
some more info here http://forum.xbian.org/thread-2760.html

Dikay900, 2015-02-21:
can confirm this, just did some testing
build XBian_2015.02.07_rpi2.img works, after update to latest packages the youtube addon is not working anymore

simoncozens, 2015-02-22:
It's not just streams; everything uses the curl library is failing:

    07:54:08 T:2834297856   ERROR: CCurlFile::Stat - Failed: Failure when receiving data from the peer(56) for https://i.ytimg.com/vi/NvjYjkIlWeA/hqdefault.jpg

curl is working from the command line. Looking at the debug logs show that curl is, for some reason, trying to use HTTP proxy: (I have no proxy configured)

````
08:19:58 T:2867852288   DEBUG: GetImageHash - unable to stat url https://yt3.ggpht.com/-B_AgxgoWCEg/VKV0RkOyXKI/AAAAAAAAABg/rBfj6_uCze0/w1280-fcrop64=1,00000000ffffffff-nd/baby%2Bgames%2Band%2Bsongs%2Bcopy.png
08:19:58 T:2782204928   DEBUG: Curl::Debug - TEXT: connected
08:19:58 T:2782204928   DEBUG: Curl::Debug - TEXT: Connected to (nil) (74.125.235.199) port 443 (#0)
08:19:58 T:2782204928   DEBUG: Curl::Debug - TEXT: Establish HTTP proxy tunnel to i.ytimg.com:443
08:19:58 T:2782204928   DEBUG: Curl::Debug - HEADER_OUT: CONNECT i.ytimg.com:443 HTTP/1.1
08:19:58 T:2782204928   DEBUG: Curl::Debug - HEADER_OUT: Host: i.ytimg.com:443
08:19:58 T:2782204928   DEBUG: Curl::Debug - HEADER_OUT: User-Agent: Kodi/14.2-BETA2 (X11; Linux armv6l) XBian/1.0 HW_RaspberryPi/1.0 App_Bitness/32 Version/14.2-BETA2-Git:2015-02-18-04acf2f-dirty
08:19:58 T:2782204928   DEBUG: Curl::Debug - HEADER_OUT: Proxy-Connection: Keep-Alive
08:19:58 T:2782204928   DEBUG: Curl::Debug - HEADER_OUT: Accept-Charset: UTF-8,*;q=0.8
08:19:58 T:2782204928   DEBUG: Curl::Debug - TEXT: Easy mode waiting response from proxy CONNECT
08:19:58 T:2782204928   DEBUG: Curl::Debug - TEXT: Proxy CONNECT aborted
08:19:58 T:2782204928   DEBUG: Curl::Debug - TEXT: Closing connection #0
````

RolfTimmerman, 2015-02-23:
If I recall correctly during one of the latest updates libproxy was updated/installed. Maybe thuis info helps somehow.

simoncozens, 2015-02-23:
Yeah, this patch looks odd: https://github.com/xbianonpi/xbian-package-xbmc/blob/master/patches/libproxy.patch

If nothing else, doesn't the `free` call mean that `m_proxy` always ends up pointing to a freed pointer?

mk01, 2015-02-26:
the patch is a "template" providing xbmc with auto proxy configuration via libproxy library.
it currently misses only condition to decide when to replace XBMC's own proxy config (for instance XBMC's proxy would be set to "auto:libproxy")

@simoncozens 
m_proxy variable is std::string so "=" is actually copying string (from proxy[i] address).

(but until that's agreed and finished - renaming to .nouse is the desired solution)

Dikay900, 2015-03-05:
is this issue fixed in latest update?

f1vefour, 2015-03-05:
It's still broken.

mk01, 2015-03-05:
@f1vefour 

so it wasn't the patch, ... ?
the streams(any) do work OK on imx6 (13.2, 14.0, 14.1. i'm not 100% sure I tested streams with 14.2B2 particularly, so can't say). but the others very OK - is it really not relevant to package updates ? (as code base beside the EGL implementations (RPI/Imx6) is 99% the same.

f1vefour, 2015-03-05:
Only youtube is broken I think, Genesis streaming works, there is an issue with plugin downloading also.

f1vefour, 2015-03-05:
We haven't had a Kodi update that I recall.

mk01, 2015-03-05:
@f1vefour 

I mean base system updates, no xbmc.

f1vefour, 2015-03-05:
How can I test?

mk01, 2015-03-05:
check /var/log/apt/history.log* what was installed at what date/time. 
we now speak about wheezy, yes? not the upgrade to jessie you have done ?

f1vefour, 2015-03-05:
I'm sorry Matus, I forgot I updated to nightly xbmc because 14.2 wont keep calibration settings. I just downgraded back to 14.2 and YouTube streaming is still broken, in 14.2 and 15.1 alpha. All other streaming is working in 14.2 and 15.1, calibration settings still not being applied in 14.2.

f1vefour, 2015-03-05:
Jessie/Wheezy should not be relevant, the above behavior is the exact same on standard wheezy also. This is all on Pi 2, I don't have another SD card to test on Pi 1.

mk01, 2015-03-05:
@f1vefour 

it is relevant because it seems we are back at my original assumption with broken libcurl & stuff. jessie by default includes BROKEN libcurl and that's why for imx6 we provide older versions ... I can copy the debs  to rpi apt folder (devel) and you can try your jessie (with any xbmc) agains them.

(why I listed the xbmc versions was to explain again that THE SAME code with 100% ok *curl* works without issues - and that's why also I was asking about updates to wheezy (as jessie I know for sure has libcurl broken)).

f1vefour, 2015-03-05:
It doesn't work %100 though, there are forum posts from other people saying 
youtube streaming is broken. Everything else seems to be working. It is an 
issue only on XBian, it isn't the youtube plugin.

I will test anything or do whatever, I personally can't even use xbian 
right now because of this problem. I use youtube daily.

mk01, 2015-03-05:
@f1vefour 

do this:
```
wget http://apt.xbian.org/pool/stable/imx6-jessie/l/libcurl3/libcurl3_7.37.1-1_armhf.deb
wget http://apt.xbian.org/pool/stable/imx6-jessie/l/libcurl3-gnutls/libcurl3-gnutls_7.37.1-1_armhf.deb
wget http://apt.xbian.org/pool/stable/imx6-jessie/c/curl/curl_7.37.1-1_armhf.deb
dpkg -i *.deb
pkill kodi
```

retest

mk01, 2015-03-05:
https://github.com/xbianonpi/xbian/issues/694

f1vefour, 2015-03-05:
Okay Matus here we go, this is after downgrading the packages you linked;

*This log is kodi 14.2 beta, tried to stream youtube...didn't work including missing thumbnails, moved over to hdnet trailer plugin and streamed a trailer which worked, moved over to Genesis plugin and streamed a show which worked: http://paste.debian.net/159800/

*This log is the same with kodi 15 alpha 1: http://paste.debian.net/159799/

f1vefour, 2015-03-05:
>xbian@xbian ~ $ apt-cache policy libcurl3
libcurl3:
  Installed: 7.37.1-1
  Candidate: 7.38.0-4
  Package pin: (not found)
  Version table:
     7.38.0-4 1001
        500 http://mirrordirector.raspbian.org/raspbian/ jessie/main armhf Packages
 *** 7.37.1-1 1001
        100 /var/lib/dpkg/status
     7.26.0-1+wheezy12 1001
        500 http://mirrordirector.raspbian.org/raspbian/ wheezy/main armhf Packages

RolfTimmerman, 2015-03-05:
I have an RPi1 with Wheezy and I'm having the issue as well. Anything I can do tot test?

mk01, 2015-03-05:
@f1vefour 

did you reinstall -gnutls variant too ? (and curl3 as well?)

f1vefour, 2015-03-05:
Sure did:

```
xbian@xbian ~ $ apt-cache policy libcurl3-gnutls
libcurl3-gnutls:
  Installed: 7.37.1-1
  Candidate: 7.38.0-4
  Package pin: (not found)
  Version table:
     7.38.0-4 1001
        500 http://mirrordirector.raspbian.org/raspbian/ jessie/main armhf Packages
 *** 7.37.1-1 1001
        100 /var/lib/dpkg/status
     7.26.0-1+wheezy12 1001
        500 http://mirrordirector.raspbian.org/raspbian/ wheezy/main armhf Packages
xbian@xbian ~ $ apt-cache policy curl
curl:
  Installed: 7.37.1-1
  Candidate: 7.38.0-4
  Package pin: (not found)
  Version table:
     7.38.0-4 1001
        500 http://mirrordirector.raspbian.org/raspbian/ jessie/main armhf Packages
 *** 7.37.1-1 1001
        100 /var/lib/dpkg/status
     7.26.0-1+wheezy12 1001
        500 http://mirrordirector.raspbian.org/raspbian/ wheezy/main armhf Packages
```

mk01, 2015-03-05:
@f1vefour 

ok, so wrap-up ? 
btw do you also have ca-certificates ?

f1vefour, 2015-03-05:
I do have ca-certificates. What do you mean wrap-up? Still broken even with older curl packages.

mk01, 2015-03-05:
@f1vefour 

but youtube only, yes ? or? that I meant with wrap-up ;)
can you post "ldd /usr/local/lib/kodi/kodi.bin" ?

f1vefour, 2015-03-05:
Some links fail on other plugins but I can't tell if it is curl related, so yes I would say it is youtube only.

KODI ldd: http://pastebin.com/8nMwaqWT

mk01, 2015-03-06:
can you create armhfjessie-rpi buildroot config file ? (probably it will 1:1 to the one for imx6, only remove IMX stuff and keep xbian-package-firmware (RPI fw & headers)

I will create it on build server and will rebuild xbmc for RPI against native Jessie (no other idea currently)

mk01, 2015-03-06:
just retested youtube and all ok, so lets do the native recompile against jessie with older curl. if imx6, there is no reason rpi not to.

bromix, 2015-03-06:
It's not YouTube only....MixCloud is also not working:
http://forum.kodi.tv/showthread.php?tid=116386&pid=1947819#pid1947819

In some log I noticed a message stating that the support for HTTPs was missing and that the libcurl should be compiled with it.
Don't know if it is related to this, but I haven't any problems with Ubuntu and Android 4.2.2

bromix, 2015-03-06:
And the same here with SoundCloud...no https support:
https://github.com/bromix/plugin.audio.soundcloud/issues/21

f1vefour, 2015-03-07:
Fixed youtube on latest 14.2 update, unfortunately the 15.0 alpha 2 update totally broke kodi. So 14.2 is okay, 15.0 alpha 2 broken. 

Great work @mk01

mk01, 2015-03-07:
@f1vefour 

any xbmc package generated from master (so xbmc-nightly in that case) is provided as is (you know that, just for others). 
there were some commits yesterday (upstream) to it which didn't build at all at first, new dependencies/libs not provided by Debian etc.

but tell me more, it didn't start at all or ... ?

f1vefour, 2015-03-07:
Yes it doesn't start, first it complained about a missing settings.xml which I replaced but then I got this.

http://paste.debian.net/160073/

mk01, 2015-03-07:
@f1vefour 

I triggered reset & rebuild on rpi-nightly, let me know later...

f1vefour, 2015-03-07:
I will Matus, thanks.

mk01, 2015-03-07:
@f1vefour 

there are two build, latest 15.0~ALPHA2-1425748527 and one before (you will find it via apt-cache policy ...).
the last one is build with extra texturepacker - as texture packer is no more linked into xbmc. I'm not sure what all impacts that might have on skins... 

meaning: last one with extra texture packer should be fine. but I appreciate also testing the other package - that are clear on impacts of removed texturepacker.

f1vefour, 2015-03-07:
Will test both now, I'm using a very complex skin so if there are issues I 
should see them immediately.

f1vefour, 2015-03-07:
Not in the repo? 

```
xbian@xbian ~ $ apt-cache policy xbian-package-xbmc-nightly
xbian-package-xbmc-nightly:
  Installed: (none)
  Candidate: 15.0~ALPHA2-1425682000
  Version table:
     15.0~ALPHA2-1425682000 0
        500 mirror://apt.xbian.org/mirror.txt/ stable/rpi-wheezy armhf Packages
        100 /var/lib/dpkg/status
     15.0~ALPHA1-1424841336 0
        500 mirror://apt.xbian.org/mirror.txt/ devel/rpi-wheezy armhf Packages
     15.0~ALPHA1-1424584643 0
        500 mirror://apt.xbian.org/mirror.txt/ devel/rpi-wheezy armhf Packages
     15.0~ALPHA1-1424275123 0
        500 mirror://apt.xbian.org/mirror.txt/ devel/rpi-wheezy armhf Packages
     14.1~RC1-1419732896 0
        500 mirror://apt.xbian.org/mirror.txt/ stable/rpi-wheezy armhf Packages
     14.0-ALPHA1-1404480020 0
        500 mirror://apt.xbian.org/mirror.txt/ devel/rpi-wheezy armhf Packages
     14.0~RC3-1419061099 0
        500 mirror://apt.xbian.org/mirror.txt/ stable/rpi-wheezy armhf Packages
     14.0~RC3-1418728114 0
        500 mirror://apt.xbian.org/mirror.txt/ stable/rpi-wheezy armhf Packages
     14.0~RC3-1418634627 0
        500 mirror://apt.xbian.org/mirror.txt/ stable/rpi-wheezy armhf Packages
     14.0~RC2-1417676350 0
        500 mirror://apt.xbian.org/mirror.txt/ stable/rpi-wheezy armhf Packages
     14.0~BETA3-1416715634 0
        500 mirror://apt.xbian.org/mirror.txt/ stable/rpi-wheezy armhf Packages
     14.0~BETA2-1416671511 0
        500 mirror://apt.xbian.org/mirror.txt/ stable/rpi-wheezy armhf Packages
```

mk01, 2015-03-07:
repo will rebuild at 2000CET so in 1h20m (each 4h).

but packages were already pushed.

f1vefour, 2015-03-07:
Will test in an hour and half then.

f1vefour, 2015-03-07:
What should I have in my sources.list @mk01 because the package is not showing up, I see it on the build server but it's not in the repo.

CurlyMoo, 2015-03-07:
From our own website:

**Raspberry Pi: /etc/apt/sources.list**

`deb http://mirrordirector.raspbian.org/raspbian/ wheezy main contrib non-free rpi`

**CuBox-i / Hummingboard: /etc/apt/sources.list**

`deb http://http.debian.net/debian jessie main non-free`

**Raspberry Pi: /etc/apt/sources.list.d/xbian.list**

```
deb mirror://apt.xbian.org/mirror.txt     stable   main    rpi-wheezy
### deb mirror://apt.xbian.org/mirror.txt     staging   main    rpi-wheezy
### deb mirror://apt.xbian.org/mirror.txt     devel   main    rpi-wheezy
```

**CuBox-i / Hummingboard: /etc/apt/sources.list.d/xbian.list**

```
deb mirror://apt.xbian.org/mirror.txt     stable   main    imx6-jessie
deb mirror://apt.xbian.org/mirror.txt     staging   main    imx6-jessie
### deb mirror://apt.xbian.org/mirror.txt     devel   main    imx6-jessie
```

f1vefour, 2015-03-07:
I know that, I'm on jessie ;)

f1vefour, 2015-03-07:
@mk01 I manually pulled the build from the server, installed libgif4, installed 15.0~ALPHA2-1425748527 and it is still broken unfortunately.

http://paste.debian.net/160159/

CurlyMoo, 2015-03-08:
What about adding an additional symlink to sourceforge called `Working`.

So we have:
`XBian_2015.03.07_rpi2.img.gz` which also is `XBian_Latest_rpi2.img.gz`

But we add:
`XBian_2015.01.31_rpi2.img.gz` which can become `XBian_Working_rpi2.img.gz` or `XBian_Stable_rpi2.img.gz`

Then users at least know what the latest image was without any issues.

Dikay900, 2015-03-08:
the latest working image for me was the image XBian_2015.02.07_rpi2.img which should be the same for the rpi.

f1vefour, 2015-03-08:
I think stable would be better terminology.

CurlyMoo, 2015-03-08:
So that's a yes?

f1vefour, 2015-03-08:
I think it's a great idea.

mk01, 2015-03-16:
@f1vefour 

wanted to check the log but no more on the server ???
```
@mk01 I manually pulled the build from the server, installed libgif4, installed 15.0~ALPHA2-1425748527 and it is still broken unfortunately.

http://paste.debian.net/160159/
```

mk01, 2015-03-16:
regardless of "stable" in the name of .img, stable img is build only from packages in stable apt repo (so STABLE main/rpi-wheezy/imx6-jessie). 

if that is not true (for instance probably imx6 builds still checkout staging repository by default (my strong feeling)), or a unreliable package is in stable repository (xbian-package-xbmc-6q  -  default xbmc for imx6), corrective action is needed - it is definitely a mistake, not intention. 

(((I do not count *-nightly xbmc builds, which indeed are in stable repo - but they don't install automatically, and ... being in stable we allow regular users to pick xbmc/xbmc-nighly easily one over the other even from xbian-config->packages->xbmc by click/clack (or apt-get if they prefer) without adding themselves to staging or devel))).

@f1vefour 

to you question regarding repository. I assumed your wheezy(or upgraded jessie) somewhere in posts before - and thought we are clear - until you have upgraded jessie, you can keep rpi-wheezy for now as your system probably still keeps pre-jessie libs needed - if any. and most of the xbian packages are anyhow lib/system release number undependable. 
BUT, we can slowly turn your upgrade into slow prepare to wheezy/jessie switch (once it comes) - also as described earlier. 
```
start with config file for build root - take imx6 as template, remove imx6 specific packages, put back rpi's (kernel and firmware). 
push it, I will run build root creation on build server then.
then we can fork current xbmc/rpi configs into rpi-jessie and let compile against the new build root. once they are building ok (so buildroot and xbmc building confs are ok), we just introduce new codename into APT repo (rpi-jessie) and can start building automatically into it.
.... and once we get there, you then just remove xbian.list from sources, reinstalls xbian-package-repo which will create it again from template but now reading your system as jessie so you get 

....... stable main rpi-jessie

into it automatically.
... and once we get HERE, we  set rpi-jessie img for auto build. 
```

actually all that is 30minutes work (and 2 days waiting and checking all logs :))) ). so please take this small challenge with you taking lead (having me to assist what ever it will need) - that we finish before summer holidays ;)

btw: with 15alhas - again due to nature of it being HEAVY development branch - changing constantly - until some more of us will take such self punishment (by deciding to run it) - best approach would be that you follow build logs on Jenkins directly. even merge upstream/master into rpi-nighly at xbian-package-xbmc yourself (so you would merge upstrem/master, commit into rpi-nightly, and push to xbian-package-xbmc. then Jenkins should auto-build providing even live console output for you). 
I'm again telling  this because - status of it (and work/doesn't work result) can change x-times a day - simply at Alphas (until push to next Tag) no special care is taken for checking / patch alignment between developers etc. with your potential and engagement - waiting for us to give you go to test and asking for logs - is simply waste of time at least. until I log into git you could be with working package for sure ;)

if you don't have access to Jenkins, ask me or Curlymo.

f1vefour, 2015-03-16:
@mk01

I didn't know but apparently pastes expire on the Debian site, it was complaining about some glES stuff from what I remember.

I will try to create the rpi-jessie branch as soon as I can, I may need a bit of hand holding since I am still getting up to speed on the way the git is setup for autobuilds. About the Jenkins stuff, I have access but have no experience with Jenkins so I will have to get with you on IRC for help as I don't want to break anything.

mk01, 2015-03-16:
@f1vefour 

sure, I'm not trying to get rid of some tasks, don't worry. just 99% of the STUFF are 4-line config files - processed then by $xbiangit handlers automatically - and until we start supporting completely new platform or will add completely new package(source tree), anything is JUST more or less adapted EXISTiNG config - so almost no knowledge needed (and nor myself nor curly will push you into edits where one can mess something). and that is generally meant for wide public. don't be asking, just do it ;) we do not speak hear about sophisticated source code edits. 
in your case this means:
```
cloning xbianonpi/xbian
opening folder xbian-build-img/build/
copying imx6/config into new rpi-jessie/config
and editing config_xbian_deb= keyword
```
(because imx6 IS jessie already - little changes are needed - I suppose removing firmware-imx, removing  linux-image-3.14XX or what kernel is there (xbian branch). and putting xbian-package-firmware (covering RPIs) and adding even the kernel you provided config for (rpi2/3.18)) - but that was said before. then you can do the same with repo xbian-package-xbmc, copying rpi-nightly/config to rpi-nightly-jessie/config - with actually just one change - the buildroot name to compile agains. plus .autobuild by adding new line for this new .deb. but you will get the point yourself).

and that is 95% job done !

Dikay900, 2015-04-10:
is there an update here?

mk01, 2015-04-19:
@Dikay900 

for stable branch this was confirmed as OK again, ... 15 alphas are being build, but I have no feedback.