xbianonpi/xbian #686: [Cubox-i] Black screen and freeze

zilexa, 2015-02-22:
On Kodi 14.2beta.
I sometimes get a black screen, it seems the Cubox-i/system is still running (SSH works) and the Kodi process is also still running but my tv acts as if there is no input anymore from HDMI (it says HDMI2 instead of "Recorder", "Recorder" is normal for OpenElec/Geexbox/Xbian).  

I haven't been able to figure out how to reproduce this one exactly but:
- when I play a video and trakt addon is installed, it happens. 
- with trakt uninstalled it does not happen
- when I stop a youtube video, it happens. 

log: I start the Youtube video by sharing it from the Youtube app via Yatse at 22:55 and I stop it at 23:00. Log from the moment I stop it:
````23:00:37 T:1952833536   ERROR: EGL error in BindContext: 3006
23:00:37 T:1952833536   ERROR: CreateWindow: Could not bind to context
23:00:37 T:1952833536   ERROR: CreateNewWindow: Could not create new window
23:00:37 T:1952833536  NOTICE: CDVDPlayer::CloseFile()
23:00:37 T:1952833536  NOTICE: DVDPlayer: waiting for threads to exit
23:00:37 T:1404040112  NOTICE: CDVDPlayer::OnExit()
23:00:37 T:1404040112  NOTICE: Closing stream player 1
23:00:37 T:1404040112  NOTICE: Waiting for audio thread to exit
23:00:37 T:1367335856  NOTICE: thread end: CDVDPlayerAudio::OnExit()
23:00:37 T:1404040112  NOTICE: Closing audio device
23:00:38 T:1429205936 WARNING: CRenderManager::WaitForBuffer - timeout waiting for buffer
23:00:38 T:1404040112 WARNING: Previous line repeats 20 times.
23:00:38 T:1404040112  NOTICE: Deleting audio codec
23:00:38 T:1404040112  NOTICE: Closing stream player 2
23:00:38 T:1404040112  NOTICE: waiting for video thread to exit
23:00:38 T:1429205936  NOTICE: thread end: video_thread
23:00:38 T:1404040112  NOTICE: deleting video codec
23:00:38 T:1952833536  NOTICE: DVDPlayer: finished waiting
23:00:39 T:1952833536  NOTICE: CDVDPlayer::CloseFile()
23:00:39 T:1952833536  NOTICE: DVDPlayer: waiting for threads to exit
23:00:39 T:1952833536  NOTICE: DVDPlayer: finished waiting
23:00:39 T:1952833536  NOTICE: CDVDPlayer::CloseFile()
23:00:39 T:1952833536  NOTICE: DVDPlayer: waiting for threads to exit
23:00:39 T:1952833536  NOTICE: DVDPlayer: finished waiting
23:00:39 T:1404040112  NOTICE: Thread JobWorker start, auto delete: true
23:00:39 T:1884103600  NOTICE: Previous line repeats 1 times.
23:00:39 T:1884103600   ERROR: ActiveAE::FlushEngine - failed to flush
23:00:45 T:1884103600   ERROR: ActiveAE::InitSink - failed to init
23:00:47 T:1873802160   ERROR: CAESinkALSA - snd_pcm_writei(-5) Input/output error - trying to recover
23:00:47 T:1873802160   ERROR: CAESinkALSA::HandleError(snd_pcm_writei(1)) - snd_pcm_writei returned -5 (Input/output error)
23:00:57 T:1873802160   ERROR: CAESinkALSA::HandleError(snd_pcm_writei(2)) - snd_pcm_writei returned -5 (Input/output error)
23:01:07 T:1873802160   ERROR: CAESinkALSA - snd_pcm_writei(-5) Input/output error - trying to recover
23:01:07 T:1873802160   ERROR: CAESinkALSA::HandleError(snd_pcm_writei(1)) - snd_pcm_writei returned -5 (Input/output error)
23:01:17 T:1873802160   ERROR: CAESinkALSA::HandleError(snd_pcm_writei(2)) - snd_pcm_writei returned -5 (Input/output error)
23:01:27 T:1873802160   ERROR: CAESinkALSA - snd_pcm_writei(-5) Input/output error - trying to recover
23:01:27 T:1873802160   ERROR: CAESinkALSA::HandleError(snd_pcm_writei(1)) - snd_pcm_writei returned -5 (Input/output error)
23:01:37 T:1873802160   ERROR: CAESinkALSA::HandleError(snd_pcm_writei(2)) - snd_pcm_writei returned -5 (Input/output error)
23:01:47 T:1873802160   ERROR: CAESinkALSA - snd_pcm_writei(-5) Input/output error - trying to recover
23:01:47 T:1873802160   ERROR: CAESinkALSA::HandleError(snd_pcm_writei(1)) - snd_pcm_writei returned -5 (Input/output error)
23:01:57 T:1873802160   ERROR: CAESinkALSA::HandleError(snd_pcm_writei(2)) - snd_pcm_writei returned -5 (Input/output error)
23:02:07 T:1873802160   ERROR: CAESinkALSA - snd_pcm_writei(-5) Input/output error - trying to recover
23:02:07 T:1873802160   ERROR: CAESinkALSA::HandleError(snd_pcm_writei(1)) - snd_pcm_writei returned -5 (Input/output error)
````
full log: http://pastebin.com/yexuYz1C

zilexa, 2015-02-23:
Reboot via SSH doesnt work, it says system is going down for reboot and I can't connect afterwards. It doesn't boot. I have to physically unplug power :(

zilexa, 2015-02-23:
It happens every time I stop a video stream, perhaps even with local video files but I have not had the chance to test this.

zilexa, 2015-02-23:
new log: I was able to watch 2 streams (without hitting stop) via the XOT/XBMC Online TV addon, watching a Dutch tv show (NPO de mol). When I try to play the third and last episode (8) I get the black screen freeze. xbmc process is still running. 
new log of this one shows lots of alsa errors:
http://pastebin.com/7k4BNK72

```
14:59:37 T:1953038336  NOTICE: DVDPlayer: Opening: http://l24m231f4a1f8c0054eb3248000000.f57f2609ab9c7ab4.adaptive-e10c1a.npostreaming.nl/lmshieldv2/3/avrotros/rest/2015/AT_2026905/AT_2026905.ism/AT_2026905-audio_eng%3D128000-video%3D1000000.m3u8
14:59:37 T:1953038336 WARNING: CDVDMessageQueue(player)::Put MSGQ_NOT_INITIALIZED
14:59:37 T:1478419376  NOTICE: Thread DVDPlayer start, auto delete: false
14:59:37 T:1478419376  NOTICE: Creating InputStream
14:59:37 T:1457517488  NOTICE: Thread FileCache start, auto delete: false
14:59:37 T:1478419376  NOTICE: Creating Demuxer
14:59:37 T:1478419376  NOTICE: Opening stream: 1 source: 256
14:59:37 T:1478419376  NOTICE: Creating video codec with codec id: 28
14:59:37 T:1478419376  NOTICE: i.MX6 VPU is not able to decode AVC profile 578
14:59:37 T:1478419376  NOTICE: CDVDVideoCodecFFmpeg::Open() Using codec: H.264 / AVC / MPEG-4 AVC / MPEG-4 part 10
14:59:37 T:1478419376  NOTICE: Creating video thread
14:59:37 T:1387070384  NOTICE: Thread DVDPlayerVideo start, auto delete: false
14:59:37 T:1478419376  NOTICE: Opening stream: 0 source: 256
14:59:37 T:1387070384  NOTICE: running thread: video_thread
14:59:37 T:1478419376  NOTICE: Finding audio codec for: 86018
14:59:37 T:1478419376  NOTICE: Creating audio thread
14:59:37 T:1378681776  NOTICE: Thread DVDPlayerAudio start, auto delete: false
14:59:37 T:1378681776  NOTICE: running thread: CDVDPlayerAudio::Process()
14:59:37 T:1378681776  NOTICE: Creating audio stream (codec id: 86018, channels: 2, sample rate: 48000, no pass-through)
14:59:37 T:1387070384  NOTICE:  fps: 25.000000, pwidth: 608, pheight: 342, dwidth: 608, dheight: 342
14:59:37 T:1387070384  NOTICE: Display resolution ADJUST : 1280x720 @ 50.00 - Full Screen (1.000) (25) (weight: 0.000)
14:59:38 T:1478419376  NOTICE: Opening stream: 0 source: 1024
14:59:38 T:1953038336   ERROR: EGL error in BindContext: 3006
14:59:38 T:1953038336   ERROR: CreateWindow: Could not bind to context
14:59:38 T:1953038336   ERROR: CreateNewWindow: Could not create new window
14:59:38 T:1953038336  NOTICE: Using GL_TEXTURE_2D
14:59:38 T:1953038336  NOTICE: GL: Selecting Single Pass YUV 2 RGB shader
14:59:38 T:1953038336   ERROR: GL: Error compiling vertex shader
14:59:38 T:1953038336   ERROR: 
14:59:38 T:1953038336   ERROR: GL: Error compiling vertex shader
14:59:38 T:1953038336   ERROR: GL: Error enabling YUV2RGB GLSL shader
14:59:38 T:1953038336  NOTICE: GL: Using software color conversion/RGBA rendering
14:59:38 T:1953038336  NOTICE: GL: NPOT texture support detected
14:59:39 T:1378681776   ERROR: CDVDAudio::AddPacketsRenderer - timeout adding data to renderer
14:59:48 T:1874850736   ERROR: Previous line repeats 6 times.
14:59:48 T:1874850736   ERROR: CAESinkALSA - snd_pcm_writei(-5) Input/output error - trying to recover
14:59:48 T:1874850736   ERROR: CAESinkALSA::HandleError(snd_pcm_writei(1)) - snd_pcm_writei returned -5 (Input/output error)
14:59:48 T:1378681776   ERROR: CDVDAudio::AddPacketsRenderer - timeout adding data to renderer
14:59:58 T:1874850736   ERROR: Previous line repeats 8 times.
14:59:58 T:1874850736   ERROR: CAESinkALSA::HandleError(snd_pcm_writei(2)) - snd_pcm_writei returned -5 (Input/output error)
14:59:59 T:1378681776   ERROR: CDVDAudio::AddPacketsRenderer - timeout adding data to renderer
15:00:08 T:1874850736   ERROR: Previous line repeats 7 times.
```

zilexa, 2015-02-23:
If I only reboot the xbmc thread, I get the same issue when I try to play a local video file. I have to unplug power to be able to play video again.

mk01, 2015-02-26:
I'm seeing the same, ... I can't find pattern currently, but definitely don't have ANY of yours addons. So this won't be it.

I see that on Hum2ex, Cubox was running until yesterday Gotham 13.2 - there never experienced EGL error 3006. Let's see what happens next on it (after installing Helix (basically the same as you have). 

Also, current XBian 13.2 and 14.X have synced (1:1) egl stuff (xbmc code speaking), what would indicate two options:
 - code break outside EGL in xbmc (between the versions) - but there EGL is wrapped and isolated from other parts so EGLContext can't be manipulated by it (at least I have the feeling).
 - kernel or galcore engine / libs.

what kernel you have? what is ```dmesg | grep -i galcore```? what is firmware-imx version?

zilexa, 2015-03-02:
http://pastebin.com/d35D6NUm
Thats my dmesg

Nothing happens when I run grep -i galcore
How do I find the firmware-imx version? It is the one from 20150206 image, no changes.

zilexa, 2015-03-04:
As reported in my other issue, latest IMX image does not boot at all. Tested with multiple SD cards also tried redownloading.

zilexa, 2015-03-04:
PS: today, without making changes, I have this issue now for every movie file I try to play!

Rebooting via SSH does not do anything, only SSH disconnects, tv stays blank with no input signal.

I have to unplug my Cubox-i after every video file I play :(

mk01, 2015-03-04:
@zilexa 

I pushed (apt repo will rebuild on 2400CET) xbian-package-xbmc-6q=14.1-1425503269 and linux-image-armmp=3.19.0+-1425494966 
Install with (you need to have staging repository allowed). 
```
apt-get install xbian-package-xbmc-6q=14.1-1425503269 linux-image-armmp=3.19.0+-1425494966
```

it looks like 14.2B is broken currently. 
Let me know.

zilexa, 2015-03-07:
I first stopped the XBMC process (had to do that anyway because I had a black screen again) then ran the command and did a reboot via SSH. Now the Cubox-i does not boot anymore.

zilexa, 2015-03-07:
It seems all streams work again, also the Radio addon, though they stutter, both video and radio streams, never experienced that before (as if internet is too slow, but its 3.5MB/s which should be plenty for radio). Not sure if that is related though. 

Also tested a few local videos and no black screen :)

zilexa, 2015-03-07:
I spoke too soon! 
Now with 14.1 all video and audio playblack stutters, also local files on a disk connected via eSATA. 
AND the black screen came back during local video playback, after I hit stop.

zilexa, 2015-03-07:
Just tested with the latest image from March 7 (which contains 14.2) and same black screen issue at the first video I try to play (after hitting play button). I am suprised my sources.list contain only 1 line:
````xbian@CuboxTV ~ $  cat /etc/apt/sources.list
deb http://http.debian.net/debian jessie main non-free````

I would expect it to also contain:
````deb http://xbian.brantje.com stable main```` 

Is this normal?

f1vefour, 2015-03-07:
```
cat /etc/apt/sources.list.d/xbian.list
```

zilexa, 2015-03-07:
ooops I did cat /etc/apt/sources.list

Now I do see it contains stable and staging. Thanks.
deb mirror://apt.xbian.org/mirror.txt      stable  main imx6-jessie
deb mirror://apt.xbian.org/mirror.txt  staging main imx6-jessie
### deb mirror://apt.xbian.org/mirror.txt  devel   main imx6-jessie

 I will try downgrade to 14.1 on the March 7 image and test if I still get the black screen.

zilexa, 2015-03-07:
@mk01 after I did the downgrade on the old image, how can I upgrade again to 14.2? with 14.2 I could still watch something without stutter, just had to unplug the device all the time.

f1vefour, 2015-03-07:
Should be

```
sudo apt-get install xbian-package-xbmc
```

mk01, 2015-03-07:
@zilexa 

only solution I have (and that's also answer why IMX6 still ships with Gotham) is not taking the path for Kodi. The past days I'm going from 14.2 down (currently at 14.0 RC1) and each build is unsatisfactory at different things. 

you wasn't responding for a while so I wasn't posting but I pusbed 14.0-1425545033 
(possible to install with)
```
apt-get install xbian-package-xbmc-6q=14.0-1425545033
```

for a while it looked ok but on HB1 I have the error constantly. On HB2ex already seen it too, only one not compromised looks Cubox4p (what is indeed strange to me as all sw is 100% the same).

I will update later

mk01, 2015-03-07:
@zilexa 

if you turn off switching resolutions, you will workaround this problem... (if an option until I find it)

zilexa, 2015-03-08:
Thanks will try tomorrow!

@mk01 perhaps you already know this but I noticed today that when I have black screen, the video actually continues playing! Because I waited 30 min before unplugging and Kodi asked to resume the video at 30min.
So it is really something with the HDMI connection as my tv simply thinks nothing is connected.

mk01, 2015-03-16:
@zilexa 

can you perhaps retest latest build from today, but on latest kernel from branch 3.19.y ?
```
linux-image-armmp=3.19.1+-1426506141
xbian-package-xbmc-6q=14.2~RC1-1426506966
```

zilexa, 2015-03-23:
Sorry, I've been travelling for work, not able to work on the Cubox. Will be available again next Monday to test.

CurlyMoo, 2015-07-06:
@hackedbellini, can we get you a CuBox-i to help debugging this?

hackedbellini, 2015-07-07:
@CurlyMoo of course I can!

CurlyMoo, 2015-07-07:
Do you have one? If one, just buy one and we'll pay you back (paypal).

hackedbellini, 2015-07-08:
@CurlyMoo how nice! No I don't have one, but I'll search and check what is the best place to buy it in Brazil. I'll send you a link (via email) of it before to see if you are ok with it.

CurlyMoo, 2015-07-08:
ok

hackedbellini, 2015-07-09:
@CurlyMoo OMG I could not find a place selling it in Brazil. There are no cubox distributors here and even a search on mercadolivre (our ebay) doesn't have anything.

My boss is going to europe next week, so maybe I can ask him. But he will be there for like a month.

Do you know if I have any other alternative? Is there a any international website where I can buy and get it delivered to me that you would recommend?

mkreisl, 2015-07-09:
@hackedbellini 
Maybe you can get it directly from here https://www.solid-run.com/shop/
But I'm not sure if they will ship it to Brazil. And I don't know how to manage the custom

Btw, what did you do that you get an CuBox from XBian :disappointed: :blush:

hackedbellini, 2015-07-09:
@mkreisl Just emailed them to ask if they ship to Brazil. Lets wait for their reply now...

> Btw, what did you do that you get an CuBox from XBian :disappointed: :blush:

hahaha no idea :)

mkreisl, 2015-07-09:
Look here:
http://www.ebay.com/itm/SolidRun-Streaming-Multimedia-KODI-Device-CuBox-i4-Pro-V2-1-2GHz-Quad-Core-/191445518148?pt=LH_DefaultDomain_0&hash=item2c930ac744
The do shipping world wide

CurlyMoo, 2015-07-09:
Maybe try SolidRun directly.

CurlyMoo, 2015-07-09:
If you are an XBian member just ask. You can get one as well @mkreisl

mkreisl, 2015-07-11:
@CurlyMoo 
Oh great :blush:, that's like Christmas
But, Christmas is in December and my birthday also :innocent:
I'll send you a mail ...

hackedbellini, 2015-07-17:
@CurlyMoo sorry for the delay here. Ok, I'll try the SolidRun then.

Should I get [this one](https://www.solid-run.com/product/cubox-i-4x4/)? It seems that it is the cheapest option there.

zilexa, 2015-07-24:
Go for a CuBox-i2eX or a CuBox-i4Pro. If you are really going to be helping build XBian for Cubox, you might want to go for the i4Pro so that you can make sure the built in Bluetooth will work. MK01 knows more about what needs to be done for that. But first see if we can get XBian to run stable on the solidrun products.