xbianonpi/xbian #687: Xbian/Kodi not keeping NFS connection alive

Smultie, 2015-02-24:
My NAS is set to hibernate after 20 minutes of no activity.
I found that watching a movie from my NAS, through an NFS share, is not keeping the connection alive.
My NAS goes to sleep and Xbian crashes, not even reacting to SSH.

Anyone with a possible solution?

mk01, 2015-02-26:
@Smultie 

and Gotham was running ok on the same setup ?
XBMC _is_ (and always has) closing idle connections, but never caused this troubles. 

do you have the NFS mounted on RPI, or you just have defined remote source pointing to NFS in XBMC ?

Smultie, 2015-02-26:
@mk01 I'm not 100% sure Gotham worked okay, but I can't remember the issue ever happening before.
I must say the issue only occurs after I've paused the movie/tv-show I'm watching, so that might be another pointer in the right direction.

I've just defined the remote sure in Kodi, so no fancy fstab stuff.

mk01, 2015-02-26:
@Smultie 

ok, then what you described is exactly what is expected to happen. it was like this (on RPI) since I remember - even I remember some _nasty_ discussions why RPI is not better for its purpose than ATV1/2. 

what I'm not sure - is that is "RPI" only. but IDLE connections are closed after some period of time by default (no platform dependent). while reading xbmc source, it happens on each 500ms for all of this:
- curl connections
- smb connections
- nfs connections
- myth
- tvheadend
- uPnP
- ftp sessions

in general all that (and no more) is exactly what I know about this. never was looking more into it because no personal interest (I use ldap pushed domain wide automount lists - what means XBMC clients do use LOCAL folders - which are auto mounted on request) and considering XBMC this is the way it works and always worked (what I remember). 

while reading the code further it looks like XBMC should keep some small reads as keepalive for the open files, but that that for sure depends on DVDPlayer handling of pause (too much code to read in one evening). 
could be that it simply is just strange combination of factors (with your NFS server).

I suppose it is worth opening a topic at xbmc forums and simply ask - how it should be (the programmers expectation). then we (you) have a point/fact to start from.

Smultie, 2015-02-26:
Thanks for your extensive reply @mk01.
One last remark, which might be Xbian related ISO kodi-related is the fact that when above situation occurs, I can't access the Pi over SSH anymore. While I somehow can understand the fact that hibernate crashes Kodi, I can't really fathom why it would also bring the complete OS to its knees.

mk01, 2015-02-26:
if I understand it right:

1. you play movie from XBMC source NFS -> which is NAS
2. you hit PAUSE and forgets about RPI
3. after a while, NAS sleeps (hdd down, eth down?)
4. you remembers RPI again and wants to PLAY movie (continue)
5. but RPI is dead too ?

Smultie, 2015-02-27:
Yeah, that's basically it.

1) Watch movie/tv-show
2) Take a shower
3) Come back, resume
4) Pi crashes, won't reply to CEC/SSH/etc. Have to replug the power to get
it to boot again.

mk01, 2015-03-16:
@Smultie 

https://github.com/xbmc/xbmc/issues/6278

and I have seen more, just cant find now again. 
that seems more to be symptomatic than random - although highly possible that limited to a final number of combinations (xbmc vs DEVICE/TYPE/MANUFACTURER/whoknows).