xbianonpi/xbian #689: USB Infrared Toy v2 from dangurous prototypes

kasteleman, 2015-02-26:
USB Infrared Toy v2 from dangurous prototypes was not working on XBian.
CurlyMo provided a patch:

wget http://download.xbian.org/testing/xbian-package-lirc20150227.deb
dpkg -i xbian-package-lirc20150227.deb

I managed to get the irtoy working:

Have added irtoy.conf in /etc/lirc/hardware with the following content:

# /etc/lirc/hardware/irtoy.conf
#
# Arguments which will be used when launching lircd
LIRCD_ARGS="--listen=8765"

#Don't start lircmd even if there seems to be a good config file
#START_LIRCMD=false

#Don't start irexec, even if a good config file seems to exist.
#START_IREXEC=true

#Try to load appropriate kernel modules
#LOAD_MODULES=true

# Run "lircd --driver=help" for a list of supported drivers.
DRIVER="usb_irtoy"
# usually /dev/lirc0 is the correct setting for systems using udev
DEVICE="/dev/ttyMCA0"
#MODULES=""

# Default configuration files for your hardware if any
#LIRCD_CONF=""
#LIRCMD_CONF=""

added --listen=8765 so my domoticasystem can use the irtoy over the network!

Also added a case to the hardware.conf file:

*"Prototypes"*)
. /etc/lirc/hardware/irtoy.conf
;;

The only thing i'm wondering about is the following in that file:

. /etc/lirc/hardware/custom.conf
if [ -z $LIRCD_ARGS ] && [ -z $LOAD_MODULES ] && [ -z $DRIVER ] && [ -z $DEVICE ] && [ -z $MODULES ] && [ -z $LIRCD_CONF ] && [ -z $LIRCMD_CONF ]; then
case "$DMESG" in
*"lirc_rpi"*)

Why do you use the custom.conf and then after "then" the case for "$DMESG"?
I thought you first catch the case for DMESG and if case match use specific conf file.
If no case match, then you use custom.conf. Specific reason for that, or do i misunderstand the syntax?

CurlyMoo, 2015-02-26:
Can you use markup code as described in the top right op this reaction box?

kasteleman, 2015-02-26:
sorry, new to this. What do you mean by that?

CurlyMoo, 2015-02-26:
What i said. You can read about it on the top right of this box `Markdown Supported`. You must agree that the first post is quit unreadble like this :smile:

kasteleman, 2015-02-26:
i Agree...better this way?

CurlyMoo, 2015-02-26:
Have you read the `markdown supported` page?