xbianonpi/xbian #691: System freeze due to NIC driver

s47, 2015-03-01:
Please see: http://forum.xbian.org/thread-2789-post-27655.html#pid27655
I've haven't had any problems until recently updating. Seems to happen only when downloading torrents with transmission.
Dmesg log: http://pastebin.com/ZLbC9tci
Kodi log: http://pastebin.com/ggAhcfxg

Latest build. Kodi 14.2 Beta RPI B model. No overclocking. Sufficient power supply. 8 GB sandisk class 10. Wired ethernet. HDMI, 2 TB EXT4 drive connected.

f1vefour, 2015-03-01:
If you have another power supply please try it, I reproduced the issue but while using a 500mah supply. Otherwise I can't reproduce on my normal 2A supply.

f1vefour, 2015-03-01:
I stressed the Pi hard with 2A supply and no issue.  This is just trying to stream an Internet video with 0.5A supply:

http://pastebin.com/LY2cjhma

s47, 2015-03-02:
Thanks for the reply. Will check with another power supply and measure the voltage when I'm home in a couple of days.
I just don't understand how it could be a voltage issue when it's been running smooth for the past year? Nothing has changed other than regular updates.

f1vefour, 2015-03-02:
Supplies get weaker as the capacitors age, maybe it's just a coincidence or maybe it's something else altogether.

siklosi, 2015-03-02:
I started seeing same messages and I think it's after latest kernel upgrade, my rpi is right now offline and should not be, so i think it's  3.18.7 that is now installed. Funny thing is that sometimes xbmc works and network works (streaming etc.) but Rpi is unreachable (ping, xbmc remote, ssh, apache2) Also radio streams are not working any more for me. I have rpi B 512mb.

s47, 2015-03-02:
Yes, that's the same thing happening to me siklosi.
Is there any way to revert to an earlier kernel and see if that helps?

CurlyMoo, 2015-03-02:
http://askubuntu.com/questions/138284/how-to-downgrade-a-package-via-apt-get

s47, 2015-03-04:
Turned out to be, as @f1vefour suspected, a low voltage problem. For some reason I was only getting 4.5V with my powered USB hub. Thanks again for the help and a great community.

f1vefour, 2015-03-04:
Thanks for coming back and reporting.

s47, 2015-03-05:
A user (can't see the username in the mail) sent me a message suggesting to downgrade the kernel software to 3.17.7-ck2+ with:
sudo apt-get install xbian-package-kernel=3.17.7-ck2+-1421550584

It's working flawlessly now so guess it's a problem with the kernel update.

f1vefour, 2015-03-05:
Well you have to give it some time to be sure.

s47, 2015-03-05:
Will do but it seems to be a lot more stable and responsive already. Will report back in a day or two

s47, 2015-03-07:
48 hours after the kernel downgrade and everything is still running smoothly.

f1vefour, 2015-03-07:
Thanks for reporting, I will try to figure out what happened between the two versions....ugh.

s47, 2015-03-07:
> I'm sorry but this morning I woke up to the same system freeze.
> I've retested the voltage and it's steady at 4,96-5,00 V.
> I've restarted the system and it's back online. But after a while it
> becomes extremely sluggish ad slow. CPU output (according to kodi) is
> consistantly at 100% and CPU-KODI is stuck around 70-80% (with only
> transmission running in the background). Transmssion download speed is
> really inconsistent and drops to 0 kb/sec when downloading popular, legal
> torrents. Also when scanning for new media it sticks at "Scanning movies
> using the movie database"
> Here's an updated kodi log: http://pastebin.com/ht8n7pEa
> And dmesg: http://pastebin.com/hq00gyXN
>
> Any suggestions?

Thank you so much. That fixed the problem. Thanks again.

I have downgraded to 3.17.7-ck2+ and now everything works OK.

hackedbellini, 2015-03-08:
I think I'm having this issue for about a week.

It works fine until (specially after being idle and ~1min after I switch from another hdmi source to xbian) it freezees everything. It freezes in a way that I can't login using ssh so I don't have logs to post here.

I thought my sdcard had some problems, so I migrated to an nfs installation. Still the same problem.

I also tried using ```3.17.7-ck2+-1421550584``` a suggested above, but no luck too.

I really don't now what is causing this... I never had any issue like this before (I've been using xbian for almost 2 years now)

f1vefour, 2015-03-08:
Try adding:
```
echo 08 > /sys/devices/platform/bcm2708_usb/regoffset
echo 0x31 > /sys/devices/platform/bcm2708_usb/regvalue

to

/etc/rc.local
```

f1vefour, 2015-03-08:
I don't either, I dont have any related issue.

hackedbellini, 2015-03-08:
@f1vefour I'll try that... will give some feedback in a couple of days to say if that worked. Thanks!

f1vefour, 2015-03-08:
You're welcome, I hope we can get your issue ironed out.

hackedbellini, 2015-03-08:
The workaround didn't work for me... Just experience the issue twice :(

hackedbellini, 2015-03-08:
Just to be sure, since @s47 's problem seems to be solved with the kernel downgrade, I'll try it again. Maybe I did something wrong on the first try.

Will give some feedback soon.

f1vefour, 2015-03-08:
What is your exact setup, hard drive (powered or not), power supply rating, exact Pi model, wired or wireless...

hackedbellini, 2015-03-08:
* Raspberry B (not B+)
* Power supply: 5.0V 1.0A
* Wired networking

I don't have anything on usb ports (my tv shows/movies are accessed via nfs).

Also, I thought my sdcard was having some problems the first time this issue happened. I tried (like I said above) to do an nfs installation (so, only /boot was being used on the sdcard). I had to go back to the sdcard since the nfs installation got messed up after I did the kernel downgrade =P.

But I got the nfs installation working and the issue happened at least twice. So, it is probably not related to usb/sdcard/etc.

Please tell me if you need some more info.

f1vefour, 2015-03-08:
Odd because my Pi 1 setup is exactly the same and I have no issue.

Do this and post the log to pastebin after you have to pull the power to reboot:
```
dmesg | tail -f /home/xbian/crash.log
```

hackedbellini, 2015-03-09:
@f1vefour dmesg unfortunately has no output, even right before the crash (as you can see [here](http://paste.debian.net/160356/), that is the latest dmesg from before the crash, and it is the same from when xbian booted). Here is the relevant [kodi log](http://paste.debian.net/160357/) for reference.

One things I noticed in all my tests is that every time, the issue happens after a specific "trigger". If I restart xbian and let the TV on its hdmi source, I can use it for as long as I want and I will not have the issue.

Now, I change to another hdmi source for some time (e.g. 2 min) and come back, after I start using it, it will freeze in less than 1 minute.

It freezes the whole system in a way that my opened ssh sessions stop responding and I cannot open another one. The only "communication" I can do with it is send a ping.

And just to be sure that this part is clear =P, this issue started to happen about ~1 week ago. It never happened before and I didn't change anything on it other than run ```apt-get upgrade``` almost every day (yes, I'm that kind of person), so, the only variable that changed on my system when this issue started to happen was the version of the installed packages.

Btw, I'm using stable & staging repositories.

hackedbellini, 2015-03-09:
OMG, for the first time, I let it frozen waiting while I was writing the reply above. It seems it recovered from the freeze.

I don't know how much time it was (probably more than 5 minutes because I had waited for that time for it to come back before). I'll try to time it the next time it happens and give some feedback.

I wonder if this has anything to do with CEC.... I have some historical problems with cec as yoou can see on #495

hackedbellini, 2015-03-09:
Maybe it was a one-time thing... I waited for about 1h in another freeze and it did not come back again.

Well, if I have any news, I'll post it here to you.

f1vefour, 2015-03-09:
@hackedbellini Now save dmesg and post it to pastebin please while the error is still there and you aren't frozen.

hackedbellini, 2015-03-09:
@f1vefour I posted it on the reply above ([here](http://paste.debian.net/160356/)).

Like I said, it didn't print anything before/during/after the freeze (its tail was exactly that before and after it returned from the freeze on the situation I mentioned).

f1vefour, 2015-03-09:
Without anything else to go on I can't help any more unfortunately, I will
leave this open for now in case something changes.

hackedbellini, 2015-03-09:
@f1vefour yes I imagined that... I'll still try to get anything more relevant regarding logs and post here if I find something.

If you have any idea on something that I can try, I'll gladly do.

hackedbellini, 2015-03-09:
Ok, here is the explanation to why we have no logs: It's a kernel panic.

Take a good look at [this picture](http://postimg.org/image/y0lvfoh1n/). It just crashed, and I just noticed that is possible to see the panic faded on the image.

It's hard to see, but it seems to be a "not syncing: Fatal exception in interrupt".

I'll try to get a better picture if it happens again on an easier to ready background. So sad that panics cannot be logged =P.

f1vefour, 2015-03-09:
There is just no way CEC should be able to bring down the kernel. Could you try a different HDMI cable and port or if no cable just a different port on the television, I'm wondering if your TV is doing something it shouldn't do.

Diak, 2015-03-10:
Hi,

I have the same problem (freeze and kernel panic on screen).
I noticed the problem happens when Kodi do a library update scan. It never ends and 100% of the CPU is used until xbian crashes.

Yesterday I tried the workaround :
root@xbian:~# echo 08 > /sys/devices/platform/bcm2708_usb/regoffset
root@xbian:~# echo 0x31 > /sys/devices/platform/bcm2708_usb/regvalue

And I installed the kernel version 3.17.7-ck2+-1421550584 as mentionned

But the problem happened again today...

I have XBian 20150304-2 on a Raspberry Pi B+ and my Kodi Library is in a USB drive

f1vefour, 2015-03-10:
Could you try moving the ~/.kodi directory, sounds like a similar issue some are having with a kodi library api update on various platforms.

```
sudo stop xbmc
mv ~/.kodi ~/kodi.org
sudo start xbmc
```

Reconfigure kodi and see if the issue persists.

Diak, 2015-03-10:
Ok i try it tonight after watching my series because it takes time to reconfigure it.
I'll inform you with the result tomorrow.
Thanks for the help.

f1vefour, 2015-03-10:
Really you could first just try moving the databases, if issue persists move whole .kodi directory.

```
sudo stop xbmc
mv ~/.kodi/userdata/Database ~/
sudo start xbmc
```

Diak, 2015-03-10:
I moved the Database and know i will let it run until tomorrow to see.
If the problem persists i'll move the .kodi directtory as you suggested for 24 more hours.
I cross my fingers ;)

s47, 2015-03-10:
@hackedbellini That's the same thing that happened to me with the kernel panic. Did your screen look like this? http://imgur.com/GjisEpy
Also had the same crash as @Diak when the library scan was stuck. Again, the kernel downgrade fixed the problem for me.

Diak, 2015-03-10:
Yes same error but as i said above the kernel downgrade didn't fix my problem :-/

hackedbellini, 2015-03-11:
hahaha yes, same problem @s47 , yours is just easier to read!

I saw [this post](http://www.arvydas.co.uk/2012/05/raspberry-pi-kernel-panic/) and since my supply (the same one I've been using for about 2 years with the same rpi) was showing 4.74V, I'm trying now with one that shower 5.12V to see if I have a different result.

But since the same issue started to happen at the same time for 3 different people (me, @s47 and @Diak ), probably there's another explanation to it.

f1vefour, 2015-03-11:
4.74v is not enough, like you said you have used the same supply for 2 years...and like I previously said power supplies get weaker as they age. This issue would be more prevalent if it was a common problem, it is a combination of things causing the perfect storm for a few users.

hackedbellini, 2015-03-11:
@f1vefour just tried that... Used another port of the TV with a different HDMI with that other supply (which was producing 5.12V) and the issue just happened again.

f1vefour, 2015-03-11:
@hackedbellini How are you triggering the panic so quickly?

Diak, 2015-03-11:
My RPi crashed this night after moving the Database.
I'll try moving the .kodi directory tonight after work.

hackedbellini, 2015-03-11:
@f1vefour in my case, these steps are almost guaranteed to make it crash:

1) I boot xbian and play with it a little to make sure it is working and will not crash at this point.
2) Then I change the hdmi source to another one for about 2 minutes
3) Then I change the hdmi source back to xbian.
4) I start to navigate through the menus with cec until it stops responding. It usually takes less than 1 minute and will crash 8 of 10 times I try.

That's the reason I can trigger the panis co quickly! =P

Diak, 2015-03-11:
FYI i moved .kodi directory and i reconfigured my kodi instance with add-on and libraries.
See you tomorrow for the update.

f1vefour, 2015-03-11:
@Diak thanks for keeping me updated.

@hackedbellini your issue is apparently directly related to HDMI, could you open a new issue?

Diak, 2015-03-12:
Last news : almost 24 hours since the beginning of the last test and my rpi still runs.

To complete my test, i'm going to delete theses lines from /etc/rc.local :
echo 08 > /sys/devices/platform/bcm2708_usb/regoffset
echo 0x31 > /sys/devices/platform/bcm2708_usb/regvalue

Moreover, i'm going to reinstall the last version of xbian-kernel.

We'll see in 24 hours ^^

hackedbellini, 2015-03-13:
@f1vefour np, I can.

Just to comment, the way I see to quickly reproduce the issue is not the only way it happens. Sometimes it will freeze a few moments after I boot.

Diak, 2015-03-14:
My RPi is still running so it seems the problem is solved with moving the .kodi directory.
If the problem finally come back i will post here but for now thanks a lot for the solution ;)

RolfTimmerman, 2015-03-18:
I've installed a fresh install of XBian on my RPi Model B and unfortunately I have the same issue. At least once a day the whole system freezes and I have to reboot to use it again. Because it happens randomly I haven't been able to save dmesg or anything.

Fabio72, 2015-03-18:
Have you addedd before or after "exit 0"? Should be before
check if you have the right values with
cat  /sys/devices/platform/bcm2708_usb/regoffset
cat /sys/devices/platform/bcm2708_usb/regvalue

RolfTimmerman, 2015-03-18:
Yes I've added it before the "exit 0":
'#!/bin/sh -e
#
# rc.local
#
# This script is executed at the end of each multiuser runlevel.
# Make sure that the script will "exit 0" on success or any other
# value on error.
#
# In order to enable or disable this script just change the execution
# bits.
#
# By default this script does nothing.

echo 08 > /sys/devices/platform/bcm2708_usb/regoffset
echo 0x31 > /sys/devices/platform/bcm2708_usb/regvalue

exit 0'

This also works correctly:
'root@xbian:/home/xbian# cat /sys/devices/platform/bcm2708_usb/regoffset
0x00000008
root@xbian:/home/xbian# cat /sys/devices/platform/bcm2708_usb/regvalue
Reg@0x000008 = 0x00000031'

Diak, 2015-03-18:
And my problem came back. Since Monday every day my RPi freezes....

Diak, 2015-03-19:
And I as imagined the edition of rc.local doesn't change anything. Xbian is still very very unstable...

CurlyMoo, 2015-03-19:
Even if you use the image we labeled stable?

Diak, 2015-03-19:
I switched to the beta image because i had a big problem with the stable image which forced me to format and reinstall all from the beginning. I thought that the beta image couldn't be more unstable and dangerous than the stable one. And in fact i had less problem with this beta image... 

Hopefully i will receive soon my RPi 2 and will be able to test my Xbian on it. If it's still unstable i will switch to another distrib.

Fabio72, 2015-03-21:
Today xbian freezed after many weeks. I was using youtube plugin.
I restarted and within 10 minutes it freezed again, while it was idle.
The only thing I changed was removing of "elevator=noop" from cmdline.txt.
Now I'm tring again with noop elevator and uptime is 1h

[15 more replies]

closed 2015-11-30 by mkreisl