xbianonpi/xbian #692: CEC fails in Jodi

ggggh, 2015-03-01:
During the last week (after one of the recent updates) CEC stops working after a while.  Interestingly, it seems to happen much sooner if overclock is on None (where it just stops working while you are using it).  If overclock is on Xbian standard setting, it seems to fail only after rpi has been idle for some time.

f1vefour, 2015-03-01:
Going to need Kodi and dmesg logs, please post them to pastebin and link here.

CurlyMoo, 2015-03-01:
I though he meant Jidi, but it was Kodi ;)

mkreisl, 2015-03-14:
Same happens here :-(

I' will turn on logs and post them after this issue happens again

mk01, 2015-03-16:
@mkreisl 

don't forget that the usual DEBUG logging is not enough. since KODI, specific subsystem must be set for DEBUG logging too (CEC in that case).

somewhere under settings->system->debug-> .....

mkreisl, 2015-03-16:
@mk01
thanks for the tip, I have now enabled CEC messages too and now waiting for the issue, which happens very seldom: 2x I remember and 1x directly after start, but unfortunately no logs preserved because I currently investigate an other issue (it's too early to talk about this)

ggggh, 2015-03-16:
Sorry I haven't come back with more info, but after a week where this was happening all the time, this is now not happening anymore.  Separate issues (such as inabillity to update kernel without rpi freezing) led me to believe it was a power problem under load.

f1vefour, 2015-03-16:
@ggggh Did you change supplies and the issue went away?

ggggh, 2015-03-17:
No, didn't change it - have a new one on order, 
Rpi seemed mostly ok for a few days...until I met a frozen rpi when I arrived yesterday.
Only other issue is that it still seems to freeze whenever the upgrade script tries to update the kernel package - and that's been happening for about a month.
I'm hoping that the CEC issue was fixed by one of the packages that did upgrade.

closed 2015-08-21 by mkreisl