xbianonpi/xbian #695: Kodi boots before MySQL server

baba420, 2015-03-06:
When you install the MySQL server on XBian, you will notice that the scrapers don't scrape anything as soon you use the advancedsettings.xml. This is because Kodi boots before MySQL server runs. 

On OpenELEC they have implemented a feature where you can toggle "start Kodi after network is ready". 

Alas on XBian there isn't a feature like this implemented inside the GUI. However you can solve this by editing /etc/init/xbmc-preload.conf.

Add before end of script the following:

[code]
.
.
.
.
        /etc/init.d/mysql start || :
        { [ -e /etc/init/xbmc.override ] && initctl emit -n xbmc-norun-event; } || start xbmc
end script
[/code]

Ctrl+X+Y

After a reboot you will notice that your advancedsettings.xml are available and you can start import your library and Kodi start scrape everything as it should be.

mk01, 2015-03-07:
@baba420 

in /etc/default/xbmc is
```
# Put name of all processes which XBMC should wait for before it will be started.
# e.g. WAITFOR=""
WAITFOR=""
```

user can put needed processes here, XBMC will then wait for those processes. 

in case of SQL, the parameter would look:
```
WAITFOR="mysqld"
```

baba420, 2015-03-08:
Thanks a lot! Indeed your way is much easier. 

But do you agree with me there has to be a kind of auto detection from Xbian as soon an user installs mysql server so that it can function properly? Or am I being a total Linux noob and your above solution is just normal routine after installing mysql server.
IMO there should be an option implemented in Xian Config where you can toggle start Kodi after network is ready.

CurlyMoo, 2015-03-08:
We are still looking for someone willing to create that 1000 pages manual about all features XBian currently has :smile: We never thought we would become so differentiated as the other 'two' and would have all these tiny nice but hidden option for common user issues...

mk01, 2015-03-16:
@baba420 

Imagine what - I fully agree with your perspective - considering you are used on how OE implemented this. I don't want to argue there or open discussion what is better/worse - but XBIan was there a 1y or possibly more ago. later XBian integrated internal mechanisms which made that "simple" and user-action needing approach _almost_ useless by changing whole boot process into EVENT based booting - meaning everything happens only for a reason (if needed). so why waiting for network if that is physically not available (for instance), why to process network drives mounting (and waiting for network to come up)  if there is no actual service/application ASKING for them, etc etc etc. not important at this point.

but back to that - YES I honestly agree with you on that. there could be such button again. only at the end it is not about the button at all. because I'm sure you as user doesn't care HOW the particular requirement is internally designed - as long as it is transparent, understandable and most important - VISIBLE to you (made to your knowledge). 

and there is simply valid (and so many times already discussed and blamed) - what @CurlyMoo said. we are terribly failing to provide proper Docs. Personally for a looong time I don't remember functional request, which would not be already provided internally by XBian (by custom toolchains or internal mechanisms/config files - easy to grasps at user level). But all systems are only that good, how they are documented. Otherwise they doesn't matter. 

There is a guy in our team @belese who had in competence (and was doing it great) the GUI config tool (xbian config within XBMC). Unfortunately he cant spend much time with us lately - and apparently there wasn't any self-nomination from user base for that part (yet). And we would appreciate it very much !

-howk

f1vefour, 2015-03-16:
I didn't know this feature existed, now I must wonder what other things I don't know about that improve usability :)

CurlyMoo, 2015-03-16:
xbian boot menu for example ;)

f1vefour, 2015-03-28:
Closing as issue has been resolved.

closed 2015-03-28 by f1vefour