xbianonpi/xbian #700: udev permission denied errors on boot

Fabio72, 2015-03-14:
Distribution: XBian 1.0 (knockout)
Kodi: 14.2-RC1 Git:Unknown
Kernel version: Linux 3.18.8+ armv6l
Firmware: 115f63aa0915cdb5b6dff23822eb16699b72ae8b (4 Feb 2015 21:04:27)
PI version B+

I'm getting two error messages at boot. They are very quick so I had to make a video to see something
udev error opening /sys/devices/platform/mmc-bcm2835.0/mmc_host/mmc0/mmc0:59b4/block/mmcblk0/queue/iosched/slice_idle
and
udev error opening /sys/devices/platform/mmc-bcm2835.0/mmc_host/mmc0/mmc0:59b4/block/mmcblk0/queue/iosched/target_latency
both for permission denied.

/sys/devices/platform/mmc-bcm2835.0/mmc_host/mmc0/mmc0:59b4/block/mmcblk0/queue/iosched is empty after boot
Everything else is working fine, so maybe is only a cosmetic problem.

mk01, 2015-03-16:
@CurlyMoo @f1vefour @anaconda 

guys, can any of you confirm that ? 

@Fabio72 

empty (non existing) queue/iosched is definitely not good ;)

CurlyMoo, 2015-03-16:
I saw them sometimes but can't remember when...

anaconda, 2015-03-16:
10-local-xbian.rules has:
```
ACTION=="add|change", KERNEL=="mmcblk0", ATTR{queue/iosched/slice_idle}="0"
ACTION=="add|change", KERNEL=="mmcblk0", ATTR{queue/iosched/target_latency}="120"
```

`slice_idle` here is 0, but I don't have a `target_latency`:
```
$ sudo cat /sys/devices/platform/mmc-bcm2835.0/mmc_host/mmc0/mmc0\:0007/block/mmcblk0/queue/iosched/target_latency
cat: /sys/devices/platform/mmc-bcm2835.0/mmc_host/mmc0/mmc0:0007/block/mmcblk0/queue/iosched/target_latency: No such file or directory
```

`iosched` is not empty here, but this a Pi 2 (3.18.8+).

Fabio72, 2015-03-17:
https://plus.google.com/118154946688569975412/posts/ertVfXjJxvE

I have too

Fabio72, 2015-03-17:
Maybe I solved!
I found scheduler=noop on my cmdline.txt. I don't kown why, maybe it comes from where I copied the configuration for nfs root.
I must wait to go home to double check but after changing scheduler on the fly queue/iosched is not empty any more

closed 2015-03-17 by Fabio72