xbianonpi/xbian #701: Kernel Module: CONFIG_NFS_FSCACHE

haribert, 2015-03-14:
Dear XBian Team

One small thing, could you please integrate the following options in the kernel:

CONFIG_NFS_FSCACHE=y
CONFIG_AFS_FSCACHE=y
CONFIG_CIFS_FSCACHE=y
CONFIG_FSCACHE_STATS=y
CONFIG_FSCACHE_HISTOGRAM=y

CONFIG_FSCACHE is enabled, but does not work with NFS. It would be great to cache the NFS connection.

Thank you

CurlyMoo, 2015-03-14:
Can you do a pull-request on all our kernel configs?

f1vefour, 2015-03-14:
It's okay, I will add it.

mk01, 2015-03-16:
@f1vefour 

just please leave the parent (CONFIG_FSCACHE=m) as module -- user is right that it itself has no use (without the appropriate support compiled for available filesystems) -- but beside such special installs has little use.

(((as this is not the standard fscache mechanism (which is always present). this one mostly needs installing of additional user space counterparts and configuration - not needed to be present in memo)))

f1vefour, 2015-03-16:
If there is no benefit then perhaps it shouldn't be added, I don't use NFS so if this has negative consequences it should be left out altogether.

mk01, 2015-03-16:
@f1vefour 

ok, I formulated not properly perhaps:

it makes no sense to compile that FSCACHE base driver into kernel directly. as it's use is only for special applications. But I use it myself on one installation. So the user request is to be fulfilled - I would consider that one missing as bug myself (meaning XBian is aware of that driver and was compiling for other kernel branches automatically).
((only saying let's keep it as MODULE, if user installs the user-space toolchain, the module gets loaded automatically and otherwise doesn't waste memory))

haribert, 2015-03-16:
Thanks for your support.

Sorry, if there was a misunderstanding. I would leave the `CONFIG_FSCACHE` as module (which can be installed with `apt-get install cachefilesd`).
At the moment NFS caching does not work, if the cachefilesd module is loaded. If the module is not loaded, there will be no negative consequences for users who do not use NFS.

@mk01 Thanks, this describes it very well.
@f1vefour It would be great, if you can add it to the kernel.

mkreisl, 2015-03-16:
@haribert 
Can you please explain the advantages of such caching. If I understand this right, you need an additionals backend like cachefilesd for caching, which caches nfs share data locally on disk. I played a little bit at Saturday with this on my notebook (runs under openSUSE 13.1) and I can't see any advantages for streaming videos from nfs share using such caching.  And it seems that it does not work if the local cache runs on a btrfs fs (which XBian also has), I had do add an loop device with an ext4 fs to make it work. And, of course, it will stress the sd-card extremely if you put the cache on the local root fs.

haribert, 2015-03-16:
@mkreisl cachefilesd works if `user_xattr` are supported. This is the case for btrfs and as far as I know it is enabled in XBian kernel. So no loop device should be necessary.
The reason I am interested in the NFS caching is my unstable wireless network. I expect that the movie gets cached at the play start and is accessed from the local drive afterwards. So there should be no interruptions if the network is slow or interrupted for a moment.

mkreisl, 2015-03-16:
@haribert 
Yes, I know this, and btrfs should support xattr's. But on my notebook cachefilesd doesn't take care of is. Don't know why. Before you try this caching, try to enable kodi's build in cache, perhaps it helps: http://kodi.wiki/view/HOW-TO:Modify_the_video_cache. Or, even better, try to optimize your wlan ;-)

f1vefour, 2015-03-16:
Okay I will integrate it tonight.

mkreisl, 2015-03-16:
@f1vefour 
What do you think about (re-)adding of following options into the kernel:

CONFIG_ZSMALLOC=y
CONFIG_PGTABLE_MAPPING is not set
CONFIG_ZRAM=m
CONFIG_ZRAM_LZ4_COMPRESS=y
CONFIG_ZRAM_DEBUG is not set

This will provide the modul zram, which is required by /etc/init/zram-swap.conf in some cases.

f1vefour, 2015-03-16:
I hadn't noticed they weren't enabled, sorry about that I will add.

mkreisl, 2015-03-16:
@f1vefour 
Thanks for this. This options have been removed a long time ago. For me it is no problem because I'm compiling my own kernel with additional special options, but for other users who only want's to use the move the zram features without the need of compiling their on kernel it is good to have it again.

f1vefour, 2015-03-28:
@mkreisl No problem.

Added, closing request.

closed 2015-03-28 by f1vefour

embdlover, 2020-09-15:
Hi,
I have disable CONFIG_FSCACHE i want to know what is impact of it on performance of system? will enabling boost improve system performance and by disabling it any degradation will happen with any subsystem ?

How important it is to have CONFIG_FSCACHE  ??

mkreisl, 2020-09-15:
If you had read this thread completely, you could give yourself the answer