xbianonpi/xbian #703: hdmi screen get disconnected after starting movie since XBain 2015.02.14 rpi2

ctugt, 2015-03-18:
Please Read the full post to get a good understanding about what i mean. There is also a video in my post:

http://forum.xbian.org/thread-2874.html

[POST COPY]
Software
XBian version: 1.0 (knockout) 3.18.8
XBMC/Kodi version: 14.2 BETA 2
Overclock settings:
1Ghz profile from xbain

Hardware
Raspberry Pi Model B (2)
Power Adapter: Blue Start 110-350V, 50-60Hz 0.1 A
SD card size and make/type: Kingston 8GB microSDHC, SDC10/8GB, Class10
Network (Ethernet or wireless): RaspBerry default port, connected with a cat 6 cable to my router.
Connected devices (TV, USB, network storage, ...): I got a 60 inch Sharp tv, running over a Cinema the vsx-922 hdmi -> to hdmi in raperri pi 2 (B) with 1080P.
Its not harware related, openElec works fine no recconections... just works...

Log files
Link to logfile(s): Will post when i know which version does work and which doesnt.

Problem description:
Hi,

Im glad to be a new user and owner of a rasperri Pi 2 (B). I just bought it yesterday and have complete set it up like i wish. Soon im trying to get my won ambilight with xbian.

Except for one little problem, as soons as i play or stop a video, it reconnects the hdmi. Is this a sync issue? or is this a xbian related issue. I wanne try the other software too Elec something... But i do think this is a settings somewhere to change the refesh rate of the tv or something idk. Could anywone look in to this for me?

Here a video with some more guided explenation about what is going wrong here.
Thanks!

How to reproduce:
Well install latest beta, leave everything default, close down the wizard window.
Connect it to a 60 inch display from the branch sharp. Boot it.
Then, start xbmc, added a nfs share for movies, play movie and then it reconnecs hdmi, like it looses it connection for a second. Then pause and its ok, Then Close the movie and it will reconnect again.

(Sometime it start on 640 x 800, i do think this is because i shutdown the system wrong, cause whenever i shutdown. Reconnect hdmi + power and boot again i get the 1080p resolution back.
common issue after installing other product or shutdown improperly?, that it resets to 640 x 800? can this question also be answered)

What i tried to solve this
* Defaulting back stable version, this works flawless.
* Connected directly to tv without cinema
* Disable the builtin tv option (swap channels automaticly of something get regonisez on hdmi) (later i realised that the raspberry is connected to cinema, so the tv input wouldnt change) (but after that i realised that i connect it to tv to test without cinema, so users can try this option if their stuck) (lol)
* Note: After every install i did a clean windows safe removal and i shutted down Kodi each time completely. I also disconnected the hdmi and power cable after shutting down. And connecting them after i putted the fresh sd card back in to prevent any issues.

Current Progress:
Narrowing down with version corrupted. I think its a xbian bug.

XBain 2015.03.07 rpi2 fails
XBian 2015.02.28 rpi2 fails
xBian 2015.02.21 rpi2 fails
xBian 2015.02.14 rpi2 fails
xBian 2015.02.07 rpi2, finally narrowed it down, this is the last version when it works normally for me and i didnt got the hdmi signal loss.

Now im going for option 2:
If your post is not related to XBMC/Kodi post the appropriate log file, if you don’t know leave it empty and ask for help.

anaconda, 2015-03-18:
Do you have "Adjust display refresh rate to match video" enabled in Settings->Video->Playback?

ctugt, 2015-03-18:
Nope, but doesnt effect it, i tried

ctugt, 2015-03-18:
added all logs files meanwhile

ctugt, 2015-03-19:
can be closed, in the first beta the adjust display was disabled by default, from that i was thinking it was a bug

closed 2015-03-19 by ctugt