xbianonpi/xbian #704: Only one package updated each time xbian-config

RolfTimmerman, 2015-03-19:
I noticed a small issue when I tried to update my RPi. While there were 4 packages which could be updated, only one (not sure if it was first or last item) was updated/selected when I selected 'update all packages' through xbian-config.

I have installed all latest updates (20150304-2) and I'm using a RPi model B.

mk01, 2015-04-07:
can you provide version of xbian-package-config-shell - 
```
dpkg-query -W xbian-package-config-shell
```

there was recent (2-3weeks ago) update to similar problems.
 I'm testing with
```
xbian-package-config-shell	20150307-0
```

and can not replicate this issue. can you recheck the version of that sub-package ?

RolfTimmerman, 2015-04-07:
``
xbian@xbian ~ $ dpkg-query -W xbian-package-config-shell
xbian-package-config-shell      20150307-0
``
It still occurs because when I select 'update all' then it only selects the last package from the 2 available.

mk01, 2015-04-07:
@RolfTimmerman 

can you post the names of the packages?

and also output of 
```
apt-mark showhold
xbian-apt-show-versions -u
```

RolfTimmerman, 2015-04-07:
```
unrar
xbian-package-kernel
```

apt-mark showhold does nothing, xbian-apt-show-versions is below:
```
xbian@xbian ~ $ sudo xbian-apt-show-versions -u
unrar/stable 5.2.6-1 upgradeable to 1:4.1.4-1
xbian-package-kernel/stable 3.17.7-ck2+-1421550584 upgradeable to 3.18.8+-1425523930
```

mk01, 2015-04-08:
please try 
```
apt-get -s upgrade
```

and tell me if simulation via apt updates both or only kernel.

RolfTimmerman, 2015-04-08:
Both of them
'''
xbian@xbian ~ $ sudo apt-get -s upgrade
[sudo] password for xbian:
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following packages will be upgraded:
  unrar xbian-package-kernel
2 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Inst xbian-package-kernel [3.17.7-ck2+-1421550584] (3.18.8+-1425523930 XBian repository:stable [armhf])
Conf xbian-package-kernel (3.18.8+-1425523930 XBian repository:stable [armhf])
Inst unrar [5.2.6-1] (1:4.1.4-1 XBian repository:stable [armhf])
xbian@xbian ~ $
'''

mk01, 2015-04-19:
@RolfTimmerman 

stupid question - but needed, sorry in advance: beside the tick on 'update all' have you tried to run through full update process ? 

this is not clear from opening post and fact is, when you  tick the 'virtual' first package named "update all available" no other checkbox is ticked. that is expected - but during actual update process, all should be processed.

RolfTimmerman, 2015-04-20:
I have updated my RPi completely. When there are more than one packages available and I select 'update all' only the last package is selected/updated. When I select the packages individually the selected packages are updated.

Smultie, 2015-04-29:
This is indeed still an issue. Selecting the"update all" only updates 1 package.

mk01, 2015-04-29:
@Smultie 

ping me again on IRC and prepare ssh in. I can not reproduce it so need again affected system.

mk01, 2015-04-29:
@RolfTimmerman 

it's fixed and already pushed to repo. after APT update at 1600CET (and installing it) you can retest

RolfTimmerman, 2015-04-29:
@mk01 Tnx! Is there a way to test it if there are no packages to update?

CurlyMoo, 2015-04-29:
Manually downgrading some packages.

RolfTimmerman, 2015-05-02:
Ok I can confirm this is fixed now