xbianonpi/xbian #705: Can't configure CEC anymore

RolfTimmerman, 2015-03-19:
Recently I've installed a fresh copy of XBian (20150304-2) and everytime I start my TV, it automatically switches back to XBian (once or twice) which is quite annoying. Before the update I was able to configure CEC through system settings (under peripherals?) but now there is no such option anymore.

bairdy, 2015-03-19:
@RolfTimmerman 

What version of kodi is running? is it the latest is 14.2-rc1 (compiled March 13th) as ive just checked and as it does have the option to disable still.

 its located still under from the main home screen
 ```System>Settings>System>Input Devices>Peripherals>CEC Adapter```

RolfTimmerman, 2015-03-20:
It is the latest. I know where it should be, but it isn't there anymore only a blank Peripherals list.

RolfTimmerman, 2015-03-29:
Somehow the option is back maybe because ive installed a new kodi version.

closed 2015-03-29 by RolfTimmerman