xbianonpi/xbian #706: Kernel Panic @mk01 from Xbian forums

The-Exnor, 2015-03-24:
As requested by f1vefour here is as picture of Kernel panic situation that is recurrent on 2 units (http://forum.xbian.org/thread-2783-page-3.html) and (http://forum.xbian.org/thread-2827.html).

Picture (http://s10.postimg.org/y6r0r3mnd/Kernel.jpg)

Thanks.

CurlyMoo, 2015-03-26:
Please try `compress=lzo` for btrfs and reboot (`/boot/cmdline.txt`)

The-Exnor, 2015-03-27:
Ok, trying that. will post progress.
Thks

The-Exnor, 2015-03-28:
@CurlyMoo  : Sorry mate this did not worked... tested on my 2 rPi 1 B and after less than 24h i've still got crashes and one Kernel Panic.

One of the units was freshly installed with the latest Xbian stable, available on the Win installer, then updated via apt-get. No other addons installed, no other configurations. Also OC @ xbian.

Thks.

CurlyMoo, 2015-03-28:
Can you try a fresh install but before you boot change the btrfs compress to lzo?

f1vefour, 2015-03-28:
@The-Exnor Please try this updated kernel for RPi 1 and let me know how it goes.

https://db.tt/sNM5ty8M

Install by doing:
```
sudo cp /boot/kernel.img /boot/kernel.good
sudo dpkg -i xbian-package-kernel_3.19.3+_armhf.deb
```

The-Exnor, 2015-03-28:
@CurlyMoo  and @f1vefour :

Ok so Curly i'm already running a fresh install with btrfs compress level at LZO on both parameters i will post if it crashes

@f1vefour  i'm going to install your Kernel on my other rPi 1 unit and will post the result (btw thank you for compiling that for me :) )

I will post results in here and on Xbian forums

CurlyMoo, 2015-03-28:
At least it booted here and the scrub succeeded.

The-Exnor, 2015-03-28:
@CurlyMoo 

Yes scrub does not crash anymore i've tested that too... but last time (after changing the btrfs compression) it also succeeded, and it lasted for several hours until it crashed.

f1vefour, 2015-03-28:
Thanks for testing, I am hoping we can move past the 3.18 branch and make 3.19 the default. I wish I could have pre-tested beforehand but I'm relatively sure the kernel will boot and function since a lot of Pi specific code has made it into the 3.19.y branch.

The-Exnor, 2015-03-28:
@f1vefour 
sorry for the noob question but after installation do i need to do anything else besides rebooting?

f1vefour, 2015-03-28:
Nope, just reboot. If you have still have issues I will help you update the firmware.

The-Exnor, 2015-03-28:
@f1vefour 
Ok i have it installed and its running fine so far.. it past btrfs scrub without issues... i hope this is the one.
Do you want me to send you any log file or info?

f1vefour, 2015-03-28:
You could pastebin dmesg and link here so I can make sure nothing is initially wrong, thanks.

The-Exnor, 2015-03-28:
@f1vefour  here you go mate (attached as .jpg... rename it to .txt :P) 
[image]

RolfTimmerman, 2015-03-28:
Here is mine: http://pastebin.com/CEjQp2BP

The-Exnor, 2015-03-28:
@RolfTimmerman 
is your running ok now?

RolfTimmerman, 2015-03-28:
For now but its only running for 5min ;)

RolfTimmerman, 2015-03-29:
Still running, fingers crossed!

The-Exnor, 2015-03-29:
Same here... so far so good.

The-Exnor, 2015-03-30:
@f1vefour 

Well the unit did not crash but kody did and i lost SSH so i had to hard reboot and now i'm stuck on the splash screen saying "init process" ... it does respond to ctrl+alt+del but i cant bypass the splash screen with any of  the key combinations that i've tried.

So i guess i have to re-install and restart the test... any suggestions? 

@CurlyMoo 
The unit running your suggestion is still up and running. Only Kodi crashed but the OS is responsive.

The-Exnor, 2015-03-30:
@CurlyMoo 
Just noticed one thing... the unit i've freshly installed and applied the LZO setting did not expand the file system... It was installed on a 8GiB card and only less than 400 Mib are used with only 44 MiB available on the system partition.
Also the swap partition was not created.

CurlyMoo, 2015-03-30:
Are you sure that's related to lzo?

The-Exnor, 2015-03-30:
@CurlyMoo  Not sure... i did nothing on that unit besides installing fresh, modify the LZO part and boot... this is the 1st time i actually take time to see how it was performing, besides not crashing.

The-Exnor, 2015-03-30:
@f1vefour 

After a fresh install on a new sdcard, and after installing the new Kernel and about 1 hour of runtime the unit froze again (no messages on screen) and on reboot the 2nd partition (btrfs) was corrupted again :/
No modifications on the system this time besides the installation of the Kernel (used latest stable image and did not performed any update.).
Any suggestions?

The-Exnor, 2015-03-30:
@CurlyMoo 

Just re-installed a new one with your idea. This time the filesystem/partitions were created normally. 

Used zswap.compressor=lz4 and compress=lzo. Last time i used zswap with lzo.

Lets see if it still holds like the last time, now that the partitions are correct.

The-Exnor, 2015-03-30:
@CurlyMoo  well scrub this idea mate... few minutes of doing nothing and i've just got Kernel Panic screen again... Is there any file i can post here to help?

CurlyMoo, 2015-03-30:
So it's not lz4 :smile:

The-Exnor, 2015-03-30:
@CurlyMoo it appears its not.... but i'm getting out of option here :(  new install crashes, new Kernel makes FS corrupt ... any ideas ??? from the forums i know i'm not the only one with the same problem.

Off topic: why LZ4 ? isn't this setting more compute demanding?

CurlyMoo, 2015-03-30:
We patch lz4 customly inside the kernel. Somehow it's IO related but i don't know where. I never had issues with normal Kodi + pilight usage.

The-Exnor, 2015-03-30:
@CurlyMoo  any theories about what is happening in this situations?

CurlyMoo, 2015-03-30:
Nope, maybe @mk01?

RolfTimmerman, 2015-03-30:
My RPi crashed again today

f1vefour, 2015-03-30:
It could be the bfs/bfq patches, the only thing left to try is build a kernel without those patches and see what happens. I think we need to provide a image without lz4 and an unpatched vanilla kernel because this is getting ridiculous.

f1vefour, 2015-03-30:
I have had no kernel panic on any of the kernels on either the Pi 1 or 2, I can get a freeze when a torrent approaches 5 megabyte/s. But it isn't locked up as if I leave it alone it comes back.

The-Exnor, 2015-03-30:
@f1vefour  so this is not an isolated case right? More people are getting the same problem for what i can see on the forums and in here.

Was there any major change on the last 2 kernels? because before everything was ok.

Also, now Kodi is very sluggish to respond to commands, either direct from keyboard or android app, and am talking about fresh out of an installation.

f1vefour, 2015-03-31:
I'm beginning to think it's a hardware issue, since 99% of all users aren't having these problems.

My current best recommendation is to downgrade to 3.17.y or earlier and see if you have the same issue, since this is the kernel you previously had no issue with.

Have you tried other SD cards?

CurlyMoo, 2015-03-31:
The most radical idea is building a ext4 root image. And that's what i'm doing now. If that is still crashing is has to be a hardware issue.

CurlyMoo, 2015-03-31:
Can you try this image:
http://download.xbian.org/testing/XBian_2015.03.31_rpi2-test.img.gz

The-Exnor, 2015-03-31:
@f1vefour yes i'm using 2 rPi 1 units and tried 3 different SDcards betwen them...

@CurlyMoo ok, just one question: this image is just for test or if is stable will i be able to continue to update it?

CurlyMoo, 2015-03-31:
Just for testing.

f1vefour, 2015-03-31:
Thanks @CurlyMo, this will definitely help determine the issue.

The-Exnor, 2015-03-31:
@CurlyMoo @f1vefour 

Thanks, tomorow i will give it a try ASAP. Today i don't have the time.

Ps: the .x.19 kernel just corrupted the 2nd partition for the 4th time today... (and its a different SDcard)

f1vefour, 2015-03-31:
Is your Pi UK or China made? Because the UK made has Samsung RAM and China is something else, this caused issues before.

The-Exnor, 2015-04-01:
@f1vefour  i bought it (both actually) from an official seller on the UK (link is directly on the Raspberry pi foundation page).  

Also i bought my 2nd Pi almost a year later than the 1st... and they both have the same problem :/

Just in case, how do identify if my rPi units are UK made or PRC?

The-Exnor, 2015-04-01:
@f1vefour @CurlyMoo 

Your test img does not boot on any of my rPi... i flash it using Win32DiskImager on 3 different SDcards.

Any suggestions?

f1vefour, 2015-04-01:
@The-Exnor The Pi on the right is UK made http://i.ytimg.com/vi/qP4qoVnhAEY/maxresdefault.jpg

The-Exnor, 2015-04-01:
@f1vefour  my oldest unit is made in PRC (black audio input jack) and my less old one (lol) is made on the UK (or at least its said so on the PCB and it has the blue audio jack).

f1vefour, 2015-04-01:
Good that you have both, that rules out the RAM theory.

What kernel is on the vanilla install?

The-Exnor, 2015-04-01:
@f1vefour as of today the x.18+ with ZL4 compression.

I'm testing something on one of the units... i got a theory about instability on some OC at the GPU (last year i've tried several setting and got some interesting results...) and i've reverted the clock to the default settings (below the ones on Xbian OC). 
So far this installation (downloaded today with the installer) with Kodi NOT running is still alive...

The other unit is also full updated and so far only Kodi crashes when idle (simple Stop/Start resolves that via SSH). Also this unit i'm using to keep seeing my shows is OC @ Turbo (you can see the pics on the forum, it has 2 active fans for heat dissipation) and never gave any problems like this.

Both units are powered by  2.5 A/h PSU's .

Nikamura, 2015-04-02:
Is there anything I can help, I have stock xbian on rpi2 with rtorrent running and after leaving it over night running it crashes or just if it uses 100% cpu for some time it also crashes

The-Exnor, 2015-04-03:
@f1vefour @CurlyMoo 

So my test unit that has kodi stopped, so far did no crash or Panic. (3.18.8+, with all the updates as of yesterday). I notice some "lag" via SSH console that was not there prior to this Kernel, but so far it passed the stress tests i made (reaching 48h without a Panic)

The other unit (where i can still watch a movie) when alone, freezes/crashes Kodi. (so typically i have to kill/start kodi twice a day on this unit).

Can this be an issue with Kodi itself or any of the services supporting it?

CurlyMoo, 2015-04-03:
Does this test unit run as download box with lots of IO?

The-Exnor, 2015-04-03:
@CurlyMoo  No... its idling with TOP running... but i did test moving big files over Wlan and it did no crash.

CurlyMoo, 2015-04-03:
Can you do a IO / CPU stresstest somehow for an extended period of time?

The-Exnor, 2015-04-03:
Yes... how do you propose i do that? Besides Kodi what else can use the CPU to the max?

CurlyMoo, 2015-04-03:
I would suggest googling for a nice script...

Smultie, 2015-04-03:
http://stackoverflow.com/questions/2925606/how-to-create-a-cpu-spike-with-a-bash-command

For example.

So 
fulload() { dd if=/dev/zero of=/dev/null | dd if=/dev/zero of=/dev/null | dd if=/dev/zero of=/dev/null | dd if=/dev/zero of=/dev/null & }; fulload; read; killall dd
should work

The-Exnor, 2015-04-03:
@CurlyMoo  just started kodi on that unit and it crashed in less than 5 min.... started slughish then input become even more slow and finally froze...

@Smultie  thnks mate going to try that one... i want to see if this is related to Kodi itself.

CurlyMoo, 2015-04-03:
The script from @smultie just tests memory IO, not disk IO.

Smultie, 2015-04-03:
It stresses all cores, which was the goal, right?

[67 more replies]

closed 2015-09-14 by mkreisl