xbianonpi/xbian #710: Kodi livetv playback crashes

zarboz, 2015-03-30:
When using livetv playback either through HDHR addon or through Mythtv.pvr 
screen will go black and pi will reboot

zarboz, 2015-03-30:
21:19:48 T:1960775680  NOTICE: special://profile/ is mapped to: special://masterprofile/
21:19:48 T:1960775680  NOTICE: -----------------------------------------------------------------------
21:19:48 T:1960775680  NOTICE: Starting Kodi (14.2-RC1 Git:Unknown). Platform: Linux ARM 32-bit
21:19:48 T:1960775680  NOTICE: Using Release Kodi x32 build (version for Raspberry Pi)
21:19:48 T:1960775680  NOTICE: Kodi compiled Mar 13 2015 by GCC 4.6.3 for Linux ARM 32-bit version 3.2.51 (197171)
21:19:48 T:1960775680  NOTICE: Running on XBian 1.0 (knockout), kernel: Linux ARM 32-bit version 3.18.8+
21:19:48 T:1960775680  NOTICE: FFmpeg statically linked, version: e197149-xbmc-2.4.6-Helix
21:19:48 T:1960775680  NOTICE: Host CPU: ARMv7 Processor rev 5 (v7l), 4 cores available
21:19:48 T:1960775680  NOTICE: ARM Features: Neon enabled
21:19:48 T:1960775680  NOTICE: special://xbmc/ is mapped to: /usr/local/share/kodi
21:19:48 T:1960775680  NOTICE: special://xbmcbin/ is mapped to: /usr/local/lib/kodi
21:19:48 T:1960775680  NOTICE: special://masterprofile/ is mapped to: /home/xbian/.kodi/userdata
21:19:48 T:1960775680  NOTICE: special://home/ is mapped to: /home/xbian/.kodi
21:19:48 T:1960775680  NOTICE: special://temp/ is mapped to: /home/xbian/.kodi/temp
21:19:48 T:1960775680  NOTICE: The executable running is: /usr/local/lib/kodi/kodi.bin
21:19:48 T:1960775680  NOTICE: Local hostname: xbian
21:19:48 T:1960775680  NOTICE: Log File is located: /home/xbian/.kodi/temp/kodi.log
21:19:48 T:1960775680  NOTICE: -----------------------------------------------------------------------
21:19:48 T:1960775680  NOTICE: load settings...
21:19:48 T:1960775680  NOTICE: Found 1 Lists of Devices
21:19:48 T:1960775680  NOTICE: Enumerated PI devices:
21:19:48 T:1960775680  NOTICE:     Device 1
21:19:48 T:1960775680  NOTICE:         m_deviceName      : HDMI
21:19:48 T:1960775680  NOTICE:         m_displayName     : HDMI
21:19:48 T:1960775680  NOTICE:         m_displayNameExtra:
21:19:48 T:1960775680  NOTICE:         m_deviceType      : AE_DEVTYPE_HDMI
21:19:48 T:1960775680  NOTICE:         m_channels        : FL,FR
21:19:48 T:1960775680  NOTICE:         m_sampleRates     : 8000,11025,16000,22050,24000,32000,44100,48000,88200,96000,176400,192000
21:19:48 T:1960775680  NOTICE:         m_dataFormats     : AE_FMT_FLOAT,AE_FMT_S32NE,AE_FMT_S16NE,AE_FMT_S32LE,AE_FMT_S16LE,AE_FMT_FLOATP,AE_FMT_S32NEP,AE_FMT_S16NEP,AE_FMT_AC3,AE_FMT_DTS,AE_FMT_EAC3
21:19:48 T:1960775680  NOTICE:     Device 2
21:19:48 T:1960775680  NOTICE:         m_deviceName      : Analogue
21:19:48 T:1960775680  NOTICE:         m_displayName     : Analogue
21:19:48 T:1960775680  NOTICE:         m_displayNameExtra:
21:19:48 T:1960775680  NOTICE:         m_deviceType      : AE_DEVTYPE_PCM
21:19:48 T:1960775680  NOTICE:         m_channels        : FL,FR
21:19:48 T:1960775680  NOTICE:         m_sampleRates     : 48000
21:19:48 T:1960775680  NOTICE:         m_dataFormats     : AE_FMT_FLOAT,AE_FMT_S32LE,AE_FMT_S16LE,AE_FMT_FLOATP,AE_FMT_S32NEP,AE_FMT_S16NEP
21:19:48 T:1960775680  NOTICE:     Device 3
21:19:48 T:1960775680  NOTICE:         m_deviceName      : Both
21:19:48 T:1960775680  NOTICE:         m_displayName     : HDMI and Analogue
21:19:48 T:1960775680  NOTICE:         m_displayNameExtra:
21:19:48 T:1960775680  NOTICE:         m_deviceType      : AE_DEVTYPE_PCM
21:19:48 T:1960775680  NOTICE:         m_channels        : FL,FR
21:19:48 T:1960775680  NOTICE:         m_sampleRates     : 48000
21:19:48 T:1960775680  NOTICE:         m_dataFormats     : AE_FMT_FLOAT,AE_FMT_S32LE,AE_FMT_S16LE,AE_FMT_FLOATP,AE_FMT_S32NEP,AE_FMT_S16NEP
21:19:48 T:1960775680  NOTICE: Loaded settings file from special://xbmc/system/advancedsettings.xml
21:19:48 T:1960775680  NOTICE: Contents of special://xbmc/system/advancedsettings.xml are...
                                            <advancedsettings>
                                              <fanartres>720</fanartres>
                                              <imageres>540</imageres>
                                              <libass>
                                                <styleoverrides>
                                                  <style name="Blur" value="0" />
                                                  <style name="BorderStyle" value="0" />
                                                  <style name="Shadow" value="0" />
                                                </styleoverrides>
                                              </libass>
                                              <network>
                                                <readbufferfactor>4.0</readbufferfactor>
                                              </network>
                                            </advancedsettings>
21:19:48 T:1960775680  NOTICE: Loaded settings file from special://profile/advancedsettings.xml
21:19:48 T:1960775680  NOTICE: Contents of special://profile/advancedsettings.xml are...
                                            <advancedsettings>
                                              <network>
                                                <cachemembuffersize>20971520</cachemembuffersize>
                                                <readbufferfactor>4.0</readbufferfactor>
                                              </network>
                                              <samba>
                                                <clienttimeout>10</clienttimeout>
[... 163 lines cut ...]
[... cut ...]
02:19:55 T:1960775680  NOTICE: Found resolution 1680 x 1050 for display 0 with 1680 x 1050 @ 59.940063 Hz
02:19:55 T:1960775680  NOTICE: Found (1920x1080@60.000000) at 22, setting to RES_DESKTOP at 16
02:19:58 T:1825559568  NOTICE: Thread PVRGUIInfo start, auto delete: false
02:19:59 T:1808782352  NOTICE: Start - EPG thread started
02:19:59 T:1687151632  NOTICE: Thread EPGUpdater start, auto delete: false
02:20:55 T:1717273616  NOTICE: ES: Client  from 192.168.1.4 timed out
02:22:55 T:1791357968  NOTICE: XBian : on saver
02:22:55 T:1791357968  NOTICE: XBian : on saver END

zarboz, 2015-03-30:
21:18:44 T:1960890368  NOTICE: special://profile/ is mapped to: special://masterprofile/
21:18:44 T:1960890368  NOTICE: -----------------------------------------------------------------------
21:18:44 T:1960890368  NOTICE: Starting Kodi (14.2-RC1 Git:Unknown). Platform: Linux ARM 32-bit
21:18:44 T:1960890368  NOTICE: Using Release Kodi x32 build (version for Raspberry Pi)
21:18:44 T:1960890368  NOTICE: Kodi compiled Mar 13 2015 by GCC 4.6.3 for Linux ARM 32-bit version 3.2.51 (197171)
21:18:44 T:1960890368  NOTICE: Running on XBian 1.0 (knockout), kernel: Linux ARM 32-bit version 3.18.8+
21:18:44 T:1960890368  NOTICE: FFmpeg statically linked, version: e197149-xbmc-2.4.6-Helix
21:18:44 T:1960890368  NOTICE: Host CPU: ARMv7 Processor rev 5 (v7l), 4 cores available
21:18:44 T:1960890368  NOTICE: ARM Features: Neon enabled
21:18:44 T:1960890368  NOTICE: special://xbmc/ is mapped to: /usr/local/share/kodi
21:18:44 T:1960890368  NOTICE: special://xbmcbin/ is mapped to: /usr/local/lib/kodi
21:18:44 T:1960890368  NOTICE: special://masterprofile/ is mapped to: /home/xbian/.kodi/userdata
21:18:44 T:1960890368  NOTICE: special://home/ is mapped to: /home/xbian/.kodi
21:18:44 T:1960890368  NOTICE: special://temp/ is mapped to: /home/xbian/.kodi/temp
21:18:44 T:1960890368  NOTICE: The executable running is: /usr/local/lib/kodi/kodi.bin
21:18:44 T:1960890368  NOTICE: Local hostname: xbian
21:18:44 T:1960890368  NOTICE: Log File is located: /home/xbian/.kodi/temp/kodi.log
21:18:44 T:1960890368  NOTICE: -----------------------------------------------------------------------
21:18:45 T:1960890368  NOTICE: load settings...
21:18:45 T:1960890368  NOTICE: Found 1 Lists of Devices
21:18:45 T:1960890368  NOTICE: Enumerated PI devices:
21:18:45 T:1960890368  NOTICE:     Device 1
21:18:45 T:1960890368  NOTICE:         m_deviceName      : HDMI
21:18:45 T:1960890368  NOTICE:         m_displayName     : HDMI
21:18:45 T:1960890368  NOTICE:         m_displayNameExtra:
21:18:45 T:1960890368  NOTICE:         m_deviceType      : AE_DEVTYPE_HDMI
21:18:45 T:1960890368  NOTICE:         m_channels        : FL,FR
21:18:45 T:1960890368  NOTICE:         m_sampleRates     : 8000,11025,16000,22050,24000,32000,44100,48000,88200,96000,176400,192000
21:18:45 T:1960890368  NOTICE:         m_dataFormats     : AE_FMT_FLOAT,AE_FMT_S32NE,AE_FMT_S16NE,AE_FMT_S32LE,AE_FMT_S16LE,AE_FMT_FLOATP,AE_FMT_S32NEP,AE_FMT_S16NEP,AE_FMT_AC3,AE_FMT_DTS,AE_FMT_EAC3
21:18:45 T:1960890368  NOTICE:     Device 2
21:18:45 T:1960890368  NOTICE:         m_deviceName      : Analogue
21:18:45 T:1960890368  NOTICE:         m_displayName     : Analogue
21:18:45 T:1960890368  NOTICE:         m_displayNameExtra:
21:18:45 T:1960890368  NOTICE:         m_deviceType      : AE_DEVTYPE_PCM
21:18:45 T:1960890368  NOTICE:         m_channels        : FL,FR
21:18:45 T:1960890368  NOTICE:         m_sampleRates     : 48000
21:18:45 T:1960890368  NOTICE:         m_dataFormats     : AE_FMT_FLOAT,AE_FMT_S32LE,AE_FMT_S16LE,AE_FMT_FLOATP,AE_FMT_S32NEP,AE_FMT_S16NEP
21:18:45 T:1960890368  NOTICE:     Device 3
21:18:45 T:1960890368  NOTICE:         m_deviceName      : Both
21:18:45 T:1960890368  NOTICE:         m_displayName     : HDMI and Analogue
21:18:45 T:1960890368  NOTICE:         m_displayNameExtra:
21:18:45 T:1960890368  NOTICE:         m_deviceType      : AE_DEVTYPE_PCM
21:18:45 T:1960890368  NOTICE:         m_channels        : FL,FR
21:18:45 T:1960890368  NOTICE:         m_sampleRates     : 48000
21:18:45 T:1960890368  NOTICE:         m_dataFormats     : AE_FMT_FLOAT,AE_FMT_S32LE,AE_FMT_S16LE,AE_FMT_FLOATP,AE_FMT_S32NEP,AE_FMT_S16NEP
21:18:45 T:1960890368  NOTICE: Loaded settings file from special://xbmc/system/advancedsettings.xml
21:18:45 T:1960890368  NOTICE: Contents of special://xbmc/system/advancedsettings.xml are...
                                            <advancedsettings>
                                              <fanartres>720</fanartres>
                                              <imageres>540</imageres>
                                              <libass>
                                                <styleoverrides>
                                                  <style name="Blur" value="0" />
                                                  <style name="BorderStyle" value="0" />
                                                  <style name="Shadow" value="0" />
                                                </styleoverrides>
                                              </libass>
                                              <network>
                                                <readbufferfactor>4.0</readbufferfactor>
                                              </network>
                                            </advancedsettings>
21:18:45 T:1960890368  NOTICE: Loaded settings file from special://profile/advancedsettings.xml
21:18:45 T:1960890368  NOTICE: Contents of special://profile/advancedsettings.xml are...
                                            <advancedsettings>
                                              <network>
                                                <cachemembuffersize>20971520</cachemembuffersize>
                                                <readbufferfactor>4.0</readbufferfactor>
                                              </network>
                                              <samba>
                                                <clienttimeout>10</clienttimeout>
[... 206 lines cut ...]
[... cut ...]
02:19:46 T:1959318544  NOTICE:         m_displayName     : HDMI and Analogue
02:19:46 T:1959318544  NOTICE:         m_displayNameExtra:
02:19:46 T:1959318544  NOTICE:         m_deviceType      : AE_DEVTYPE_PCM
02:19:46 T:1959318544  NOTICE:         m_channels        : FL,FR
02:19:46 T:1959318544  NOTICE:         m_sampleRates     : 48000
02:19:46 T:1959318544  NOTICE:         m_dataFormats     : AE_FMT_FLOAT,AE_FMT_S32LE,AE_FMT_S16LE,AE_FMT_FLOATP,AE_FMT_S32NEP,AE_FMT_S16NEP
root@xbian /home/xbian/.kodi/temp #

CurlyMoo, 2015-03-30:
Please use github markdown. Search for it if you don't know what it is.

zarboz, 2015-03-30:
@CurlyMoo  What style of markdown are you looking for as im not sure where in the logs the issue lies i just knew i should probably supply some kind of logs

CurlyMoo, 2015-03-30:
You obviously changed something... Good.

zarboz, 2015-03-30:
Also would like to note Both the mythtv.pvr and hdhomerun plugin work on openelec
So I figured It was an issue directly related to  Xbian
This problem is also specific to my raspberry pi 2
my hummingboard works fine

mk01, 2015-04-07:
@zarboz 

can you check some last lines from /var/log/upstart/xbmc.log (after the crash), ... or can you check 
``
ldd XXX.so
```

on those two binary plugins ? xbmc is monolitic (one big binary) but PVR addons are linkable modules at /usr/local/lib/kodi so try:
```
find /usr/local/lib/kodi -type f | xargs ldd 
```

and check if there is no "not found"

zarboz, 2015-04-08:
``find /usr/local/lib/kodi -type f | xargs ldd
/usr/local/lib/kodi/system/players/dvdplayer/libdvdcss-arm.so:
	libGLESv2.so => /opt/vc/lib/libGLESv2.so (0x76f7f000)
	libEGL.so => /opt/vc/lib/libEGL.so (0x76f55000)
	libopenmaxil.so => /opt/vc/lib/libopenmaxil.so (0x76f47000)
	libbcm_host.so => /opt/vc/lib/libbcm_host.so (0x76f27000)
	libvcos.so => /opt/vc/lib/libvcos.so (0x76f15000)
	libvchiq_arm.so => /opt/vc/lib/libvchiq_arm.so (0x76f07000)
	libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0x76ee7000)
	librt.so.1 => /lib/arm-linux-gnueabihf/librt.so.1 (0x76ed8000)
	libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0x76e67000)
	libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0x76d37000)
	libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0x76d2c000)
	/lib/ld-linux-armhf.so.3 (0x76fb9000)
/usr/local/lib/kodi/system/players/dvdplayer/libdvdnav-arm.so:
	libGLESv2.so => /opt/vc/lib/libGLESv2.so (0x76f56000)
	libEGL.so => /opt/vc/lib/libEGL.so (0x76f2c000)
	libopenmaxil.so => /opt/vc/lib/libopenmaxil.so (0x76f1e000)
	libbcm_host.so => /opt/vc/lib/libbcm_host.so (0x76efe000)
	libvcos.so => /opt/vc/lib/libvcos.so (0x76eec000)
	libvchiq_arm.so => /opt/vc/lib/libvchiq_arm.so (0x76ede000)
	libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0x76ebe000)
	librt.so.1 => /lib/arm-linux-gnueabihf/librt.so.1 (0x76eaf000)
	libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0x76e3e000)
	libgcc_s.so.1 => /lib/arm-linux-gnueabihf/libgcc_s.so.1 (0x76e16000)
	libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0x76ce6000)
	libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0x76cdb000)
	/lib/ld-linux-armhf.so.3 (0x76fbf000)
/usr/local/lib/kodi/system/players/paplayer/vgmstream-arm.so:
	libGLESv2.so => /opt/vc/lib/libGLESv2.so (0x76f2b000)
	libEGL.so => /opt/vc/lib/libEGL.so (0x76f01000)
	libopenmaxil.so => /opt/vc/lib/libopenmaxil.so (0x76ef3000)
	libbcm_host.so => /opt/vc/lib/libbcm_host.so (0x76ed3000)
	libvcos.so => /opt/vc/lib/libvcos.so (0x76ec1000)
	libvchiq_arm.so => /opt/vc/lib/libvchiq_arm.so (0x76eb3000)
	libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0x76e93000)
	librt.so.1 => /lib/arm-linux-gnueabihf/librt.so.1 (0x76e84000)
	libstdc++.so.6 => /usr/lib/arm-linux-gnueabihf/libstdc++.so.6 (0x76db7000)
	libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0x76d46000)
	libgcc_s.so.1 => /lib/arm-linux-gnueabihf/libgcc_s.so.1 (0x76d1e000)
	libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0x76bee000)
	libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0x76be2000)
	/lib/ld-linux-armhf.so.3 (0x76fae000)
/usr/local/lib/kodi/system/players/paplayer/timidity-arm.so:
	libGLESv2.so => /opt/vc/lib/libGLESv2.so (0x76d96000)
	libEGL.so => /opt/vc/lib/libEGL.so (0x76d6c000)
	libopenmaxil.so => /opt/vc/lib/libopenmaxil.so (0x76d5e000)
	libbcm_host.so => /opt/vc/lib/libbcm_host.so (0x76d3e000)
	libvcos.so => /opt/vc/lib/libvcos.so (0x76d2c000)
	libvchiq_arm.so => /opt/vc/lib/libvchiq_arm.so (0x76d1e000)
	libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0x76cfe000)
	librt.so.1 => /lib/arm-linux-gnueabihf/librt.so.1 (0x76cef000)
	libstdc++.so.6 => /usr/lib/arm-linux-gnueabihf/libstdc++.so.6 (0x76c22000)
	libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0x76bb1000)
	libgcc_s.so.1 => /lib/arm-linux-gnueabihf/libgcc_s.so.1 (0x76b89000)
	libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0x76a59000)
	libdl.so.2 => /lib/arm-linux-gnueabihf/libdl.so.2 (0x76a4d000)
	/lib/ld-linux-armhf.so.3 (0x76f1d000)
/usr/local/lib/kodi/system/players/paplayer/nosefart-arm.so:
	libGLESv2.so => /opt/vc/lib/libGLESv2.so (0x76ece000)
	libEGL.so => /opt/vc/lib/libEGL.so (0x76ea4000)
	libopenmaxil.so => /opt/vc/lib/libopenmaxil.so (0x76e96000)
	libbcm_host.so => /opt/vc/lib/libbcm_host.so (0x76e76000)
	libvcos.so => /opt/vc/lib/libvcos.so (0x76e64000)
	libvchiq_arm.so => /opt/vc/lib/libvchiq_arm.so (0x76e56000)
	libpthread.so.0 => /lib/arm-linux-gnueabihf/libpthread.so.0 (0x76e36000)
	librt.so.1 => /lib/arm-linux-gnueabihf/librt.so.1 (0x76e27000)
	libm.so.6 => /lib/arm-linux-gnueabihf/libm.so.6 (0x76db6000)
	libgcc_s.so.1 => /lib/arm-linux-gnueabihf/libgcc_s.so.1 (0x76d8e000)
	libc.so.6 => /lib/arm-linux-gnueabihf/libc.so.6 (0x76c5e000)
[... 486 lines cut ...]
	liblzma.so.5 => /lib/arm-linux-gnueabihf/liblzma.so.5 (0x750e8000)
	libjson.so.0 => /lib/arm-linux-gnueabihf/libjson.so.0 (0x750d8000)
	libpulsecommon-2.0.so => /usr/lib/arm-linux-gnueabihf/pulseaudio/libpulsecommon-2.0.so (0x75076000)
	libattr.so.1 => /lib/arm-linux-gnueabihf/libattr.so.1 (0x75069000)
	libxcb.so.1 => /usr/lib/arm-linux-gnueabihf/libxcb.so.1 (0x7504a000)
	libcrypt.so.1 => /lib/arm-linux-gnueabihf/libcrypt.so.1 (0x75013000)
	libkrb5support.so.0 => /usr/lib/arm-linux-gnueabihf/libkrb5support.so.0 (0x75003000)
	libkeyutils.so.1 => /lib/arm-linux-gnueabihf/libkeyutils.so.1 (0x74ff8000)
	libsasl2.so.2 => /usr/lib/arm-linux-gnueabihf/libsasl2.so.2 (0x74fd9000)
	libX11-xcb.so.1 => /usr/lib/arm-linux-gnueabihf/libX11-xcb.so.1 (0x74fd0000)
	libICE.so.6 => /usr/lib/arm-linux-gnueabihf/libICE.so.6 (0x74fb2000)
	libSM.so.6 => /usr/lib/arm-linux-gnueabihf/libSM.so.6 (0x74fa4000)
	libXtst.so.6 => /usr/lib/arm-linux-gnueabihf/libXtst.so.6 (0x74f98000)
	libwrap.so.0 => /lib/arm-linux-gnueabihf/libwrap.so.0 (0x74f88000)
	libsndfile.so.1 => /usr/lib/arm-linux-gnueabihf/libsndfile.so.1 (0x74f1f000)
	libasyncns.so.0 => /usr/lib/arm-linux-gnueabihf/libasyncns.so.0 (0x74f12000)
	libXau.so.6 => /usr/lib/arm-linux-gnueabihf/libXau.so.6 (0x74f08000)
	libXdmcp.so.6 => /usr/lib/arm-linux-gnueabihf/libXdmcp.so.6 (0x74efc000)
	libuuid.so.1 => /lib/arm-linux-gnueabihf/libuuid.so.1 (0x74eef000)
	libXext.so.6 => /usr/lib/arm-linux-gnueabihf/libXext.so.6 (0x74ed9000)
	libXi.so.6 => /usr/lib/arm-linux-gnueabihf/libXi.so.6 (0x74ec4000)
	libFLAC.so.8 => /usr/lib/arm-linux-gnueabihf/libFLAC.so.8 (0x74e79000)
xbian@xbian ~ $ 
``
@mk01

mk01, 2015-04-19:
@zarboz 

for  test can you move your advancedsettings.xlm file out from /home/xbian/.kodi/userdata ?
(so that xbmc would start without it)

zarboz, 2015-06-25:
@mk01 that makes no difference

closed 2016-06-17 by mkreisl