xbianonpi/xbian #711: Today's updates (xbian-update and xbian-sources-xbmc and xbian-package-xbmc) broke something

Smultie, 2015-04-07:
Kodi doesn't start at bootup and the /home directory is empty

Whenever I login with SSH:

*Last login: Tue Apr  7 15:14:48 2015 from 10.0.1.4
Could not chdir to home directory /home/xbian: No such file or directory
xbian@xbian / $ *

Smultie, 2015-04-07:
Narrowed it down to xbian-update 20150407-1
All the other updates work fine.

mk01, 2015-04-07:
see my comment in that issue

thomasbishop, 2015-04-08:
Same problem here. Applied today's update via the SSH menu, rebooted and... it fails with the same error message as @Smultie is seeing.

@mk01 I can't answer your question in the other thread as I can't get in via SSH (my Pi is wireless only, I guess the boot sequence is failing before the wireless connection is initialised) but I've never manually set anything on the Pi... I did a clean installation when the Pi v2 arrived, then applied every update through the SSH menu. No manual changes to config files, no installations via apt-get etc. Just a 'vanilla' configuration! Hope that helps.

mk01, 2015-04-08:
@thomasbishop 

sorry for the break. I removed that particular xbian-update from apt already - hopefully you are just few users who managed to process that update almost instantly. 

indeed that info helped and I have the feeling it will be ok with the fixed package (already released). just need it to get confirmed from you (once you manage to get to your RPI) and @Smultie. 

@Smultie will just reinstall xbian-update because his PRI manage to start with network, ... you will have to remove /etc/default/tmpfs file (if my theory is correct), then reboot, ... then your RPI should boot correctly, then try the last xbian-update. it should be ok after this update.

I have the feeling the problem with /home mounting (or even bigger with you) wasn't caused directly by the new code - and being dormant, but with wrong /etc/default/tmpfs file update -> some parts updated to garbage (as @mkreisl said earlier). 

mount/mountall is reading this file and that's the moment when it probably due to the garbage collapses (whole boot in your case, "only" incorrect mounting for @Smultie)).

once you guys recover and retest, report please back.

thomasbishop, 2015-04-08:
OK, thanks - I'll try that. I just tried booting with keyboard plugged in but the system is completely frozen after the line:

    [ ok ] Setting up X socket directories... /tmp/.X11-unix /tmp/ .ICE-unix.

so I'll need to pull the SD card and open it on another computer to delete the `tmpfs` file. Will report back asap.

mk01, 2015-04-08:
uncomment initramfs.gz line in /boot/config.txt to have boot with initramfs (or if it is booting like that) - just hold left shift on keyboard. you will get recovery shell right away. with some short howto.

mk01, 2015-04-08:
@thomasbishop 
you won't mount the root of another (not patched accordingly) kernel.

still, to save you from panic and reflash - edit cmdline.txt on the 1st fat partition, put keyword "rescue" and save, edit config.txt and uncomment "initramfs.gz" line. 

you will get recovery shell (with rootfs mounted under /rootfs/). so delete 
```
rm /rootfs/etc/default/tmpfs
reb
```

(reb instead of reboot is correct)

Smultie, 2015-04-08:
Your comments didn't work @mk01 
After doing what you told, still no Kodi and no home-folder.

tom4242, 2015-04-08:
I have the same problems and think that it is a general problem with an updated package for the pi 2.
i tried the stable release and also the relase of 2015.04.04. after an apt-get update & upgrade and a reboot xbian stopped working. Error message described at the top from Smultie . Very strange.

mk01, 2015-04-08:
@Smultie 

and if you revert to xbian-update last before april ? 20150304-2

apt-get install xbian-update=20150304-2

?

Smultie, 2015-04-08:
No problem with all xbian-update version before 20150307-1.
So only problems with 20150307-1 and 20150308-0.

agijsberts, 2015-04-08:
I ran into the same error reported by thomasbishop when I update to 20150408-0 (either via 20150407-0 or directly from an earlier bcakup). Pinning xbian-update to version 20150304-2 and all is ok again.

Before getting stuck at the (unrelated?) [i]X socket directories[/i] message it is displaying the [i]Loading[/i] splash screen for an unusually long time.

The-Exnor, 2015-04-08:
Same problem here... fortunately i only updated 1 of my units....

mkreisl, 2015-04-08:
Hmm, why do I have no problems with 20150307-1 and 20150308-0 ?
Tested on my 2 PIs, no issue

Smultie, 2015-04-08:
@mkreisl what RPi version do you have? I have an RPi 2.

agijsberts, 2015-04-08:
I have an RPi 1 model B instead.

mkreisl, 2015-04-08:
@Smultie 
2x RPi 1B

er0l, 2015-04-08:
rpi2 narrowed down the issue back to update 20150407-1

kataclank, 2015-04-08:
same problem with my rpi b

Lexmark0381, 2015-04-08:
Same here, RPI2.
Even cable-connection is giving troubles...

kataclank, 2015-04-09:
sudo apt-get install xbian-update=20150304-2 Fix it and everything works fine now

Smultie, 2015-04-09:
sudo apt-get install xbian-update=20150407-0 works fine too

er0l, 2015-04-09:
I can confirm apt-get install xbian-update=20150407-0 works on my rpi2 too.

CurlyMoo, 2015-04-09:
I've removed `20150408-0` and see that `20150407-0` was removed earlier by another dev. That should at least fix the spread of the issue to more users.

Smultie, 2015-04-09:
If I'm current, only 20150407-1 was removed.

http://apt.xbian.org/pool/stable/main/x/xbian-update/

20150407-0 is still on there.

CurlyMoo, 2015-04-09:
I updated my post. `20150407-0` was wrongly named on the server so i couldn't find it.

dams1911, 2015-04-09:
Hello, I had the same problem yesterday and I've reinstalled my Pi2... (didn't know I could just downgrade to fix it).

Can I turn the auto update ON again now or should I wait ?

The-Exnor, 2015-04-09:
@CurlyMoo just reinstalled the OS (again) an hour ago, updated (minus xbian-package-xbmc because of the reasons on the other thread) and so far no problems.

m-urban, 2015-04-09:
Hi there,
I was running into the same issue yesterday with my RPi 1 B+. Running "apt-get install xbian-update=20150304-2" fixed it (thank you, @mk01!).

xiolo, 2015-04-15:
20150407-0 works, but the list of local installed packages (nzbget etc..) si empty.

xiolo, 2015-04-15:
also 20150407-0 on rpi2 doesn't allow for additional packages to be installed, as it brakes dependencies (libc6 error?).
Has to flash back to 20150304-2, install all updates except xbian-update-20150407-0, and the packages are then fine.

Moral of the story from my side on an RPI2: 20150304-2 is fine, 20150407-0 has broken packages list/installation.

xiolo, 2015-04-16:
update on this. I restored a previous 20150304-2 backup, and carefully did an apt-get update/apt-get upgrade, and now running 20150407-0 on the pi2 with the list of packages all in order...

So unsure what was my previous issue (dodgy file/fs?).

Still getting a kernel panic/freeze every few days, so me thinking that 3.18 kernel (especially around the ethernet driver) is still a bit unstable :-)

mk01, 2015-04-19:
@xiolo 

xbian-update itself didn't change own dependencies for a while so you are right it was "something in the clouds". 

@all

returning back to original topic - I would like to deploy the code (changed xbian-run.conf). as I can't replicate the issue on mine boxes, I would need someone who original had problem with the mentioned xbian-update and can provide remote access (so I would ssh in the box). 
that also means that the test subject's "broken" system should be like @Smultie's. no /home mounted, xbmc not starting (due to /home), but ssh works - so I would see untouched system after reboot. 

if nobody goes voluntarily, I will go one by one without asking :D

Smultie, 2015-04-19:
@mk01 I can test for you, no problem.

Lexmark0381, 2015-04-19:
@mk01 Sorry man, i made a clean installation after that bug..

mk01, 2015-04-19:
@Smultie 

this time no "test for me" ... I  would need to see it myself (so I need logging in).

Smultie, 2015-04-19:
Okay. I can provide you access

mk01, 2015-04-19:
tell me a approx timing when you could loose maybe 1h doing anything but RPI.
and
I would send you deb. you dpkg -i , reboot... and provide access credentials. so ~10min, then I won't need you.
(we do snap before and rollback after so as I logout, you have your system back)

Smultie, 2015-04-19:
Now would be fine
Can you come to IRC?

B0rax, 2015-08-08:
>also 20150407-0 on rpi2 doesn't allow for additional packages to be installed, as it brakes dependencies (libc6 error?).
Has to flash back to 20150304-2, install all updates except xbian-update-20150407-0, and the packages are then fine.

how did you flash back? I'm on 20150407-0 on rpi2 and can't install anything (even the old updates), I'm quite stuck as it is now. I'm unable to physically access the SD card, so I would really prefer a solution over ssh (or directly via keyboard). Can anyone help me with this?

closed 2016-06-17 by mkreisl