xbianonpi/xbian #715: RPI2 xbian kernel panic while intense use

ErickPH, 2015-04-11:
I now have an RPI2, and I think Im facing same issues as in RPI1 when I was in the middle of massive downloads (lots of IO to my USB3 BTRFS disk) perhaps is the same as [issue](https://github.com/xbianonpi/xbian/issues/709)
@f1vefour do you have a kernel for RPI2 as the one you provided for  issue 709?

Attached are screen captures. Can I please be reffered to an article that can help me setup a proper capture process?

```
uname -a
Linux xbian 3.18.8+ #1 SMP PREEMPT Thu Mar 5 03:13:36 CET 2015 armv7l GNU/Linux

xbian@xbian /boot $ cat config.txt
#initramfs initramfs.gz  0x1000000
gpu_mem_1024=256
gpu_mem_512=144
gpu_mem_256=100
initial_turbo=3
hdmi_ignore_cec_init=1
disable_overscan=1
disable_splash=1
xbian@xbian /boot $

xbian@xbian /boot $ cat cmdline.txt
telnet zswap.enabled=1 zswap.compressor=lz4 sdhci-bcm2708.sync_after_dma=0 dwc_otg.lpm_enable=0 console=tty1 root=/dev/mmcblk0p2 rootflags=subvol=root/@,thread_pool=2,autodefrag,compress=lz4,commit=120 rootfstype=btrfs rootwait smsc95xx.turbo_mode=N logo.nologo quiet noswap loglevel=0 mod_scsi.scan=sync partswap startevent=mountall splash nohdparm --startup-event mountall
xbian@xbian /boot $

```
thanks
[image]
[image]
[image]

FatFairFoot, 2015-04-27:
Got the same issue here !
[image]

root@xbian /boot # uname -a
Linux xbian 3.18.8+ #1 SMP PREEMPT Thu Mar 5 03:13:36 CET 2015 armv7l GNU/Linux
root@xbian /boot # cat config.txt
#initramfs initramfs.gz  0x1000000
gpu_mem_1024=256
gpu_mem_512=144
gpu_mem_256=100
initial_turbo=3
hdmi_ignore_cec_init=1
disable_overscan=1
disable_splash=1
arm_freq=1000
core_freq=500
sdram_freq=500
over_voltage=2
root@xbian /boot # cat cmdline.txt
telnet zswap.enabled=1 zswap.compressor=lz4 sdhci-bcm2708.sync_after_dma=0 dwc_otg.lpm_enable=0 console=tty1 root=/dev/mmcblk0p2 rootflags=subvol=root/@,thread_pool=2,autodefrag,compress=lz4,commit=120 rootfstype=btrfs rootwait smsc95xx.turbo_mode=N logo.nologo quiet noswap loglevel=0 mod_scsi.scan=sync partswap startevent=mountall splash nohdparm --startup-event mountall
root@xbian /boot #

Thanks for your help !
This could happen several times per day !!

OtakuWally, 2015-04-30:
Running on Pi2 as well and using transmission pretty extensively, nut no kernel panics yet (thank god). But today I had a complete crash (everything simply stops, no network connections possible anymore), will try to reproduce it these days. But since I regularly backup my SD it's not (yet) a BIG issue for me right now ...

By the way: I am really missing the "screen" deb package on xbian ;-)

CurlyMoo, 2015-04-30:
Screen is available from the default debian repository.

OtakuWally, 2015-05-01:
So WHAT exactly is right now? I thought one should NOT install packages from the default repositories? Or is this just true for the apt utility to upgrade the system? What is OK and what should be avoided in the meaning of package management? Thanks for your help CurlyMoo.

BTW: Does CurlyMoo mean that you are some hairy cow? o.O :-D

CurlyMoo, 2015-05-01:
Ofc you can install packages from the normal repository. That's the whole point of us building on top of debian. Just don't run `apt-get dist-upgrade` or `rpi-update`.

OtakuWally, 2015-05-01:
Thanks for the correction of my mental error, this makes many of my wishes true :-D (How could one ever live WITHOUT screen on remote logins ;-) )

HugoAdrian, 2015-05-29:
The first time you install Xubuntu was a success, but the second I get this error in the Raspberry Pi b 2 in the installation of Xubuntu of   https://wiki.ubuntu.com/ARM/RaspberryPi   following the video    https://www.youtube.com/watch?v=UGSQ7nzVCs4       in the last step, but not my desktop image below appears                                               
[image]

HugoAdrian, 2015-05-29:
help

CurlyMoo, 2015-05-29:
EU call 112, US call 911.

The-Exnor, 2015-05-29:
Did i miss something here? Is this a Xubuntu help request or a Xbian related problem?

HugoAdrian, 2015-05-29:
Quizás quisiste decir: Es un problema de Ubuntu corriendo en Raspberry Pi B 2, en las imagenes de abajo esta el primer test que tube con exito pero la otra imagen es un error en la segunda instalacion que hice, ayuda por favor, no se que hacer
Xubuntu is a problem running on Raspberry Pi B 2, in the images below is the first test tube successfully but the other image is an error on the second installation I did, please help, not to do
[image]
[image]

HugoAdrian, 2015-05-29:
Xubuntu is a problem running on Raspberry Pi B 2, in the images below is the first test tube successfully but the other image is an error on the second installation I did, please help, not to do
[image]
[image]

CurlyMoo, 2015-05-29:
So back to the question of @The-Exnor:
> Is this a Xubuntu help request or a Xbian related problem?

The-Exnor, 2015-05-30:
@CurlyMoo i still don't understand if this guy is mistaken or i missed something... isn't this Xbian issues section?

CurlyMoo, 2015-05-30:
Ofc, that's why i repeated your question.

The-Exnor, 2015-05-30:
@CurlyMoo  lol. We should tell him that he is mistaken...

s47, 2015-06-06:
I've got the exact same problem as @ErickPH when using transmission downloading big files.
Any news on this?

CurlyMoo, 2015-06-06:
Maybe you guys can try this:
http://forum.xbian.org/thread-2783-post-29202.html#pid29202

OtakuWally, 2015-06-06:
I got the same problem over here, already two times right now:

[image]
[image]

In case you need more info, just tell me, but it simply seems to be a kernel issue.

s47, 2015-06-08:
I ended up removing my wifi dongle and switching to cable and the issue hasn't happened to me since

OtakuWally, 2015-06-08:
Well, thanks for the info. But I have never seen the colors-square so I assume my cable should be ok.

coderpop, 2015-06-18:
Same here on a pi2.

This time on wired ethernet (no wifi). Happens while remuxing videos using avconv.
Oddly it seems to complete the task of remuxing the videos and then crashes sometime after it's finished. (?)...
Only way out is to pul the power cord and restart the pi.
Screen looks similar to the first (erik PH) pictures above with the kodi desktop showing under the text from the kernel panic.

mkreisl, 2015-06-18:
@coderpop 
Do you overclock the RPi2? I had same trouble with my brand new RPi2 (1GHz clocked) which I bought for testing. I used a lot of kernels (3.18, 3.19, 4.01), nothing helped. Only back to standard clock (900MHz) seems to help, no kernel oops/panic since many days.

coderpop, 2015-06-19:
mkreisl,

You are correct yet again!

I was running in overclock mode and went back to default last night. I left the pi remuxing videos for 12 hours and no problems. Then it was left alone for 8 hours and is now remuxing again. No panic and no reboots.

Thanks!

CP

OtakuWally, 2015-06-19:
Well, MINE is NOT overclocked. Never do that since the performance gain is marginal and all you possible get from it is PROBLEMS ;-)

mkreisl, 2015-06-19:
@OtakuWally 

> Well, MINE is NOT overclocked. Never do that since the performance gain is marginal and all you possible get from it is PROBLEMS ;-)

Sounds bad :-(

In case of RPi2 you are right. But on RPi1 there is a difference if you run it with 700MHz or 1Ghz (like my 2 RPI1's). Never had trouble overclocking my RPi1's up to 1GHz.

OtakuWally, 2015-06-20:
You mean, you never had trouble you were able to notice ;-) Data can get corrupted in RAM etc ... Overclocking is useless in my eyes, but you can of course do whatever you want ;-)

closed 2015-09-14 by mkreisl