xbianonpi/xbian #716: Xbian GNU/Linux 7 xbian tty1 xbian login

brentonmeeks, 2015-04-15:
I have 2 pi2's and both up to date yet one keeps kicking me out of kodi to a black screen that says what is my title. I have started from scratch 3 times now yet this one pi keeps doing the same thing. I deleted my boot/config thinking my overclock was doing it i used a wireless connection a lan connection i restored an image that was working for a day but this continues. Please anyone with any ideas please let me know did i get a bad rpi2.

mk01, 2015-04-19:
@brentonmeeks 

is  there any interesting output in /var/log/upstart/xbmc.log ?
can you check ?

brentonmeeks, 2015-04-20:
i started all over from scratch and took my time and i have not had this happen again. not sure what was going on but it seems some of the updates were causing me some issues so i am not updating anything right now im scared it will break it. thankyou for responding and all your hard work we can close this case.

closed 2015-04-20 by brentonmeeks