xbianonpi/xbian #717: XBian/Kodi not always switching refresh rate

Smultie, 2015-04-16:
I think the title says all.

Seems to happen only when the TV has been off for a while.
Might be related to XBian Saver (whatever that may be).

From kodi.log:
19:27:59 T:1781527568  NOTICE: XBian : on saver
19:27:59 T:1873294352  NOTICE: Found resolution 640x480 (640x480) @ 60.00 - Full Screen
19:27:59 T:1873294352  NOTICE: Found resolution 640x480 (640x480) @ 59.94 - Full Screen
19:27:59 T:1873294352  NOTICE: Found resolution 720x480 (720x480) @ 60.00 - Full Screen
19:27:59 T:1873294352  NOTICE: Found resolution 720x480 (720x480) @ 59.94 - Full Screen
19:27:59 T:1873294352  NOTICE: Found resolution 1280x720 (1280x720) @ 60.00 - Full Screen
19:27:59 T:1873294352  NOTICE: Found resolution 1280x720 (1280x720) @ 59.94 - Full Screen
19:27:59 T:1873294352  NOTICE: Found resolution 1920x1080 (1920x1080) @ 60.00 - Full Screen
19:27:59 T:1873294352  NOTICE: Found resolution 1920x1080 (1920x1080) @ 59.94 - Full Screen
19:27:59 T:1873294352  NOTICE: Found resolution 720x576 (720x576) @ 50.00 - Full Screen
19:27:59 T:1873294352  NOTICE: Found resolution 1280x720 (1280x720) @ 50.00 - Full Screen
19:27:59 T:1873294352  NOTICE: Found resolution 1920x1080 (1920x1080) @ 50.00 - Full Screen
19:27:59 T:1873294352  NOTICE: Found resolution 1920x1080 (1920x1080) @ 24.00 - Full Screen
19:27:59 T:1873294352  NOTICE: Found resolution 1920x1080 (1920x1080) @ 23.98 - Full Screen
19:28:00 T:1781527568  NOTICE: XBian : on saver deactivated
19:28:00 T:1781527568  NOTICE: XBian : on saver deactivated END

After this example refresh rate switching worked well.

Another example later that day:
20:51:44 T:1781527568  NOTICE: XBian : on saver
20:51:44 T:1873294352  NOTICE: Found resolution 1920x1080 (1920x1080) @ 60.00 - Full Screen
20:51:44 T:1781527568  NOTICE: XBian : on saver deactivated

After that instance, refresh rate switching no longer worked.

mk01, 2015-04-19:
@Smultie 

from a reason unknown (never seen before), after HPE events on HDMI (also TV going ON counts) only one resolution (I assume the default you run) is enumerated as available.

is that 14.2 ? can you ~ define time period, when it started ? can you (for future) keep at least +-5,+-10 lines around this resolution refresh action ?

mk01, 2015-04-19:
@Smultie 

please also edit xbmc.conf in /etc/init, change for debug logging (DEBUG='' to DEBUG='--debug').

Smultie, 2015-04-19:
That is indeed on 14.2 
Not really sure when exactly it started, might have been quite a while. I've noticed XBian not switching the resolution before, just never thought about checking kodi.log before

mk01, 2015-04-19:
@Smultie 

ok, try guessing. it's been moths, weeks, days ? was it still Gotham, or already Kodi  ?

Smultie, 2015-04-19:
Pure guess would be it started with Kodi Helix!

mk01, 2015-04-19:
hot plug events... 
hdmi standard defines 2x2 hi-lo switch-overs to signal presence - but that also means when looking at the event in minimal steps, it signals - 1. cable was unplugged, 2. replugged, 3. unplugged, 4. replugged.

mk01, 2015-04-28:
@Smultie 

confirm the custom xbmc you are testing that we can roll out new rpi and rpi-nighty packages.

Smultie, 2015-04-28:
If I don't reply in this thread the following 5 days I think we can assume the issue is fixed.