xbianonpi/xbian #718: High CPU Load on kodi 14.2-1429432779 whern idle

Fabio72, 2015-04-22:
Since i updated kodi to 14.2-1429432779 I'm getting this

top - 12:05:48 up 17:30,  1 user,  load average: 2,71, 2,67, 2,65
Tasks:  72 total,   1 running,  71 sleeping,   0 stopped,   0 zombie
%Cpu(s): 78,8 us,  2,5 sy,  0,0 ni, 11,4 id,  0,0 wa,  7,1 hi,  0,3 si,  0,0 st
KiB Mem:    366012 total,   263304 used,   102708 free,      512 buffers
KiB Swap:   501756 total,        0 used,   501756 free,   163468 cached

  PID USER      PR  NI  VIRT  RES  SHR S  %CPU %MEM    TIME+  COMMAND
 1323 xbian    -26   0  439m  81m  37m S  79,9 22,7 314:23.97 kodi.bin

mk01, 2015-04-28:
start "top" utility. sort by cpu usage, show threads instead of tasks.

(after top started, press "P" (shift+p), then "H")

what threads are taking the cpu time?

f1vefour, 2015-07-15:
It's also relevant which screen you are leaving kodi idling at.

closed 2015-11-30 by mkreisl

The-Exnor, 2015-11-30:
I had that problem too (and posted on the forums) but it got solved with the last update.