xbianonpi/xbian #719: Last xbian-package-cec update breaks all

Smultie, 2015-04-26:
Saw there was an update today (xbian-package-cec), installed it, saw it also updated xbian-package-kernel. After a reboot all I see is a rainbow screen.

CurlyMoo, 2015-04-26:
Was there a backup created of the kernel and initramfs? If so, restore them and try again...

Smultie, 2015-04-26:
Since I had a backup I tried reinstalling. Unfortunately the backup doesn't work either. So now I lost all my settings/MySQL/SickRage etc.

Great..

Smultie, 2015-04-26:
Updating has been a real pita lately.

CurlyMoo, 2015-04-26:
I meant a backup of those files on your original SD.

Smultie, 2015-04-26:
Yeah, like I said: I thought that since I have a backup (from xbian-config, option 7) It would be fine to just reinstall. It wasn't.

CurlyMoo, 2015-04-26:
Maybe that some juice for another issue. I never had problems with the backup feature, but i always have my settings on my NAS symlinked to the pi so haven't tested in a while.

rafdelre, 2015-04-27:
I can confirm that on PI2 CEC does not work anymore withe latest update CEC

CurlyMoo, 2015-04-27:
Are you sure this is CEC and not the kernel?

Smultie, 2015-04-27:
Yesterday there was a wrong dependency in xbian-package-cec, last night's update fixes that but apparently has a dependency that's not available on fresh installs.

bairdy, 2015-04-27:
I Had updated yesterday and it also broke with package 9.2.2.0-1-1430045420 but there was also another package released yesterday 9.2.2.0-1-1430044352 which was in the repo but no longer there which was working. when going to try and test cec is active and working over ssh i get the following on  9.2.2.0-1-1430045420
```root@xbian ~ # cec-client
No device type given. Using 'recording device'
libXrandr.so.2: cannot open shared object file: No such file or directory
Cannot load libcec.so```
where previously it gives the debug and allows to test input

CurlyMoo, 2015-04-27:
So the very latest CEC doesn't `break` the system. It just breaks CEC?

bairdy, 2015-04-27:
yeah thats correct

CurlyMoo, 2015-04-27:
Update the title.

mk01, 2015-04-28:
libcec 2.2 by default links against libXrandr.so (X11) support.

it was corrected immediately on next APT rebuild window (1200am yesterday). 
```
xbian-package-cec:
  Installed: 9.2.2.0-1-1430117049
  Candidate: 9.2.2.0-1-1430117049
  Version table:
 *** 9.2.2.0-1-1430117049 0
        500 mirror://apt.xbian.org/mirror.txt/ stable/rpi-wheezy armhf Packages
        100 /var/lib/dpkg/status
```

RolfTimmerman, 2015-04-28:
For me CEC works again even when using the screensaver

doming52, 2015-04-29:
In my case, it doesn´t work properly. It works when I boot the Pi, but after a while it stops working and the Tv (Samsung) doesn´t recognize it. I tried to force hotplug in the config.txt but it dind´n make any difference.

bairdy, 2015-04-29:
@doming52, what version of of the cec package are you running as anything below 9.2.2.0-1-1430117049 is broken with the latest kodi and doing what you have said as i had those issues but seem fixed with the latest update (running for 24 hours on still it has recognized it after the tv had been off overnight)

doming52, 2015-04-29:
I have the last version released 24 hours ago, but the problem is still on

RolfTimmerman, 2015-04-29:
Ok unfortunately it also stopped working for me again after a while :(

RolfTimmerman, 2015-04-30:
When CEC isn't working anymore and I connect to Kodi with Yatse than whole Kodi crashes... I'm trying to get a log but I can't login remotely anymore it seems.

jwamsterdam, 2015-04-30:
I'm not sure if this is the same, but I have the latest updates as of now, and have these messages in kodi.log:

17:02:46 T:3034959872  NOTICE: special://profile/ is mapped to: special://masterprofile/
17:02:46 T:3034959872  NOTICE: -----------------------------------------------------------------------
17:02:46 T:3034959872  NOTICE: Starting Kodi (14.2 Git:Unknown). Platform: Linux ARM 32-bit

[[skipped some rows]]

17:02:53 T:2957071376   ERROR: CecLogMessage - Open - vc_cec could not be initialised
17:02:53 T:2957071376   ERROR: CecLogMessage - could not open a connection (try 1)
17:02:54 T:3034959872   ERROR: DBus: Error org.freedesktop.UDisks.Error.Failed - Error mounting: mount exited with exit code 1: helper failed with:
                                            mount: according to mtab, /dev/root is already mounted on /home
                                            mount failed

OtakuWally, 2015-05-01:
I am not sure, if this is really connected to CEC, so please correct me, if I am wrong. This does NOT happen for the first time as well and maybe it is just a problem of Kodi in conjunction with screensavers, but it happens quite often, that kodi will just hang when it is in screensaver mode when I switch off my Samsung TV + my receiver (which it is connected to).

Following is the kodi.log excerpt for the time when it happens (according to the clock shown on the screensaver, 04:17 in this case):

04:13:33 T:1441788944  NOTICE: Thread FileCache start, auto delete: false
04:15:36 T:1376777232  NOTICE: Previous line repeats 2 times.
04:15:36 T:1376777232  NOTICE: Thread JobWorker start, auto delete: true
04:17:37 T:1376777232  NOTICE: Thread BackgroundLoader start, auto delete: false
04:17:37 T:1700787216  NOTICE: XBian : on saver deactivated
04:17:37 T:1700787216  NOTICE: XBian : on saver deactivated END
04:17:37 T:1376777232  NOTICE: Thread Timer start, auto delete: false
04:17:38 T:1368388624  NOTICE: Feedreader screensaver: exit
04:17:42 T:1376777232  NOTICE: Found resolution 1920x1080 (1920x1080) @ 60.00 - Full Screen
04:20:37 T:1778381840  NOTICE: Thread JobWorker start, auto delete: true
04:23:25 T:1778381840  NOTICE: Previous line repeats 1 times.
04:23:25 T:1778381840  NOTICE: Thread FileCache start, auto delete: false
04:23:32 T:1385165840  NOTICE: Previous line repeats 7 times.

I have the rest of the log as well in case it should be needed.
The system has been updated up to 2015-04-30, no additional packages have been installed yet.

OtakuWally, 2015-05-01:
Oh by the way, it is being said way to seldom anyway: Thanks to all of you guys from the XBian team for spending your time on the best Kodi distribution in our universe ;-)

OtakuWally, 2015-05-01:
Well, too late, now it crashed totally again :-(

OtakuWally, 2015-05-03:
And another one, since this happens EVERY TIME I switch off my receiver and TV. Should I enable DEBUG for any addons (or something else) to get more info?

Kodi crashed at 01:30

```
01:29:57 T:1389360144   ERROR: Failed to fetch checksum for directory listing http://hitcherrepo.googlecode.com/svn/trunk/addons.xml for repository xbmc.repo.hitcher.
01:29:57 T:1431303184  NOTICE: Thread FileCache start, auto delete: false
01:30:02 T:1431303184  NOTICE: Previous line repeats 2 times.
01:30:02 T:1431303184  NOTICE: Thread JobWorker start, auto delete: true
01:30:47 T:1698690064  NOTICE: XBian : on saver deactivated
01:30:47 T:1698690064  NOTICE: XBian : on saver deactivated END
01:30:47 T:1502606352  NOTICE: Feedreader screensaver: exit
01:30:47 T:1389360144  NOTICE: Thread Timer start, auto delete: false
01:30:52 T:1389360144  NOTICE: Found resolution 640x480 (640x480) @ 60.00 - Full Screen
01:30:52 T:1389360144  NOTICE: Found resolution 640x480 (640x480) @ 59.94 - Full Screen
01:30:52 T:1389360144  NOTICE: Found resolution 720x480 (720x480) @ 60.00 - Full Screen
01:30:52 T:1389360144  NOTICE: Found resolution 720x480 (720x480) @ 59.94 - Full Screen
01:30:52 T:1389360144  NOTICE: Found resolution 1280x720 (1280x720) @ 60.00 - Full Screen
01:30:52 T:1389360144  NOTICE: Found resolution 1280x720 (1280x720) @ 59.94 - Full Screen
01:30:52 T:1389360144  NOTICE: Found resolution 1920x1080 (1920x1080) @ 60.00 - Full Screen
01:30:52 T:1389360144  NOTICE: Found resolution 1920x1080 (1920x1080) @ 59.94 - Full Screen
01:30:52 T:1389360144  NOTICE: Found resolution 720x576 (720x576) @ 50.00 - Full Screen
01:30:52 T:1389360144  NOTICE: Found resolution 1280x720 (1280x720) @ 50.00 - Full Screen
01:30:52 T:1389360144  NOTICE: Found resolution 1920x1080 (1920x1080) @ 50.00 - Full Screen
01:30:52 T:1389360144  NOTICE: Found resolution 1920x1080 (1920x1080) @ 24.00 - Full Screen
01:30:52 T:1389360144  NOTICE: Found resolution 1920x1080 (1920x1080) @ 23.98 - Full Screen
01:30:52 T:1389360144  NOTICE: Found (1920x1080@60.000000) at 22, swapping with 16
01:34:13 T:1389360144  NOTICE: Thread JobWorker start, auto delete: true
01:39:40 T:1648358416  NOTICE: Previous line repeats 1 times.
01:39:40 T:1648358416  NOTICE: Thread FileCache start, auto delete: false
01:39:51 T:1389360144  NOTICE: Previous line repeats 7 times.
01:39:51 T:1389360144   ERROR: CCurlFile::FillBuffer - Failed: HTTP returned error 401
01:39:51 T:1389360144  NOTICE: CCurlFile::FillBuffer - Reconnect, (re)try 1
01:39:51 T:1389360144   ERROR: CCurlFile::FillBuffer - Failed: HTTP returned error 401
01:39:51 T:1389360144   ERROR: CCurlFile::CReadState::Connect, didn't get any data from stream.
01:39:51 T:1389360144   ERROR: Open - failed to open source <http://hitcher-repo.googlecode.co/svn/trunk/addons.xml.md5>
```

nbarrientos, 2015-05-11:
Sorry for hijacking the thread, but I'm experiencing the same:

```
22:01:05 T:2829054992  NOTICE: Previous line repeats 3 times.
22:01:05 T:2829054992  NOTICE: Thread VideoInfoScanner start, auto delete: false
22:01:05 T:2829054992  NOTICE: VideoInfoScanner: Starting scan ..
22:01:10 T:2829054992  NOTICE: VideoInfoScanner: Finished scan. Scanning for video info took 00:04
22:01:10 T:2829054992  NOTICE: Thread JobWorker start, auto delete: true
22:01:11 T:2905801744  NOTICE: Thread BackgroundLoader start, auto delete: false
22:20:21 T:2829054992  NOTICE: Previous line repeats 1 times.
22:20:21 T:2829054992  NOTICE: Thread JobWorker start, auto delete: true
22:30:29 T:2863858704  NOTICE: XBian : on saver
22:30:29 T:2863858704  NOTICE: XBian : on saver END
22:30:53 T:2923238416  NOTICE: Thread VideoInfoScanner start, auto delete: false
22:30:53 T:2923238416  NOTICE: VideoInfoScanner: Starting scan ..
22:30:57 T:2923238416  NOTICE: VideoInfoScanner: Finished scan. Scanning for video info took 00:03
22:30:57 T:2923238416  NOTICE: Thread JobWorker start, auto delete: true
22:30:57 T:2872247312  NOTICE: Thread BackgroundLoader start, auto delete: false
23:00:38 T:2829054992  NOTICE: Previous line repeats 1 times.
23:00:38 T:2829054992  NOTICE: Thread VideoInfoScanner start, auto delete: false
23:00:38 T:2829054992  NOTICE: VideoInfoScanner: Starting scan ..
23:00:41 T:2829054992  NOTICE: VideoInfoScanner: Finished scan. Scanning for video info took 00:02
23:00:41 T:2829054992  NOTICE: Thread JobWorker start, auto delete: true
23:00:42 T:2872247312  NOTICE: Thread BackgroundLoader start, auto delete: false
```

Kodi crashed at 22:30. Think it was right after switching to another CEC source on the TV. The "on saver" entries in the log look a bit suspicious.

```
ii  xbian-package-cec            9.2.2.0-1-141675380 armhf               libcec 2.1.4 for XBian (rpi/master)
ii  xbian-package-xbmc           14.2-1428397803     armhf               XBMC (rpi/rpi)
```

jwamsterdam, 2015-05-22:
Is there an update on the CEC issue yet? I still experience issues, primarily immediately after boot the CEC gets lost. But also when it does get recognized, after a few hours of inactivity, the CEC is lost again without being able to connect anymore. Only rebooting (a few times) solves the issue.

My log:
```
21:33:06 T:2956665872   ERROR: CecLogMessage - Open - vc_cec could not be initialised
21:33:06 T:2956665872   ERROR: CecLogMessage - could not open a connection (try 1)
21:33:19 T:3034554368   ERROR: DBus: Error org.freedesktop.UDisks.Error.PermissionDenied - Not Authorized
21:33:19 T:2923111440  NOTICE: Thread CECAdapterUpdate start, auto delete: false
21:33:19 T:3034554368   ERROR: DBus: Error org.freedesktop.UDisks.Error.Failed - Error mounting: mount exited with exit code 1: helper failed with:
                                            mount: according to mtab, /dev/sda1 is already mounted on /home
                                            mount failed
```
I'm happy to provide more instructions if advised.

CurlyMoo, 2015-05-22:
Yes, not sure if it fixes this issue, but there is:
https://github.com/xbianonpi/xbian-sources-libcec/pull/1

jwamsterdam, 2015-05-22:
> Yes, not sure if it fixes this issue, but there is:
> xbianonpi/xbian-sources-libcec#1

Thnx for your quick reaction, but how do I install? Can you give the commands for putty?

CurlyMoo, 2015-05-22:
All updates will always eventually be available through apt.

jwamsterdam, 2015-05-22:
Thnx, I'll be patient then ;-)

opdenkamp, 2015-05-23:
So it looks like you're doing something that calls cec-client by default? If so, then you have to change it to use Kodi's json-rpc instead, or only do whatever you're doing _before_ Kodi is started. Failing to do so will introduce new issues, and the bug that blocks things for 500 seconds is actually a good effect in this case, as it shows what is being tried here.

Even though this specific bug (seconds vs milliseconds) will be fixed soon, I'm going to introduce some new lock to prevent cec-client from accessing libCEC while Kodi is running, as indicated in the other ticket.

mk01, 2015-06-07:
@opdenkamp 

actually problem of this I pinpointed to be BFS/BFQ patches for 3.19 kernel. it behaves quite like all the users said and most close was @mkreisl - although even this is just symptom of some bad in kernel spinlock or similar. I spent few days on that with @pop and Martin? his ex-co-worker with code access to PI firmware. same kernel same patches same libcec same firmware is fine on RPI2.

@jwamsterdam @CurlyMoo @nbarrientos 
the kernel without those two patches is at staging (rpi-wheezy), sources pushed to repo and config too. 

@opdenkamp 
don't do any 'hard' blocking, we have imx6 cec driver with shared use - with each new libcec access it creates new device on the bus with separate LA, PA and manages inter client communication. 
the RPI adapter could be adapted for this too. It would need one change to rpi firmware to allow use of full uint16 as LA registering, not only 0x1-0xf (to allocate more LAs on RPI cec hw).

opdenkamp, 2015-06-08:
@mk01 libCEC has no support for things like this, and you'll confuse a number of TVs out there by doing this (including it's users by creating stale device entries). If you want to do this, you'll have to solve the problem in userspace, by writing a daemon that uses libCEC and that applications connect to without allocating a new logical address.

er0l, 2015-06-09:
Today I am faced with following after xbian-package-cec upgrade to xbian-package-cec 9.2.2.0-1-1433853654:

18:21:18 T:1960624128   ERROR: Unable to load libcec.so.2, reason: libXrandr.so.2: cannot open shared object file: No such file or directory
18:21:18 T:1960624128   ERROR: Initialise - failed to initialise bus cec

It seems that it was built with libXrandr2 support so following package needs to be installed in order to get it working again:

apt-get install libxrandr2

Strange how it was not resolved via dependancy.

Duvian, 2015-06-10:
er0l : after having installed the latest Xbian-image today on a B+ I was faced with the same problem. 
Your solution worked out of the box (Raspi connected to an LG 'Smart'-TV with Simplink (Pulse-Eight-CEC) turned ON)...
THANK YOU ;-)

libidae, 2015-06-14:
+1 for er0l's solution: Rpi2 with the latest Xbian, CEC worked fine until it crashed updating the library for the first time. Installed libxrandr2 through SSH and rebooted, CEC is fine again.

Thank you, I was about to give up on Xbian because this issue had already caused me to reload four times. Now I don't have to go back to an inferior media centre!

CurlyMoo, 2015-06-14:
The latest CEC packages already installs `libxrandr2` as a dependency.

closed 2015-06-14 by CurlyMoo