xbianonpi/xbian #720: Dependencies on Upstart

airend, 2015-04-27:
Things work pretty well on Debian 8/Jessie, except that xbian-update (and other xbian packages) have stringent dependencies on upstart. Now that Jessie is released, could these dependencies be fixed so that systemd-sysv can replace upstart?

CurlyMoo, 2015-04-27:
No, XBian is too dependent on upstart.

closed 2015-04-27 by CurlyMoo

airend, 2015-04-27:
Of course, I understand that's very much the case with Debian 7. I guess I'm wondering about plans for Debian 8 and systemd, where upstart is not needed. I remember reading that XBian devs already checked the compatibility of the current RPi packages on jessie.

CurlyMoo, 2015-04-27:
Yes, we already use jessie on the CuBox-i with upstart.

mk01, 2015-04-28:
@alrend

there is no plan to migrate the INIT to systemd.
but systemd is present and fully working (so starting / stopping own services etc) - that means if user/you want to adapt system with own/3rd party services with provided .systemd service files, current XBian setup is no obstacle.

airend, 2015-04-28:
Thanks for the information. For some reason, I thought systemd can deal with the legacy init infrastructure transparently.

mk01, 2015-04-28:
@airend 

yes, it can. but legacy is SYSV. so runlevels and /etc/init.d.

it doesn't handle upstart at all and never will as systemd is looking at the whole booting process from target perspective (what is needed to do to full-fill final "task"'s requirement - this iterates until all depends are managed (so systemd progressed to very begging of boot).

upstart goes from boot (what I can do now, as I finished this and that already).

but again, since udevd was integrated with systemd, as soon as upstart launches udevd, systemd starts in parallel too. only difference is that (as you wrote) systemd-sysv is missing so systemd is not started as init(pid=1) process (so is not managing the initial namespace).