xbianonpi/xbian #722: No SSA/ASS subtitles displayed with Kodi v15

mkreisl, 2015-04-30:
It's currently no critical issue, just for notice: 
Kodi 15 uses a new libass call `ass_set_line_position`, which does not exist in the current libass-0.10.0 library coming with debian wheezy.
I build a new library 0.10.1 which provides this new call. 
It can be downloaded from here: http://www.mediafire.com/download/ycbp3t2j2etcqdj/libass-0.10.1.tar.xz

CurlyMoo, 2015-04-30:
@mkreisl, on a sidenote. I see a lot of technical and valueble input from you. Care to join XBian:smiley: in whatever way?

mkreisl, 2015-05-06:
@CurlyMoo 
Yes, if I can help, why not. We can talk about that

CurlyMoo, 2015-05-06:
What would you like and what can you do?

mkreisl, 2015-05-06:
Hmm, that's the question. I already thought about that an unfortunately I still search for the right answer

CurlyMoo, 2015-05-06:
I'm ofc not going to give you access to everything directly. So if you want to work on the kernel with @mk01, i'll give you access to that. Or if you want to contribute to xbian-package-config-xbmc, i'll give you access to that.

Smultie, 2015-05-06:
Yeah, would be nice to have a Kodi 15 we can test ;)

mkreisl, 2015-05-06:
Yes, that's clear, and I don't want that. What I like to do is to help by bringing Kodi 15 on the way - as far as I can see there are a lot of things to do

mkreisl, 2015-05-07:
@Smultie 
If you want to test Kodi 15 you can download it from this folder:
https://www.mediafire.com/folder/2ngwumkbzw36z/XBian_Packages

Smultie, 2015-05-07:
Just to be sure; in running raspberry pi 2 with Debian Jessie, will that work?

mkreisl, 2015-05-07:
Oh, I don't know, probably not. RPi2 and Jessie, from where did you get this?. It sounds very experimental

Smultie, 2015-05-07:
Did that together with mk01. Works like a charm!

closed 2015-08-21 by mkreisl

noodle94, 2015-08-26:
Thanks mkreisl! i installed the libs today and it seems to work!

mer2329, 2016-03-21:
i have been searching for about 2 days, this fixed my sub woes.
thank you so much.