xbianonpi/xbian #724: dispman_vncserver framerate hardcoded to 15 FPS

fschaer, 2015-05-03:
Hi,

The framebuffer vnc server is really usefull for remotely debugging (even if my mouse strangely is moving down when I move it up ;) ) , thanks for this thing !
Unfortunately, its framerate is hardcoded to be limited to 15 fps.

Looking at the code (https://github.com/hanzelpeter/dispmanx_vnc/blob/master/main.c) it can be easily changed to something higher (#define PICTURE_TIMEOUT (1.0/15.0)) : I compiled it for 24 FPS and it worked quite well on my pi2, using only 30% of one core.

I don't know what would be the impact on a pi1 though...
Might you consider increasing the default FPS (and/or eventually allowing an option for it to be defined at run time ?) ?

Thanks

mk01, 2015-05-07:
@fschaer 

I'm currently rewriting the rpi vncserver to use our logic from imx6 vnc-server https://github.com/xbianonpi/xbian-package-vnc-server/blob/master/building/imx-vncserver/imx-vncserver.c .

just pushed
https://github.com/xbianonpi/xbian-package-vnc-server/tree/master/building/rpi-vncserver

---
just missed proper RGB offset, ... if you grab the code and fix it push back please.

fschaer, 2015-05-11:
Hi,

Thanks for the links. I took a bit more time to test, and noticed that once the full hd TV was plugged in, the frame rate was far for 15 fps anyway when using 1920x1080... that was only with the default resolution where there is no screen attached (640x480 ?).

Will test the new vncserver asap :]
Thanks

mkreisl, 2017-05-04:
After a loooong time a new version of VNC server is available, see http://forum.xbian.org/thread-3268-post-34113.html#pid34113

All known issues should be solved now

closed 2017-05-04 by mkreisl