xbianonpi/xbian #725: issue with mpeg2 and VC1 licenses breaking xbian plugin

fschaer, 2015-05-11:
Hi,

So, I just installed xbian, and must have skipped the license setup, since I don't own them anyway.
But when I try to open the xbian  plugin in kodi, it just doesn't open anything, it just says initialising or something like that in a popup and quickly reverts to the settings menu.

The kodi logs show this kind of exceptions : 
```

2:06:23 T:1701823504   ERROR: Traceback (most recent call last):
22:06:23 T:1701823504   ERROR:   File "/usr/local/share/kodi/addons/plugin.xbianconfig/default.py", line 222, in <module>
22:06:23 T:1701823504   ERROR:     win.show()
22:06:23 T:1701823504   ERROR:   File "/usr/local/share/kodi/addons/plugin.xbianconfig/default.py", line 94, in show
22:06:23 T:1701823504   ERROR:     self.onShow()
22:06:23 T:1701823504   ERROR:   File "/usr/local/share/kodi/addons/plugin.xbianconfig/default.py", line 125, in onShow
22:06:23 T:1701823504   ERROR:     catmodule = __import__('%s.%s'%(CATEGORY_PATH,module), globals(), locals(), [module])
22:06:23 T:1701823504   ERROR:   File "/usr/local/share/kodi/addons/plugin.xbianconfig/categories/10_system.py", line 264, in <module>
22:06:23 T:1701823504   ERROR:     class mpeg2License(Setting) :
22:06:23 T:1701823504   ERROR:   File "/usr/local/share/kodi/addons/plugin.xbianconfig/categories/10_system.py", line 267, in mpeg2License
22:06:23 T:1701823504   ERROR:     OKTEXT = _('xbian-config.license.updated')%_('xbian-config.licensempg2.name')
```

If I change the OKTEXT strings (?) to "ok" in both the mpeg2 and vc1 functions (mpeg2License and vc1License), then it fails on BADUSERENTRYTEXT.
If again I change this to "bad", then it opens a menu.
I tried with "_('xbian-config.dialog.ok')" instead of "ok" or "bad"

I'm not sure where the problem lies, I'm just discovering both the rpi and kodi - but not linux.
the config.txt in /boot does not contain anything related to any license.

How may I help to fix that ?

```
root@xbian ~ # cat /boot/config.txt 
initramfs initramfs.gz  0x1000000
gpu_mem_1024=256
gpu_mem_512=144
gpu_mem_256=100
initial_turbo=3
hdmi_ignore_cec_init=1
disable_overscan=1
disable_splash=1
root@xbian ~ # 
```

janbar, 2015-06-12:
Not alone, i have this issue too and licenses are configured for mpg2 and vc1 in my config.txt:
`06:25:28 T:1603269648   ERROR:   File "/usr/local/share/kodi/addons/plugin.xbianconfig/categories/10_system.py", line 267, in mpeg2License`

mkreisl, 2017-09-15:
Tested this week and after [this commit](https://github.com/xbianonpi/xbian-package-config-xbmc/commit/47aafb2c2ae78d15c50041eee21a8bd5ff5cf1e6) everything works as it should

closed 2017-09-15 by mkreisl