xbianonpi/xbian #726: french locale breaks the xbian plugin

fschaer, 2015-05-11:
Hi again.

So, if in the CLI xbian settings you change the system locale to "french (france)" and then go in the system settings => xbian , then most of the options cannot be "clicked"

The only things that can be "clicked" are Services and Preferences.

The kodi logs show these python exceptions : 

```
22:45:07 T:1800401936  NOTICE: XBian : Cannot add category: 10_system 
                                            (<type 'exceptions.UnicodeEncodeError'>, UnicodeEncodeError('ascii', u'XBian-config : Initialise category Syst\xe8me', 39, 40
, 'ordinal not in range(128)'), <traceback object at 0x6631dee0>)
22:45:08 T:1800401936  NOTICE: XBian : Cannot add category: 40_update 
                                            (<type 'exceptions.UnicodeEncodeError'>, UnicodeEncodeError('ascii', u'XBian-config : Initialise category Mettre \xe0 jour', 
42, 43, 'ordinal not in range(128)'), <traceback object at 0x44905d0>)
22:45:09 T:1800401936  NOTICE: on inint, before xbmc onInit
22:45:09 T:1800401936  NOTICE: <resources.lib.xbmcguie.xbmcContainer.GroupListControl instance at 0x475e5a8>
22:45:09 T:1800401936  NOTICE: <resources.lib.xbmcguie.xbmcContainer.GroupListControl instance at 0x4835ee0>
22:45:09 T:1800401936  NOTICE: <resources.lib.xbmcguie.xbmcContainer.GroupListControl instance at 0x4576850>
22:45:09 T:1800401936  NOTICE: <resources.lib.xbmcguie.xbmcContainer.GroupListControl instance at 0x45260a8>
22:45:09 T:1800401936  NOTICE: <resources.lib.xbmcguie.xbmcContainer.GroupListControl instance at 0x451aad0>
22:45:09 T:1800401936  NOTICE: on inint, after xbmc onInit
22:45:09 T:1800401936   ERROR: EXCEPTION Thrown (PythonToCppException) : -->Python callback/script returned the following error<--
                                             - NOTE: IGNORING THIS CAN LEAD TO MEMORY LEAKS!
                                            Error Type: <type 'exceptions.UnicodeEncodeError'>
                                            Error Contents: 'ascii' codec can't encode character u'\xe8' in position 19: ordinal not in range(128)
                                            Traceback (most recent call last):
                                              File "/usr/local/share/kodi/addons/plugin.xbianconfig/resources/lib/xbianWindow.py", line 35, in onInit
                                                xbmc.executebuiltin('Skin.SetString(%sloadingvalue,%s)'%(title,_('xbian-config.common.clicktoload')))
                                            UnicodeEncodeError: 'ascii' codec can't encode character u'\xe8' in position 19: ordinal not in range(128)

```

What I see in there is that it fails to understant the french accentuated chars "à" in "mise à jour" (==upgrade) and "è" in "système" (==system).

While searching for this error for kodi in general, I found the following link that might help fixing the issues : 
http://forum.kodi.tv/showthread.php?tid=144677

Looks like a unicode thing, no ?

Regards

fschaer, 2015-05-11:
Well. the dialog says "xbian-config.common.clicktoload", but in fact I can't open anything.
If I switch to the english locale, I can click everything - but I can't do that because of the 'WAF'...

CurlyMoo, 2015-05-13:
To all xbian-config-xbmc error reports. The package is not being maintained at the moment. Until someone can pick it up again, we can't do anything with the errors.

mkreisl, 2017-09-09:
Testing today, all categories are selectable.

So issue closed

closed 2017-09-09 by mkreisl