xbianonpi/xbian #728: Upstart bridge fails when using profiles without autologin

RabidArts, 2015-05-14:
Hi all

I had this situation with a kodi installation, specifically with profiles set up, and no auto-login.

Kodi would throw an error on startup (onscreen) saying that the xbian upstart bridge script failed.
Following that, a number of other things in kodi are unstable - I would get random reboots.
This is likely due to memory leaks.

I have already fixed the issue on my system, and am contributing the cause and solution to the community.

# Investigation: #

In the kodi log (~/.kodi/temp/kodi/log) you will find that the upstart bridge script fails on line 119.

## Example kodi log error: ##

    ERROR: EXCEPTION Thrown (PythonToCppException) : -->Python callback/script returned the following error<--
                                             - NOTE: IGNORING THIS CAN LEAD TO MEMORY LEAKS!
                                            Error Type: <type 'exceptions.KeyError'>
                                            Error Contents: ('result',)
                                            Traceback (most recent call last):
                                              File "/usr/local/share/kodi/addons/script.service.xbian.upstart-bridge/service.py", line 202, in <module>
                                                service = UpstartBridge()
                                              File "/usr/local/share/kodi/addons/script.service.xbian.upstart-bridge/service.py", line 20, in __init__
                                                self.monitor = XBMCMonitor(self)
                                              File "/usr/local/share/kodi/addons/script.service.xbian.upstart-bridge/service.py", line 120, in __init__
                                                json_version = json_response['result']['version']
                                            KeyError: ('result',)
                                            -->End of Python script error report<--

Looking on line 119 you find the following:

		# This should never fail
        json_response = json.loads(xbmc.executeJSONRPC('{"jsonrpc": "2.0", "id": "ApiVersion", "method": "JSONRPC.Version"}'))
        json_version = json_response['result']['version']
        self.jsonrpc_api_ver = (json_version['major'], json_version['minor'], json_version['patch'])

If you log the return value from JSONRPC, you'll find that it says 'Method not found'

BTW, the *This should never fail* comment really cracked me up.
Nice one **Anaconda**! :D (https://github.com/xbianonpi/xbian-package-upstart-xbmc-bridge/commit/de89836e2578ba4f1d18123bdba7d534e7df3b33)

# **TL;DR: ** #

# Cause: #
The JSON RPC service is not started when xbian starts, only once you log in.
If you have multiple profiles, well, you can guess the rest.

# Solution: #
Change the xbian upstart script to run on login, not on startup.

How, you say?
sudo nano /usr/local/share/kodi/addons/script.service.xbian.upstart-bridge/addon.xml
Find the line:
	<extension point="xbmc.service" library="service.py" start="startup"/>
Replace start="startup" with start="login"
Save, reboot, celebrate, profit!

# Confirmation: #
I logged out the response from JSONRPC.Version in the xbian upstart script - result is:
XBian XBMC-Upstart bridge: JSON RPC Version Response: { " i d " : " A p i V e r s i o n " , " j s o n r p c " : " 2 . 0 " , " r e s u l t " : { " v e r s i o n " : { " m a j o r " : 6 , " m i n o r " : 2 1 , " p a t c h " : 2 } } }

RabidArts, 2015-05-15:
Update: 
I'm experiencing a potential side effect of this solution - sometimes I have to log in twice. Possibly a race condition on the upstart script when set to run on login, and some other part of the system.

This doesn't bother me a lot, but obviously a problem for a stable build release.

I'll have a look at the logs and see if I can resolve it or at least get more information.

Possible solutions that come to mind:
1. Upstart on startup, but delay json rpc calls until login.
2. Find the script that crashes on login, and make it more resilient by , e,g, waiting and retrying

Smultie, 2015-05-18:
I don't use any profiles, and XBian has been throwing me the same error since a couple of updates, but the solution you posted isn't helping in my situation.

Thanks for trying to fix it, though!!

RabidArts, 2015-05-19:
Must be a different reason in your case... Could you link us a pastebin of
your kodi log?

Smultie, 2015-05-19:
Sure, see below:
http://pastebin.com/iNu0kBAY

RabidArts, 2015-05-26:
Update: Haven't had a chance to look again yet - hope to try looking at it tomorrow.

RabidArts, 2015-05-28:
Smultie:
I had a look at the code to work out where your problem is, but your version of the usptart bridge addon appears to be quite different.
It's either custom-modified or very old - I cannot find your version anywhere online.

My advice - uninstall and reinstall the upstart bridge from the xbian config dialog on ssh login.
But this is just my suggestion as an amateur - use at your own risk :)

Smultie, 2015-05-28:
Yeah, might be something I should've told you from the start; I'm running
XBian on Jessie.
http://apt1.xbian.org/pool/stable/rpi2-jessie/ and
http://apt1.xbian.org/pool/staging/rpi2-jessie/

RabidArts, 2015-05-28:
Smultie:
I can't seem to find the service.py file for the Jessie version.

Could you start a thread elsewhere (if you haven't already)?
And in it post the content of the
file /usr/local/share/kodi/addons/script.service.xbian.upstart-bridge/service.py

That way I can help advise you based on the actual code, and more
importantly, give other (more knowledgeable) people in the forum a chance
to answer :)

RabidArts, 2015-06-09:
In the pull request, mk01 says : "the needed "initctl emit xbmc-loaded" was moved directly into xbmc code, so non starting of this service (without login - in case of profiles) should not longer be an issue. or ?"

So gonna have a look sometime and try it out