xbianonpi/xbian #730: Cubox not booting

zilexa, 2015-05-25:
This is a general bug report related to the issues that has plagued every iMX release: the system fails to boot Kodi. Though in the forum http://forum.xbian.org/newreply.php?tid=2607 reports have been made the image from April 25 is the only one that works, but it only works on 2GB sd cards. 

What is needed to move forward and kill this bug? It seems to be preventing Cubox buyers to use XBian. The following issues already contain log files:

zilexa, 2015-06-11:
Update: testing the latest image (XBian_2015.06.06_imx6) resulted in errors on the TV, no successful boot. Image attached. Also SSH is not possible. 
[image]

zilexa, 2015-06-18:
This is with a Sandisk Ultra microSDHC class 10 with the I mark of UHS on it.

zilexa, 2015-06-18:
another user reported the specified Sandisk does not work.
A Transcend 8GB microSDHC class 6 does work!! So does micro sd kingston 8gb class 4. 

The Sandisk is 7.39GB, the Transcend is 7.46GB.

j4n-co, 2015-07-14:
This was driving me crazy! I experienced the exact same issue with my cubox-i pro and a 16GB Sandisk Ultra microSDHC class 10 card. With a Kingston 16GB microSDHC class 10 card, xbian works perfectly. The Sandisk card also fails when I try to copy the kingston card onto it, so I guess if you're on iMX hardware: stay away from the red and grey!

closed 2015-11-09 by mkreisl