xbianonpi/xbian #732: Sound drop outs and stuttering with I2S DAC connected.

coderpop, 2015-06-13:
Hi,

With XBMC 13.2 there was no problem, but with Kodi 14.2 (with the latest updates and upgrades to todays date) there is a problem with this.

The sound will blank for short periods, sometimes longer and sometimes shorter. It seems to get worse the longer a track is played.

I have noticed that a way to make the issue to appear is to simply move the mouse around the screen. This is repeatable. Also, scanning in a network directory to the library will also cause the problem.

If I start a track and then don't touch the mouse or the library scanning is stopped, then it is OK (from limited testing today)......

Hope someone can look into this and come up with a solution as others are reporting similar problems on the xbian forum.

CP

coderpop, 2015-06-14:
Just to add that the hardware acceleration for video doesn't help and it's not that the songs carry on playing and the sound goes off, the song is actually pausing for short intervals.
I'll try and generate a log and post if it helps (where to post though?).

coderpop, 2015-06-15:
Update...

Setting overclock from 'xbian' to 'high' has helped, but navigating menus still causes the sound to drop out, sometimes for many seconds at a time.

mkreisl, 2015-06-15:
@coderpop 
Did you have the dynamic priority feature enabled? If yes, could you turn it off and test again.

coderpop, 2015-06-15:
Hi,

Yes DP was on (must be the default setting as I never touched it). Turning it off and rebooting and testing again, it is better, but not fixed. For example I tried with a streaming radio source to avoid any problems at my server and when I went to open the music menus to play a flac file the sound blanked out for a while. Then, with the radio stream still running I scrolled up and down the artists menu and it stutters like hell.
I hope this helps narrow down the problem.

CP

f1vefour, 2015-06-17:
@mkreisl
When was the firmware last updated?

coderpop, 2015-06-17:
/opt/vc/bin/vcgencmd version
Jun 10 2015 20:40:12
Copyright (c) 2012 Broadcom
version a0f9db571a2bb980469502bd4de2c9793b3223c4 (clean) (release)

uname -a
Linux lounge 3.18.8+ #1 PREEMPT Thu Mar 5 03:43:34 CET 2015 armv6l GNU/Linux

I hope this helps!

mkreisl, 2015-06-18:
@coderpop 
> Turning it off and rebooting and testing again, it is better, but not fixed

Ok, let's make a further test:
1 make sure that the variable `DPRIORITY` is set to `no` in `/etc/default/xbmc`
2 1. is not enough to turn of DP completely. Unfortunately, kodi.bin will be launched with priority 49 and will newer changed later. So, to turn it completely off you have to modify the file /etc/init/xbmc.conf. Please change the line `exec schedtool -R -p 49 -e sudo -u $USER -g $GROUP -s $DAEMON $DAEMON_ARGS  $DAEMON_LIRC $DEBUG 2>&1 | grep -v ES30 || :` to `sudo -u $USER -g $GROUP -s $DAEMON $DAEMON_ARGS  $DAEMON_LIRC $DEBUG 2>&1 || :`
and run a new test

coderpop, 2015-06-19:
mkreosl,

You, Sir, are a genious! :)

I can confirm that on the one pi I tested this did the trick. No more stuttering sounds no matter what I do!

I admit that I don't understand what it all means, but I am very happy about this!

Thank You!

CP

closed 2015-06-23 by coderpop