xbianonpi/xbian #733: no way to downgrade from kodi to xbmc gotham

irisv, 2015-06-16:
see http://forum.xbian.org/thread-3029-post-29348.html

CurlyMoo, 2015-06-16:
Can you please describe the full problem as described here:
https://github.com/xbianonpi/xbian/blob/master/CONTRIBUTING.md

irisv, 2015-06-16:
From the forum thread linked above:

>    CurlyMo Wrote:  
>  Just search google how to downgrade packages with apt.
>
>Thanks for the response! You mean with apt-get install <package>=<version>, right? That's what I >tried last week, but there was only one version available and that failed to install due to unmet >dependencies. After that, it all got rather messy so I can't reproduce the exact errors.
>
>There's a detailed writeup for the downgrading process over on the Kodi wiki, but it's for Apple TV. >It says, "Due to the name change, the steps for downgrading between Kodi (starting at v14 and up) >to XBMC (v13.2 and lower) requires slightly different steps" - could that also be why my downgrade >attempt with apt failed? 

>    CurlyMo Wrote:  
>    If downgrading doesn't work well, please post an issue about it on github. It should be supported.
>
>Will do! Which package should I be downgrading and to which version (so I can post the results on >github)?
>
>And just to be clear, you're saying the kodi/xbmc name change thing is not an issue when >downgrading on Xbian? Or it shouldn't be? 

But then the thread veered of in a different direction and that at last set of questions never got answered.

Based on this thread - http://forum.kodi.tv/showthread.php?tid=213001 - this is what I've done: 

root@xbian ~ # apt-cache policy xbmc
xbmc:
  Installed: (none)
  Candidate: 2:11.0~git20120510.82388d5-1
  Version table:
     2:11.0~git20120510.82388d5-1 0
        500 http://mirrordirector.raspbian.org/raspbian/ wheezy/main armhf Packages
root@xbian ~ # apt-get install xbmc=2:11.0~git20120510.82388d5-1
Reading package lists... Done
Building dependency tree
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 xbmc : Depends: xbmc-bin (>= 2:11.0~git20120510.82388d5-1) but it is not installable
        Depends: xbmc-bin (< 2:11.0~git20120510.82388d5-1.1~) but it is not installable
        Recommends: python-qt3 but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

CurlyMoo, 2015-06-16:
Don't just post out in the open but actually read the page i linked and try to answer it as best as possible. This is not a user forum but a bug tracker. Different rules apply here.

irisv, 2015-06-16:
On the page you linked to it says to 'try to answer the following questions' - so that's what I tried to do. 

what did you do?
apt-cache policy xbmc
apt-get install xbmc=2:11.0~git20120510.82388d5-1

what did you expect to happen?
I was hoping to downgrade from kodi to xbmc gotham

what happened instead?
dependencies were not met 

I'm using a RPi model B. Kodi runs just fine, that's why I didn't include a log file. The problem is that I don't want kodi to run, I want to go back to xbmc.

I had assumed you were the same CurlyMo who had asked me to 'post an issue about it on github' on the kodi forum and were thus already familiar with the issue - sorry!  What other information do you need?

CurlyMoo, 2015-06-16:
I am, but the main developers are not reading the forum.

mk01, 2015-07-08:
@irisv 

assuming Gotham and XBian's packages, then it's xbian-package-xbmc, latest 13.2 was 13.2-RLS-1419734381.

apt-get install xbian-package-xbmc=13.2-RLS-1419734381

irisv, 2015-07-09:
Thanks mk01!

didn't work out for me though. I first got my xbian installation entirely up to date, then ran that apt-get install and here's what i got:

root@xbian ~ # apt-get install xbian-package-xbmc=13.2-RLS-1419734381
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Version '13.2-RLS-1419734381' for 'xbian-package-xbmc' was not found

btw- I think CurlyMoo removed (hid?) my previous error reports, but perhaps you still have access to those?

closed 2016-06-20 by mkreisl