xbianonpi/xbian #734: libcec3 missing dependency

hackedbellini, 2015-06-16:
After upgrading to the latest `xbian-package-xbmc` from staging on rpi2-jessie, `libcec` was upgraded to `libcec3`.

After that, I noticed that cec stopped working completely. Looking at the kodi log I found the following:

```
ERROR: Unable to load libcec.so.3.0, reason: liblockdev.so.1: cannot open shared object file: No such file or directory
```

After installing `liblockdev1` the error message disappeared and cec started working again. So, I think that `libcec3` should probably depend on that package too.

CurlyMoo, 2015-06-16:
It should be installed by default:
https://github.com/xbianonpi/xbian-package-cec/blob/master/build/imx6-cec3/content-tpl/DEBIAN/control#L7
https://github.com/xbianonpi/xbian-package-cec/blob/master/build/rpi-cec3/content-tpl/DEBIAN/control#L7

@mkreisl ?

hackedbellini, 2015-06-16:
@CurlyMoo here is the output of `apt-cache show libcec3` for me:

```
$ apt-cache show libcec3
Package: libcec3
Source: libcec
Version: 3.0.0-2-1434269756
Architecture: armhf
Maintainer: Lars Op den Kamp <lars.opdenkamp@pulse-eight.com>
Installed-Size: 4
Conflicts: libcec (<< 2.2), libcec1, libcec2
Replaces: libcec (<< 2.2), libcec1, libcec2
Provides: libcec, libcec1, libcec2
Filename: pool/staging/rpi2-jessie/l/libcec3/libcec3_3.0.0-2-1434269756_armhf.deb
Size: 199306
MD5sum: ce88f21fbf939401b9f3f1596dec80ab
SHA1: 0639bd47caa4c2ec97165c7fedd12ee0c54789ee
SHA256: 479cb79e7c5dcdbbf14774b7b3e625daee5066845fb4b3f015c5f22be676213f
Section: libs
Priority: optional
Multi-Arch: same
Homepage: http://libcec.pulse-eight.com/
Description: libCEC communication Library (shared library)
Description-md5: 4497f888803d6da622268354170e0e5d
This library provides support for the Pulse-Eight USB-CEC adapter.
.
This package provides the shared library.
```

The one that has those dependencies is `xbian-package-cec`:

```
$ apt-cache show xbian-package-cec
Package: xbian-package-cec
Status: deinstall ok config-files
Priority: optional
Section: utils
Installed-Size: 1711
Origin: XBian
Maintainer: mk01 <development@xbian.org>
  (at commit src a348cb900d383741c9809607ef9501e516ffc0f8, commit xbian repo 5661e8db9ea4e1fc2764d66ac4f64abab0476dee)
Architecture: armhf
Version: 9.2.2.0-1-1431069396
Config-Version: 9.2.2.0-1-1431069396
Provides: libcec-dev, xbian-package-cec-6q, xbian-package-cec
Depends: liblockdev1, libudev0 | libudev1, udev, liblockdev1, makedev
Conflicts: libcec-dev, xbian-package-cec-6q, xbian-package-cec
Description: libcec 2.2.0 for XBian (rpi/master)
Description-md5: f88d85d2e5cef41d6b68236515a95b24
```

But I cannot install it because `libcec3` provides it:

```
$ sudo apt-get install xbian-package-cec
[sudo] password for xbian: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package xbian-package-cec is a virtual package provided by:
  libcec2 2.2.0-1-1431042557
  libcec 3.0.0-2-1434269756
You should explicitly select one to install.

E: Package 'xbian-package-cec' has no installation candidate
```

CurlyMoo, 2015-06-16:
Maybe you can search for the correct file yourself and do a PR:
https://github.com/xbianonpi/xbian-package-cec

hackedbellini, 2015-06-16:
And btw, shouldn't `xbian-package-xbmc` depend on `libcec` instead of `libcec3`?

Its control looks like this:

```
$ apt-cache show libcec
Package: libcec
Version: 3.0.0-2-1434269756
Architecture: armhf
Maintainer: Lars Op den Kamp <lars.opdenkamp@pulse-eight.com>
Installed-Size: 4
Depends: libcec3 (= 3.0.0-2-1434269756), python-libcec, cec-utils, libcec-dev
Provides: xbian-package-cec
Filename: pool/staging/rpi2-jessie/l/libcec/libcec_3.0.0-2-1434269756_armhf.deb
Size: 2006
MD5sum: 1a126800ce031bd83e9864781885f38c
SHA1: 78870f8c53802759b0975f9bc6f084a6a297db9d
SHA256: f39ac51856a61a2474fbef8fb498e799b9ad62d2e59211423a16d04a7f776cfe
Section: libs
Priority: optional
Homepage: http://libcec.pulse-eight.com/
Description: Meta package libCEC.
Description-md5: 150735e0ae5e5b86e3730a666cdd6377
```

It depends on `libcec3` and some other utilities. But it is actually not installable because one of its dependencies, `cec-utils`, depends on `libcec2`.

> Maybe you can search for the correct file yourself and do a PR:

Np, I can do that later on the week when I have some more time for that.

mkreisl, 2015-06-16:
@CurlyMoo 
for me everything looks ok, but I don't have a jessie system right now. So I can not make further tests. Maybe we should wait if @hackedbellini finds out whats going wrong.

@hackedbellini 
> And btw, shouldn't xbian-package-xbmc depend on libcec instead of libcec3?

No, libcec3 will replace libcec for latest builds in staging area

hackedbellini, 2015-06-18:
I was trying to build the package here, using the instructions on [this page](https://github.com/xbianonpi/xbian/wiki/Building-XBian-packages).

I could create the armhfjessie schroot, but the script is failing on `do_other` part. First it was because the [xbian-package-repo](https://github.com/xbianonpi/xbian/blob/master/create.buildroot#L118) it was trying to download didn't exist. After fixing that, it is now failing to [install config_install_deb](https://github.com/xbianonpi/xbian/blob/master/create.buildroot#L123) with this:

```
The following packages have unmet dependencies:
 libcurl4-gnutls-dev : Depends: libcurl3-gnutls (= 7.38.0-4+deb8u2) but 7.37.1-1 is to be installed
E: Unable to correct problems, you have held broken packages.
```

That is because there's a pin on `libcurl` to force the version on xbian repositories instead of the ones on debian.

Do you know how to workaround that? Can I just remove that pin or would that cause other issues for me?

Also, another question: When I fix that issue, what config should I try to build? I noticed that both `rpi` and `rpi-cec3` builds for `wheezy`, which is not the case here. Should I use `imx6-cec3`?

And regarding the issue itself: Although I still don't know where the problem is (will have to make the build work to do some experiments), I just noticed something: If you see [my output of the libcec3 control](https://github.com/xbianonpi/xbian/issues/734#issuecomment-112429784), it is actually the same as the original one from https://github.com/xbianonpi/xbian-sources-libcec/blob/masterv3/debian/control. So, it seems that the changes on `content-tpl/DEBIAN/control` are not being applied.

mkreisl, 2015-06-18:
@hackedbellini 
I've no a lot of experiences using jessie, so I cant give an answer for 
> Do you know how to workaround that? Can I just remove that pin or would that cause other issues for me?

> Should I use imx6-cec3?
No, because in this tree there is no compilation for RPi.
You have to make a new tree `rpij-cec3` for example and copy all content of the `build/rpi-cec3` to `build/rpij-cec3` and change in `build/rpij-cec3/config` `config_build_env=schroot -c armhfwheezy -p --` to `config_build_env=schroot -c armhfjessie -p --`. You also need the firmware files in /opt/vc in the , a simple copy of them from armhfwheezy to armhfjessie schroot should be enough

hackedbellini, 2015-06-20:
@mkreisl thanks for the instructions, it worked!

So, that `config_deb_pkgcreatenative` seems to be the reason of this issue. By Removing it from the `rpij-cec3` config (more specifically, reverting [this commit](https://github.com/xbianonpi/xbian-package-cec/commit/d037c44913ad3b3e0f21a1716cba642267a0a50e) for it) made the deb be created with the right dependencies.

Actually, by using the `config_deb_pkgcreatenative`, the whole `debian` directory from upstream is used and thus, its original control will dictate the packages that will be created, the dependencies and etc. For example, the upstream's control created packages for `cec-utils`, `libcec2`, `libcec3`, `python-libcec`, etc, while the one from `content-tpl` created only the `libcec3`.

I took a `debdiff` between the `libcec3` package created by xbian's control (the first file) and the upstream's one (the second file). Here is the result:

```
debdiff libcec33.0.0-2-1434807485.deb libcec3_3.0.0-2-1434806544_armhf.deb                                                                                      hackedbellini@aenima
[The following lists of changes regard files as different if they have
different names, permissions or owners.]

Files in second .deb but not in first
-------------------------------------
-rw-r--r--  root/root   /usr/share/doc/libcec3/copyright

Files in first .deb but not in second
-------------------------------------
-rw-r--r--  root/root   /usr/bin/pyCecClient.py
-rw-r--r--  root/root   /usr/include/libcec/cec.h
-rw-r--r--  root/root   /usr/include/libcec/cecc.h
-rw-r--r--  root/root   /usr/include/libcec/cecloader.h
-rw-r--r--  root/root   /usr/include/libcec/cectypes.h
-rw-r--r--  root/root   /usr/include/libcec/version.h
-rw-r--r--  root/root   /usr/include/platform/os.h
-rw-r--r--  root/root   /usr/include/platform/posix/os-socket.h
-rw-r--r--  root/root   /usr/include/platform/posix/os-threads.h
-rw-r--r--  root/root   /usr/include/platform/posix/os-types.h
-rw-r--r--  root/root   /usr/include/platform/sockets/cdevsocket.h
-rw-r--r--  root/root   /usr/include/platform/sockets/socket.h
-rw-r--r--  root/root   /usr/include/platform/sockets/tcp.h
-rw-r--r--  root/root   /usr/include/platform/threads/atomics.h
-rw-r--r--  root/root   /usr/include/platform/threads/mutex.h
-rw-r--r--  root/root   /usr/include/platform/threads/threads.h
-rw-r--r--  root/root   /usr/include/platform/util/StdString.h
-rw-r--r--  root/root   /usr/include/platform/util/StringUtils.h
-rw-r--r--  root/root   /usr/include/platform/util/atomic.h
-rw-r--r--  root/root   /usr/include/platform/util/buffer.h
-rw-r--r--  root/root   /usr/include/platform/util/timeutils.h
-rw-r--r--  root/root   /usr/include/platform/util/util.h
-rw-r--r--  root/root   /usr/lib/arm-linux-gnueabihf/libcec.a
-rw-r--r--  root/root   /usr/lib/arm-linux-gnueabihf/libplatform.a
-rw-r--r--  root/root   /usr/lib/pkgconfig/libcec.pc
-rw-r--r--  root/root   /usr/lib/pkgconfig/platform.pc
-rw-r--r--  root/root   /usr/lib/platform/platform-config.cmake
-rw-r--r--  root/root   /usr/lib/python2.7/dist-packages/cec/__init__.py
-rw-r--r--  root/root   /usr/lib/python2.7/dist-packages/cec/_cec.so
-rw-r--r--  root/root   DEBIAN/debian-binary
-rw-r--r--  root/root   DEBIAN/md5sums
-rwxr-xr-x  root/root   /usr/bin/cec-client-3.0.0
-rwxr-xr-x  root/root   /usr/bin/cecc-client-3.0.0
-rwxr-xr-x  root/root   DEBIAN/postinst
-rwxr-xr-x  root/root   DEBIAN/postrm
-rwxr-xr-x  root/root   DEBIAN/preinst
-rwxr-xr-x  root/root   DEBIAN/prerm
lrwxrwxrwx  root/root   /usr/bin/cec-client -> cec-client-3.0.0
lrwxrwxrwx  root/root   /usr/bin/cecc-client -> cecc-client-3.0.0
lrwxrwxrwx  root/root   /usr/lib/arm-linux-gnueabihf/libcec.so -> libcec.so.3.0

Control files: lines which differ (wdiff format)
------------------------------------------------
Conflicts: [-libcec-dev, libplatform-dev, xbian-package-cec-6q, xbian-package-cec-] {+libcec (<< 2.2), libcec1, libcec2+}
Depends: {+libc6 (>= 2.4), libgcc1 (>= 1:4.4.0),+} liblockdev1, [-libudev0 | libudev1, udev, liblockdev1, makedev-] {+libstdc++6 (>= 4.5), libudev1 (>= 183)+}
Description: [-libcec 3.0.0-] {+libCEC communication Library (shared library)+}
{+ This library provides support+} for [-XBian (rpij/cec3)-] {+the Pulse-Eight USB-CEC adapter.+}
{+ .+}
{+ This package provides the shared library.+}
{+Homepage: http://libcec.pulse-eight.com/+}
Installed-Size: [-5344-] {+28+}
Maintainer: [-mk01 <development@xbian.org>-]
[-  (at commit src 86315d9a72e52fb046efe61d9f47a78c97d440ac, commit xbian repo d037c44913ad3b3e0f21a1716cba642267a0a50e)-]
[-Origin: XBian-] {+Lars Op den Kamp <lars.opdenkamp@pulse-eight.com>+}
{+Multi-Arch: same+}
Provides: [-libcec-dev, libplatform-dev, xbian-package-cec-6q, xbian-package-cec-] {+libcec, libcec1, libcec2+}
{+Replaces: libcec (<< 2.2), libcec1, libcec2+}
Section: [-utils-] {+libs+}
{+Source: libcec+}
Version: [-3.0.0-2-1434807485-] {+3.0.0-2-1434806544+}
```

mkreisl, 2015-06-20:
@hackedbellini 
> thanks for the instructions, it worked

Good to hear ;-)
I think it's better to ask @mk01, what you already did

mk01, 2015-07-07:
@hackedbellini 

to respect to opening post - I miss-ordered two dh_* scripts in the main builder script.
just committing the changes. then also liblockdev1 is in the dependencies.