xbianonpi/xbian #737: Rpi 2 issues with xbian 1.0

coderpop, 2015-06-18:
Hi,

A few things about xbian 1.0 on rpi2.

1. Setting locale in xbian doesn't change units of measurement in kodi (ie setting to English UK still give temp in degF and not degC when in kodi>system>infomation>hardware.
2. Within Kodi>System>settings>xbian>network, the settings are not held when entered vai kodi menu and you have to ssh in to set them from xbian-config. Also the notice "are you sure to apply modification" keeps popping up, even if no changes have been made! 
3. When trying to scan the music folder (SMB link) the display would blank for a few seconds and then come back up, but no scanning happened. This happened a few times, but after successfully scanning in a video folder (on the same server) another attempt to scan the music folder to library succeeded. Odd.

Hope this helps, thanks for all the good work.

CurlyMoo, 2015-06-18:
It's more handy if you create open an issue for each problem with a clear description of the problem in both the title and body. Also post all version information possible.

coderpop, 2015-06-19:
Aha! - OK Understood.

Would you like me to delete this post and make three seperate ones now?

Thanks.

CurlyMoo, 2015-06-19:
Make

coderpop, 2015-06-19:
OK, I'll do it in the morning!