xbianonpi/xbian #738: IR "buffer" full

johnjore, 2015-06-18:
As per http://forum.xbian.org/thread-3041-post-29381.html#pid29381, there appears to be a problem sending many IR commands:

--
Hi!
 
I've downloaded latest xbian, updated what could be updated and configured lirc and it works, but only the first 8-10 key presses on the remote. Then I need to wait, 2-5 seconds before another keypress works. the longer I wait, the more buttons work until a "buffer" is full again. So, if pressing buttons very slowly and waiting 2-5 seconds between, then problem never occours.
 
I've tested this on two different receivers: Built in receiver on Solidrun (cubox) (Name: gpio_ir_recv) + hp media center USB device (Name: Media Center Ed. eHome Infrared). Same problem with both receivers.
 
Remote is the standard microsoft mce style.
 
Downloaded and compiled latest LIRC (0.9.3-devel) - Same issue. Problem with some settings within the kernel xbian uses?
 
Anyone else seen this or know what's going on?
 
JJ

CurlyMoo, 2015-06-18:
Can you try the lirc_gpio module? Does the same thing happen when running `mode2`?

johnjore, 2015-06-18:
Not sure if its supposed to terminate after each keypress:
# mode2 -d /dev/lircd
mode2: could not open logfile "/root/.cache/mode2.log"
mode2: No such file or directory
Using device: /dev/lircd
Using device: /dev/lircd
Warning: Running as root.
space 3158064
pulse 3158064
space 6710839
pulse 6709815
space 3158048
pulse 5850443
pulse 7151665
space 7566693
readdata() failed

After running mode2 8-10 times it hangs here:
	mode2 -d /dev/lircd
	mode2: could not open logfile "/root/.cache/mode2.log"
	mode2: No such file or directory
	Using device: /dev/lircd
	Using device: /dev/lircd
	Warning: Running as root.

A keypress does nothing. After waiting a few seconds and then pressing a key, it responds with the usual "space" & "pulse".

CurlyMoo, 2015-06-18:
Can you place use markdown in your post (read about it on google if you are not familiar with it).

johnjore, 2015-06-18:
1) Calm down and let me edit the post before complaining. 2) not everyone lives on github and wants to learn another editing format. Focus on the issue reported...

CurlyMoo, 2015-06-18:
> 1) Calm down and let me edit the post before complaining

Is was just a question. The edit did not do what i requested.

> 2) not everyone lives on github and wants to learn another editing format. Focus on the issue reported...

Not everyone spends a great deal of their spare time into creating free software. The least we can ask of users is to report problems in a decent manner. With enough information to work with and properly readible.

johnjore, 2015-06-18:
Sorry, no. I've reported a problem in a clear manner. Learing markdown is not something I have any interest in spending time on. I spend plenty of my time developing free software and fully appreciate the need for clear bug information, but text formatting of bug reports is not high on my requirements list for helping others...

CurlyMoo, 2015-06-18:
If that's too much too ask then i also don't feel invited to spend time into finding a solution to the problem.

closed 2015-06-18 by CurlyMoo

johnjore, 2015-06-18:
Nice!  No wonder xbian has the repuation it has.

CurlyMoo, 2015-06-18:
I think OpenElec and OSMC will suit your specific needs better then XBian will.