xbianonpi/xbian #739: [imx6] Unsharp video and strange viewing mode

zilexa, 2015-06-18:
with the latest build from June 13:

1. video is clearly fuzzy/unsharp compared to playback using OpenElec 5.0.8. Even high res 1080p bluray material looks not 100% sharp as it used to. I tested with normal hdtv (non720p) material and with 300:Rise of an Empire Bluray 1080p backup of 18GB. It is definitely unsharp overall.

2. Kodi UI looked a bit streched, calibration via Settings>System>Video seemed to fix it, the square was too wide. Without messing with settings, normal hdtv material like Big Bang Theory (non-720p) shows black bars. I have never seen that with such material. Not by default. When I check Video settings in playback mode "VIEW MODE = Normal". This is what I am used to. but I have to change it to "stretch 16:9" to get full screen video.

3. When I play a 1080p bluray, I get extra fat black bars and need to use "original size" to get normal black bars and people with normal faces. Normally, I never even had to touch that setting. Is this a 

4. Subtitles seem a bit big/stretched (even after calibration).

Smultie, 2015-06-18:
You know you're on the XBian Github, not Openelec.... ?

CurlyMoo, 2015-06-18:
He said, "compared to".

Smultie, 2015-06-18:
I should really start learning to read...... :dash:

zilexa, 2015-06-19:
I can only compare with OpenElec since it's the only other Kodi 14 distribution for Cubox :)
Geexbox is still on 13 I think.

zilexa, 2015-06-20:
@Smultie can you confirm these issues? 

After playing a few videos I noticed a hiccup or pause of 1-2 seconds, every 3-5min.

zilexa, 2015-06-28:
Is there any theory why this issue exist and what could fix it?

gitr0b, 2015-09-09:
Does the screen resolution in the video settings match to the native resolution of your display? 
(I once noticed it being different. Maybe because my TV was still switched off while booting xbian.)

mk01, 2015-09-11:
@gitr0b 

Even if modelist changes, it is not an issue as far as new modelist has from X/Y dimensions standpoint same resolution. It will be rematched.

If there is no such resolution available, Kodi will choose closest one. At some corner cases this can lead to change/save as default something different as expected... (like for instance monitor will be changed from CEA modes compatible to DVI with Vesa modes only, etc - there Pixelratio is not considered while looking up for new mode as it would never match). 

But diff pixelration should also not lead to aspect changes as each mode has recalculated PAR from monitor SAR... 
(but of course everything can happen ;) )