xbianonpi/xbian #741: VNC upgrade not working at all on RPi 2

coderpop, 2015-06-20:
After todays VNC server upgarde to xbian it is now not possible to start the VNC server from the xbian-config menu on Rpi 2 and no connection is possible via vnc.

CP

mk01, 2015-07-08:
there was reissue of ver 2.0.0 (ver 2.0.0fix)

coderpop, 2015-07-13:
Still not working on Rpi 2. VNC server never starts up. I will check RPi 1 tonight if I get chance.

baba420, 2015-07-14:
Yes, unfortunately even after the ver 2.0.0fix VNC still doesn work.
(Rpi2, fully up-to-date)

coderpop, 2015-07-23:
Can anyone tell me how to revert to the old working vnc server please?

mkreisl, 2015-07-23:
@coderpop 
Not tested because still not needed, for example look here:
http://blog.andrewbeacock.com/2007/03/how-to-install-specific-version-of.html

mkreisl, 2017-05-04:
New version available, see http://forum.xbian.org/thread-3268-post-34113.html#pid34113

So this issue can be closed now :smile:

closed 2017-05-04 by mkreisl